betaflight/betaflight #6683: RC Smoothing doesn't work well with CRSF (12 channels)

haplm, 2018-09-02:
RC Smoothing is producing following patterns with CRSF (BF 4.0 build 1092):
[image]

When in real flight, the issue seems to be even more pronounced:
[image]

Steps to reproduce the behavior:
Crossfire is running at full speed (Q X7 with the hardware inverter mod)
CRSF link set to 12 channels
```
rc_smoothing_type = FILTER
rc_smoothing_input_hz = 0
rc_smoothing_derivative_hz = 0
rc_smoothing_debug_axis = ROLL
rc_smoothing_input_type = BIQUAD
rc_smoothing_derivative_type = BIQUAD
debug_mode = RC_SMOOTHING_RATE
```

I can compare the situation depicted above with FrSky (12ch, cutoffs set manually to 25Hz), and what I'm getting here is much worse than that - hence this bug report.

Flight controller configuration:
[BF40_1092_diff_all.txt](https://github.com/betaflight/betaflight/files/2343633/BF40_1092_diff_all.txt)

Output of rc_smoothing_info:
```
# 150 Hz
# rc_smoothing_info
# RC Smoothing Type: FILTER
# Detected RX frame rate: 6.667ms
# Input filter type: BIQUAD
# Active input cutoff: 67hz (auto)
# Derivative filter type: BIQUAD
# Active derivative cutoff: 67hz (auto)

# 50 Hz
# rc_smoothing_info
# RC Smoothing Type: FILTER
# Detected RX frame rate: 19.957ms
# Input filter type: BIQUAD
# Active input cutoff: 23hz (auto)
# Derivative filter type: BIQUAD
# Active derivative cutoff: 23hz (auto)
```

BB log (use the second "flight"):
https://drive.google.com/open?id=1xlGqJLETr61N1qF__wdjEv6qDddUnr-P
(it also captures change from 50Hz update frequency to 150Hz)

kenkamm, 2018-09-02:
Is the stair stepping occurring when crossfire switches from 150 to 50hz? That’s what I noticed in my logs. So it doesn’t happen that many times during flight and I was told this is normal. I guess it depends on how often crossfire is switching.

haplm, 2018-09-02:
I don't think so. There's just one switch in the attached log, for example. 

During the real flight (https://drive.google.com/open?id=1X81hiTHBQljtrfyLMsjSiKAToF-n4Vlg), it is there pretty much all the time, yet there were just few switches.

Is there an explanation of what the debug values produced by debug_mode = RC_SMOOTHING_RATE mean?

etracer65, 2018-09-03:
Everything seems to be working normally, but you have a lot of glitches coming from your crossfire receiver.  Looking at debug[0] which shows the actual received RX packet rate, you can see the receiver running at 50hz with lots of glitches.  Then it switches to 150hz and continues to have glitches.  I don't know the internals of Crossfire so I can't tell you what's causing this or whether it's hardware or firmware on the RX/TX.

[image]

If you overlay debug[0] with rcCommand and zoom in you'll see the "steps" correspond to the glitches in the rx packets.

[image]

For `RC_SMOOTHING_RATE`:
debug[0] = the current rx frame rate sample
debug[1] = retraining sample count
debug[2] = currently active rx rate used to set the filters
debug[3] = internal state flag indicating the stage of retraining

etracer65, 2018-09-03:
The "glitches" are actually dropped frames. When in slower 50hz (20ms) mode the glitches are showing 30ms as that's the maximum limit that the Betaflight firmware waits before processing. A frame was skipped and not sent to the flight controller and this would have resulted in 40ms delay.

During 150hz mode you can see that the glitches are around 13.3ms which is double the 150hz rate of 6.67ms.  So packets were being dropped and not sent from the rx to the flight controller.  I don't know about Crossfire, but for other protocols this indicates poor signal with the transmission not making it from the TX to the RX and subsequently getting to the flight controller.

haplm, 2018-09-03:
Understood.

The log that also captures RC_SMOOTHING_RATE information was intentionally recorded in a difficult environment, so I could demonstrate both the switch and dropped frames (I have simply positioned the transmitter so the antenna alignment was worst possible, and moved across the house. 

The second log (regular flight) was in open environment, and I was flying all around me (so there were probably sports with worse signal too). LQ (Link Quality) was decent (90%+) for the whole flight. This tells me that dropped frames are (will be) the reality, and there should be a way how to deal with them. 

Is there a way how to increase resiliency to dropped frames (perhaps at cost of some additional latency)?

etracer65, 2018-09-03:
That would be a question for TBS I think. The flight controller can't do anything about the receiver not sending rx frames.  And I don't think what you're seeing is absolutely normal.  I've seen other crossfire logs and have not seen that problem - at least not to the extent you're having.

haplm, 2018-09-04:
@etracer65 - I'm sorry, but I have to disagree that this is only my problem, or that the rate of dropped frames is significantly higher compared to other logs. I have to admit that I don't have a broad enough set of logs (yet), but I have downloaded at least what's in issue #6239, which is related to this, and that log shows very similar fluctuation of rx frame rate:
[image]

Both logs side-by-side:
[image]

Yes, I have some higher spikes in my log, but he actually has few which are even worse. So IMO the change of frame rate is a reality which needs to be somehow dealt with. Do I really need to collect logs from even more people to prove that this is a real problem?

etracer65, 2018-09-04:
@haplm You referenced and old pre-3.4 issue that has been resolved and closed.  RC smoothing does properly deal with Crossfire changing from 150hz to 50hz mode and back.  At the time the receiver transitions there will be a short period (< 1 second) where the filtering will be suboptimal while the smoothing logic retrains on the new observed rate.  Then the filters adjust to the new frame rate.

So I don't believe there are any problems.  However if you want to examine further please sure to capture logs with `debug_mode = RC_SMOOTHING_RATE`.  Note that this has been reviewed multiple times with many logs and all show the logic behaving as it should.

etracer65, 2018-09-05:
Regardless, what do you expect we should do about dropped packets?  If the receiver doesn't send any data what do you hope that the flight controller is going to do to fix it?  Filter based smoothing actually handles it much better than interpolation, but ultimately we have no data to work with.

haplm, 2018-09-05:
That's the question, I know. Perhaps some approximation? I'm pretty sure that even simple linear regression might be better than nothing, but I'm sure there might be even more suitable algorithms able to guess the next value in case that the frame doesn't come. I'm pretty sure that in most cases this will be better than staying on the previous value, which I guess is what's happening now, right?

etracer65, 2018-09-05:
We can't simply "guess" what the next value from the receiver will be.

Also you're misinterpreting the log screenshots you posted above.  The debug[0] traces in both just show minor jitter related to variances in the PID loop time.  This has no little to no effect on filter based smoothing (it has a big effect on interpolation and one of the primary reasons filter based smoothing was developed).  Neither of this images show any "dropped" frames.

haplm, 2018-09-06:
I see. Since I have another log now, I'll use screenshot out of that log:
[image]

Are the bigger spikes in debug[0]. Are those dropped frames? Based on one of your previous screenshot, I assume they are. Sorry for overlooking that one.

If that's true, then I'm curious why there's such ugly RC Command trace for the pitch axis before the frame is lost. This just doesn't look right to me...

Full log (different copter, different transmitter) is here:
[BWhite_rc_smoothing_rate_test.BFL.zip](https://github.com/betaflight/betaflight/files/2357963/BWhite_rc_smoothing_rate_test.BFL.zip)

As for guessing - I'm not an expert in this. I know something, but definitely not enough. But IMO if you detect that RX didn't send an update on time, then you basically have 2 choices. Use values from last frame received, or try to approximate. I believe that approximation may work, as long as you limit it to several consecutive lost frames. In other words, it is clear that you can't be approximating for longer than let's say 2 or 3 cycles (60ms - 90ms).  But I'm really curious why this is actually happening. I have always thought that this approximation is done on the RX side...

DieHertz, 2018-09-06:
Approximation happens betweem a number of given samples, what you're talking about is extrapolation/prediction and it's not that easy to do right in each case, and impossible to do in general.
We could expect the receiver to provide a solid refresh rate and predict values if an update doesn't come in time, but there are multiple shortcomings to this:
* There are receivers that have floating refresh rate, e.g. some older FrSky receivers do that
* What if due to clock differences an update comes right after you decided it's "too late" and used a predicted value for your calculations? What if this lately arrived value is highly different from the one you guessed?

haplm, 2018-09-06:
Thanks for the response, DieHertz.

My bad, yes, extrapolation/prediction. And yes, I would also expect the receiver to provide solid refresh rate and predict values in case of frames being lost during the transmission. So I have opened a support ticket with TBS about that, as etracer suggested (and pointed TBS to this ticket).

However, I see quite ugly trace even in case when there are no dropped frames (I hope that I understand the debug traces correctly now). Do you have any theory why this is happening?

etracer65, 2018-09-06:
@haplm Turn expo off in blackbox viewer for the rcCommand traces (set the expo to 100% or click the expo on/off button).  The viewer has some strange defaults for rcCommand expo and it's making the graphs look weird.  For example, the section from your screenshot above should look like this:

[image]

As you can see, no problems with the area you were concerned about.  The section on the left half shows that your receiver switched to 50hz mode and rc_smoothing started retraining.  However the receiver switched back to 150hz before the retraining completed.  So during that time the smoothing isn't great because the filters haven't adjusted.

[image]

Another interesting section at around 11.9 seconds.  Here the receiver switched to 50hz and the retraining happened normally.  Then after being at 50hz for approximately 1.8 seconds it switched back to 150hz mode.  The rc_smoothing logic implements a delay of 2 seconds after a switch before retraining will start again (to prevent flipping the filter around too quickly).  So there was a period where the receiver was back at 150hz, a delay and then rc_smoothing retrained again.

[image]

In this next section I don't know what the receiver was doing. It was in 50hz mode and it looks like it was trying to switch back to 150hz but was dropping frames since some of the intervals appear to be multiples of the 150hz rate (but this is only a guess).

[image]

Throughout the entire log I don't see any problems with the rc_smoothing behavior.  In every case it responds correctly to the signal from the receiver.

However I can say that I've never seen a Crossfire log that has switched so frequently or spent so much time at 50hz.  I still believe that you have some issue with your tx/rx or with your antenna placement or health.

haplm, 2018-09-07:
Funny thing is, that this log isn't my quad. It is a log which I've got from a guy in one of the BF tuning groups on FB (I should have perhaps mention it more clearly in my post).

Let's put the switching aside for a while (perhaps just flying into some blind spots). But even if I turn off expo, I see patterns like this (throttle, no retraining, no dropped frames):
[image]

Anyway, I'm going to fly a bit more tomorrow, and I'm going to use diamond antenna on TX instead of the simple dipole. Let's see if it helps with dropped frames. Anything to try to get some additional insight into what's happening here?

etracer65, 2018-09-07:
I don't see any problems with those traces at all.  Looks perfectly normal.  The filter cutoffs are chosen as a balance between minimizing delay and overall smoothing.  Making them smoother would result in more latency.

haplm, 2018-09-09:
I get that with more filtering, latency would increase, and I'm fine with that. Before, I was using 16 channels on FrSky with manual cutoff set to 25Hz, which is tons of latency. CRSF is much faster than FrSky, so a bit more latency during filtering won't IMO kill anybody (flying freestyle). Would it be perhaps possible to choose "strength" of the auto setting (because manual doesn't make much sense here)?

ctzsnooze, 2018-09-18:
And would you then make it stronger or weaker in your particular case?  And what would you expect stronger or weaker to do?  The code is already automatic and optimised very effectively to the best balance between spiking due to inadequate smoothing and delay from excessive smoothing.

haplm, 2018-09-29:
I have received following from TBS:

> thanks for bringing this to our attention. we have actually looked into this quite some time ago and have not been able to identify a problem. I believe that most of this is related to the lower latency of the crossfire system as well as the higher refresh rate (and the fluctuations between 50Hz and 150Hz).

I believe it doesn't address the issue, so I have replied with this:

> I'm not sure about this. The betaflight guys say that this is caused by inconsistent frame rate generated by the CRSF receiver. That I think is not related related to the high frame rate you are mentioning as a possible cause (or rate changes). High or low frame rate, I would expect that the receiver keeps consistently sending frames unless there's a failsafe condition. If the link between Rx and Tx drops a frame or two, I would expect RX to do some approximation, or simply repeat last received frame (until failsafe kicks in). This is unfortunately not the case (and again, it is not related to the rate switch).

> Please try to read the whole BF report again - there's bunch of logs and screenshots showing what's really going on. Or perhaps ask one of the engineers working on it to read it?

closed 2018-11-05 by stale[bot]