betaflight/betaflight #6684: OMNIBUS target FAILSAFE problem.

vvac, 2018-09-02:
**Describe the bug**
Failsafe is not triggered when BF receives iBus frame shorter than rx_min_usec. Applies OMNIBUS target. Works correctly on Matek F405 and Kakute F4 AIO.

**To Reproduce**
Set rx_min_usec = 950.
Set your iBUS flysky receiver to send 900us on all aux channels as a failsafe.
Observe how Betaflight does not enter failsafe when you turned off your transmitter.

**Expected behavior**
Betaflight should enter failsafe when it gets iBUS frame with invalid value.

**Flight controller configuration**
```
# diff
# Betaflight / OMNIBUS (OMNI) 3.5.0 Aug 14 2018 / 03:49:05 (1d710afd7) MSP API: 1.40

name Omni

feature -RX_PPM
feature RX_SERIAL
feature MOTOR_STOP
beacon RX_LOST
beacon RX_SET
serial 2 64 115200 57600 0 115200
aux 0 0 3 1700 2100 0 0
aux 1 1 2 900 1300 0 0
aux 2 27 1 1700 2100 0 0
aux 3 13 0 1700 2100 0 0
aux 4 19 0 1300 1700 0 0
aux 5 28 2 1700 2100 0 0
set acc_calibration = -70,-65,-388
set rssi_channel = 14
set serialrx_provider = IBUS
set rx_min_usec = 950
set rx_max_usec = 2050
set dshot_idle_value = 700
set motor_pwm_protocol = DSHOT600
set ibata_scale = 179
set small_angle = 30
set osd_rssi_pos = 2477
set osd_tim_2_pos = 2497
set osd_flymode_pos = 2509
set osd_current_pos = 2518
set osd_mah_drawn_pos = 2488
set osd_warnings_pos = 2122
set osd_avg_cell_voltage_pos = 2465
set osd_battery_usage_pos = 457
set osd_disarmed_pos = 2283
profile 0

rateprofile 0
```

**Setup / Versions**
 - Omnibus F3 from Banggood
 - flysky x6b receiver (rssi mod firmware), i6x transmitter (newest availalble firmware)
 - standard wiring, ibus connected to uart3 on omnibus.

**Additional context**
Works as it should on other targets I tested. Same receiver with same RSSI mod works well with Matek F405 and Kakute F4 AIO. 900us value sent in the frame triggered failsafe correctly,

vvac, 2018-09-08:
I performed some further tests and it turned out to have sth to do with throttle. When I send -935us on throttle channel in addition to others it started to work. You might have been right that detection works only on throttle channel. Or more general on AETR channels and not on AUX channels. 

Best Regards,
Tomasz

mikeller, 2018-09-08:
@vvac: The check definitely only covers non-AUX channels:

https://github.com/betaflight/betaflight/blob/39ced6bbfefa52a6f605ed6635b7e62105c71672/src/main/rx/rx.c#L518-L520

closed 2018-10-15 by stale[bot]