betaflight/betaflight #6685: Request/Suggestion:  Betaflight Build Bot

ijustwannafly, 2018-09-02:
With 3.5 released and with the f3 performance edition builds in existence I was wondering if there was ever a thought to create a web based build bot where users could specify their target board (from existing targets) and requirements (ie baro and acc disabled by deafult, only sbus or fport protocol, no acro trainer etc) and have the build bot spit out a hex file with those options.

Just a thought.

wind0r, 2018-09-02:
Yes. People already had the Idea. Similar like OpenTX already does it. 
With BF4 and unified targets this may gets easier for the server since it doesnt need to compile thousands of different versions (on request and cached also saves resources) but still there is a lot to implement

SRWieZ, 2018-09-02:
I may be able to do it. I'm web developer and i know how to compile Betaflight. 

Recently I compiled a custom firmware for my babyhawk R because i needed LED feature. Babyhawk doesn't have the capability to do blackbox logging so that was an easy one. Without blackbox, there is enough space to activate others things. 

I will dig more into it unless a top contributor say it's a bad idea.

mikeller, 2018-09-02:
The big problem here will be finding / financing the resources to run this. So unless we've got a clear plan of where we can host this, and who is going to pay for it then I think investigating this doesn't make much sense.

Also, the problem with this will be that test coverage will obviously be a lot worse if this is implemented, as there are lots of potential for interdependencies between features, and any bug report will have to be investigated for the exact build that the reporter is using, and any fix will potentially introduce regression bugs in other builds.

closed 2018-10-09 by stale[bot]