betaflight/betaflight #6690: Documentation on MSP and SPort-MSP bridge.

vodka-bears, 2018-09-03:
Hi Betaflight folks. I'm trying to code a script for ersky9x to configure VTX from the radio. I couldn't find a reference about MSP_VTX_CONFIG and MSP_SET_VTX_CONFIG because last MSP docs avaliable are of 2016. Aslo there isn't ANY documentation on SPort-MSP bridge. I could've ported it from OpenTX script but i don't know Lua at all. Please someone update docs/development or docs/API.

mikeller, 2018-09-03:
Pretty much the full definition for `MSP_VTX_CONFIG` and `MSP_SET_VTX_CONFIG` is in the firmware source code (sorry there's not more documentation): https://github.com/betaflight/betaflight/blob/e2b91ae0f1693fc216be44710584c6deea4f808e/src/main/interface/msp.c#L1328-L1349
https://github.com/betaflight/betaflight/blob/e2b91ae0f1693fc216be44710584c6deea4f808e/src/main/interface/msp.c#L1965-L2002

A lot of the discussion around the MSP over SmartPort specification can be found in #1401.

vodka-bears, 2018-09-04:
Thanks mikeller, the code is really self-explaining. But you know the it's better to have a documentation rather than code. I hope someone (maybe me) will write some in the future.

mikeller, 2018-09-04:
@vodka-bears: Yes please, we are grateful for contributions of documentation.

closed 2018-10-11 by stale[bot]