betaflight/betaflight #6694: obf4 system crashes at disarm

m013, 2018-09-04:
**Describe the bug**
OMNIBUSF4 (OBF4) 4.0.0 fc crashes at disarm

**To Reproduce**
fw 1075, 1099, a quad doesnt reacts on commands, a beeper beeps continuously, only power cycle helps until the next crash

**Expected behavior**
to be able to re arm

**Flight controller configuration**
'''
diff all

# version
# Betaflight / OMNIBUSF4 (OBF4) 4.0.0 Aug 22 2018 / 12:36:34 (f8b601ee4) MSP API: 1.41

board_name OMNIBUSF4
manufacturer_id AIRB
mcu_id 0028004b3436510730383737
signature 

# reset configuration to default settings
defaults nosave

# name
name F100

# resources
resource MOTOR 1 A03
resource MOTOR 2 A02
resource MOTOR 3 B00
resource MOTOR 4 B01
resource MOTOR 5 NONE
resource LED_STRIP 1 NONE
resource SERIAL_TX 1 NONE
resource SERIAL_TX 11 A09

# mixer

# servo

# servo mix

# feature
feature -RX_PARALLEL_PWM
feature -ANTI_GRAVITY
feature RX_SERIAL
feature SOFTSERIAL

# beeper
beeper -RX_LOST_LANDING
beeper -ARMING_GPS_FIX
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -MULTI_BEEPS
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE

# beacon

# map
map AERT1234

# serial
serial 0 64 115200 57600 0 115200
serial 30 8192 115200 57600 0 19200

# led

# color

# mode_color

# aux
aux 0 0 0 1850 2100 0 0
aux 1 1 1 1375 1625 0 0
aux 2 4 1 1875 2100 0 0
aux 3 13 2 1400 1600 0 0
aux 4 28 1 1900 2100 0 0
aux 5 35 1 900 1075 0 0
aux 6 39 3 1875 2100 0 0

# adjrange
adjrange 0 0 5 900 2100 12 5 0 0

# rxrange
rxrange 0 987 2010
rxrange 1 987 2010
rxrange 2 987 2010
rxrange 3 988 2011

# vtx

# rxfail
rxfail 5 s 1500
rxfail 6 s 1500

# master
set gyro_lowpass_hz = 110
set gyro_lowpass2_hz = 0
set gyro_overflow_detect = OFF
set dyn_fft_location = BEFORE_STATIC_FILTERS
set dyn_filter_range = HIGH
set acc_lpf_hz = 7
set acc_trim_pitch = -8
set acc_trim_roll = -8
set acc_calibration = 29,-47,484
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1010
set max_check = 1998
set rssi_channel = 16
set rc_interp = MANUAL
set rc_interp_int = 15
set serialrx_provider = SBUS
set rx_min_usec = 900
set rx_max_usec = 2100
set min_throttle = 1010
set dshot_idle_value = 400
set motor_pwm_protocol = DSHOT600
set failsafe_delay = 15
set failsafe_off_delay = 30
set failsafe_throttle = 1150
set failsafe_throttle_low_delay = 150
set failsafe_procedure = AUTO-LAND
set align_board_roll = 180
set align_board_yaw = -90
set vbat_min_cell_voltage = 32
set vbat_warning_cell_voltage = 34
set current_meter = NONE
set vbat_scale = 111
set yaw_motors_reversed = ON
set small_angle = 180
set deadband = 2
set yaw_deadband = 2
set pid_process_denom = 2
set tlm_halfduplex = OFF
set osd_cap_alarm = 200
set osd_vbat_pos = 2403
set osd_rssi_pos = 2423
set osd_tim_1_pos = 54
set osd_tim_2_pos = 2435
set osd_flymode_pos = 407
set osd_throttle_pos = 225
set osd_vtx_channel_pos = 377
set osd_ah_pos = 200
set osd_current_pos = 385
set osd_mah_drawn_pos = 353
set osd_craft_name_pos = 362
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 329
set osd_avg_cell_voltage_pos = 2454
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 392
set osd_disarmed_pos = 2412
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set displayport_max7456_wht = 1

# profile
profile 0

set dterm_notch_cutoff = 0
set anti_gravity_gain = 1500
set crash_time = 650
set crash_delay = 15
set crash_recovery = ON
set throttle_boost = 25
set p_pitch = 65
set i_pitch = 65
set d_pitch = 31
set p_roll = 60
set i_roll = 60
set d_roll = 31
set d_yaw = 25
set p_level = 25
set d_level = 50

# profile
profile 1

set dterm_notch_cutoff = 0
set anti_gravity_gain = 1500
set crash_time = 650
set crash_delay = 15
set crash_recovery = ON
set p_pitch = 55
set i_pitch = 60
set d_pitch = 28
set p_roll = 50
set i_roll = 55
set d_roll = 27
set p_yaw = 75
set d_yaw = 25
set p_level = 25
set d_level = 50

# profile
profile 2

set dterm_lowpass_hz = 120
set dterm_lowpass2_hz = 230
set dterm_notch_cutoff = 0
set anti_gravity_gain = 2000
set iterm_rotation = OFF
set iterm_relax = RPY
set throttle_boost = 27
set p_pitch = 56
set i_pitch = 61
set f_pitch = 122
set p_roll = 56
set i_roll = 56
set d_roll = 24
set f_roll = 117
set p_yaw = 85
set i_yaw = 120
set f_yaw = 112

# restore original profile selection
profile 2

# rateprofile
rateprofile 0

set roll_rc_rate = 125
set pitch_rc_rate = 125
set yaw_rc_rate = 126
set roll_srate = 75
set pitch_srate = 75
set yaw_srate = 75
set tpa_rate = 0
set tpa_breakpoint = 1050

# rateprofile
rateprofile 1

set thr_mid = 0
set roll_rc_rate = 142
set pitch_rc_rate = 142
set yaw_rc_rate = 141
set roll_expo = 8
set pitch_expo = 8
set yaw_expo = 8
set roll_srate = 62
set pitch_srate = 62
set yaw_srate = 62
set tpa_rate = 0
set tpa_breakpoint = 1550

# rateprofile
rateprofile 2

set tpa_rate = 0

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 1

# save configuration
save
# 
'''

m013, 2018-09-04:
For some reason i decided to reupload fonts and it helped. Never had this issues with stable releawes and rcs

jflyper, 2018-09-05:
@m013 What happens if you re-enable the OSD feature with re-uploaded font?

m013, 2018-09-05:
Re enablling the feature only didnt help as far as i remember.
I have another quad with another fc but with the same target. Did it immediately (uploading fonts, without osd disable enable) after flashing 1099 and saw no issues

closed 2018-10-12 by stale[bot]