betaflight/betaflight #6695: GPS as a MODE that can be put on a switch

BludzUK, 2018-09-04:
I would like to unplug my GPS module and fly closer range. Even with setting min satelites set to zero, I'll still get all the GPS info on my OSD, this being changed to a mode, am I right in saying wouldn't be displayed unless it's activated?

McGiverGim, 2018-09-05:
The solution for this is different osd in a switch, like iNav has. It will serve for GPS and other situations.

BludzUK, 2018-09-05:
I was thinking more about this and had the same idea. A set of OSD elements in a group that can be turned off and on with a switch. Just adding to the feature that turns them all off.

pkruger, 2018-09-07:
There is a feature request for this, i.e. OSD pages or profiles, see issue #4155.  I have a version of this working, happy to give you a hex file if you want to try it.  It's not reviewed yet and not part of the official Betaflight -yet.

mikeller, 2018-09-08:
Adding a switch for this sounds like overkill, and bad usability - we should just change the code so that the GPS coordinates are only displayed in OSD after the GPS unit has been detected.

BludzUK, 2018-09-08:
Thanks @pkruger but I think @mikeller is correct. After further testing and more thought on my particular issue, setting min sats to zero is not a good idea. So the plan to just ignore and hide all the GPS OSD stuff isn't really ideal. Surely must be wasting CPU load too.
If the firmware detected the GPS is not connected it could just never display any GPS OSD elements, but most importantly it should let you arm still and ignore min sats for rescue mode. Maybe could display something saying the GPS not detected, for those who always run GPS and have a broken connection rather than disconnected the GPS.

mikeller, 2018-09-08:
@BludzUK: That's kind of a side effect of my suggestion: If you want GPS to be connected, and if the GPS info is not shown in the OSD, you know that something is broken.

BludzUK, 2018-09-08:
Ah yes, of course. Good call Mike. That would work well for both users. Currently nothing displays until you get a lock or it times out and shows all zeros (maybe 1 minute wait). So some sort of way to see it isn't connected might be preffered otherwise you will wait for ages to get a lock and nothing ever shows up, before you know it isn't connected/detected.

mikeller, 2018-10-08:
@BludzUK: Can you check this again? Looking at the code for 3.5, it should already be the case that the GPS information isn't displayed in the OSD if no data from a GPS unit is received:

https://github.com/betaflight/betaflight/blob/8283aba668d8ec032317c9424c3be936dbe12734/src/main/io/osd.c#L1072-L1079

BludzUK, 2018-10-20:
@mikeller Sorry for the delay in checking this, but I had to remove GPS completely due to the issue. Checked it today. When you unplug the GPS module so it can't detect it, power it up, and can't arm. Warning says 'GPS' on the osd when you try to arm. Obviously I can set GPS min sats to zero but then when I use gps I have no min sats, which I want for rescue mode. The GPS related OSD elements are not displayed but you can't arm, which is the issue.
So the code can clearly detect the sensor, we now need a way to allow arming when it can't be detected. Currently it is still carrying out sanity checks for gps when the sensor is disconnected. Perhaps you would want this to remain but I think min sats is enough and sanity should be ignored when no gps sensor is detected.

mikeller, 2018-10-20:
@BludzUK: I think this is how it should be really. If GPS rescue and the sanity checks are enabled, then checking for the presence of a GPS device should be performed - otherwise the case where the GPS isn't found because of a wiring fault will be missed, which is counter-intuitive to the idea of sanity checking.

BludzUK, 2018-10-20:
@mikeller If there is no GPS then you won't ever get a satalite, so why would you need both checks? If it needs to stay as a originally thought it would, then could we get a way to switch it off via tx, so we can still fly quads without having to do it in the configurator. I don't always need gps and sometimes can't get satalites, so unplugging the sensor and being able to fly is my hope.

mikeller, 2018-10-21:
@BludzUK: To keep you from flying away, trusting that GPS rescue will work, when in fact no GPS data is received because the wiring is faulty.

BludzUK, 2018-10-21:
If wiring is faulty and no data received, then you wouldn't have satalites and rescue mode would fail anyway, I presume. Perhaps if it's an intermitant connection fault you could lock onto sats and then gps disconnects. No idea what the code does then, whether it immediately drops all stored sats or whether it tries to keep connecting to them. I can't think of a situation where you would have sats locked (pass sanity) but gps isn't connected (fail sanity).

Is there a way to have sanity checks just on rescue mode and not on arming? Or at least just remove gps sensor detection from the arming sanity checks and leave it in the rescue mode checks? So when gps is disconnected (or faulty), rescue mode will fail as it should but you can still arm it. That way works for both. Disconnect gps and not use rescue mode. If somehow you get sats locked before arming with a faulty gps (or if it fails while up in the air), then when you try to use rescue mode it will still fail. 

Wouldn't the gps osd elements also disappear making it clear that you no longer have gps? Or does that only detect on startup?

mikeller, 2018-10-23:
@BludzUK: I don't know what you think the GPS rescue sanity checks should be doing if you think they should not prevent arming when a condition exists that makes it impossible for GPS rescue to work? If you don't care if GPS rescue will work or not then just disable the sanity checks.

BludzUK, 2018-10-23:
Sorry if my comments are confusing, let me try and simplify.

My goal - Have a quad with a GPS installed so I can go long range and use rescue mode. When not going long range and not needing the GPS, I simply want to remove it and fly without rescue mode.

The Issue - When you remove the GPS sensor you can't arm the quad. Only way to fly the quad with the GPS sensor disconnected is to use the configurator to disable GPS.

Solution - Have a way to remove the GPS sensor and still fly, without having to use the configurator.

Please just disregard all my suggestions, I was just trying to help acheive a solution. I really don't mind how you can get this to work. To me it still sounds like we need a switch or osd menu item to disable gps when we don't want to use it, as just disconnecting it will always trigger code that disables arming which we want to keep.

mikeller, 2018-10-23:
@BludzUK: This is easy. Just set `gps_rescue_min_sats` to 0 and arming without GPS or GPS lock will work again.

BludzUK, 2018-10-23:
Yes, as I've said above this is a solution but then renders rescue mode useless as it could fly away with 0 sats.

mikeller, 2018-10-23:
@BludzUK: Well yes, you apparently don't care if GPS rescue works or not, otherwise why would you want it to arm if there is no GPS detected. You can't have your cake and eat it too.

BludzUK, 2018-10-23:
Yea. I don't want rescue mode to work when I disconnect the gps, but I do want it to work when I have gps connected. A switch would solve this and we can all share cake together :P

closed 2018-11-29 by stale[bot]

peteoz, 2019-03-17:
I understand exactly what @BludzUK is asking for, unfortunately it seems he didn't explain his request very well and this whole thread looks like misunderstanding.

he is basically asking for same request as #6841

scenario - you drive 50km from home to a remote location, or you hike for 4 hours to remote location only to find out your GPS is not working (various reasons - can't get sat lock, GPS unit faulty, GPS connector broken, cable to GPS cut after first battery,...), but you still want to be able to arm the quad and take off.
you have no laptop with you (maybe because you only own a desktop, or didn't want to drag heavy laptop with you on a 4 hour hike).

there should be a way to allow the pilot to make a conscious decision to fly without GPS.

this would also be handy in other scenarios - going to fly at local football field where you don't need GPS (and don't want to wait for sat lock before each flight), or just an easy way to test/spin motors at home where you can't get sat lock. 

if FW is already able to detect there is no GPS connected, wouldn't the easiest solution be to just unplug the GPS from my quad and have the FW to take it as a sign that my intention is to fly with no GPS ?

BludzUK, 2019-03-17:
Thanks @peteoz, it's a pretty simple request but as the FC can't tell if you disconnected the GPS on purpose or if you haven't built your quad right/it has developed a fault, using the auto detect will conflict with the sanity checks. GPS as a mode would solve this I think. Like any other flight mode. Not like we have to fly angle mode when it detects the accelerometer.

BludzUK, 2019-03-17:
I think they actually solved this issue, but haven't test.
https://github.com/betaflight/betaflight/commit/007e14f348e41b7b957f019eb03b91640b3d363f

peteoz, 2019-03-17:
@BludzUK 
> FC can't tell if you disconnected the GPS on purpose or if you haven't built your quad right/it has developed a fault

I'm aware of this possibility as @mikeller mentioned above a case of wiring fault, but I run sanity checks OFF and still cannot arm when my GPS is removed from the quad.

for me personally it's a non-issue for the FC not knowing whether my GPS was disconnected due to wiring fault or whether it was my intention to remove the GPS unit. 

if I definitely want to fly with GPS and know my GPS is connected and yet I see 0 satellites on my OSD or GPS coordinates sitting at 0.00000 then I can decide not to fly, I don't need FC/FW to make that decision for me (remember sanity checks OFF)
- this is no different to taking off with discharged battery, or leaving accidentally my crossfire on 10mW for long range flight. we do pre-flights check and likewise I do check my sat count before taking off.