betaflight/betaflight #6704: Problem connecting PPM receiver to Omnibus F4 V3

PackyFPV, 2018-09-06:
Hi...
I'm trying to connect a PPM receiver to an Omnibus F4 V3 FC flashed with Betaflight 3.4.0 without any results; in fact in the receiver tab no channel moves when i move the sticks;
First of all, i can say that the binding procedure wad done correctly (tested in PWM) and so i think that the problem is the connection with the board.
I soldered the PPM wire to the J7 connector and i moved the solder bridges from sbus to ppm, chosen the UART6 and selected the PPM Rx input in beta flight.
I tried also to move the PPM wire to RX1 (UART1) which is not inverted (in case the RX is of that type) but the result is the same.

Do you know if i missed something or if i make any errors?
Thank you very much for your help!!! 
 
[image]

etracer65, 2018-09-06:
PPM connections don't use UARTs and there's nothing to set up on the Ports tab.  You should not enable the "Serial Rx" toggle on any port.  Connect to the dedicated PPM pin on the flight controller and be sure to select "PPM RX Input" under Receiver on the Configuration tab.

PackyFPV, 2018-09-06:
First of all thank you for the answer!!!
I tried what you suggested but the result is always the same... I disabled every toggle on every port, and selected PPM RX Input in the Configuration tab... But the channel are still fixed!!!

etracer65, 2018-09-06:
Well are you sure you're receiver works and actually outputs PPM?  Beyond that I don't have any more suggestions.  Possibly you might get more help with your specific hardware here:

https://www.rcgroups.com/forums/showthread.php?2464844-Betaflight-Flight-Controller-Firmware-Discussion-Thread

Or possibly from the manufacturer of the flight controller.

PackyFPV, 2018-09-06:
To be honest i'm using an old "recycled" RX (F801) since i would like to use it to connect my radio with the simulator wireless.
I tested the RX in PWM and it works normally! On the RX is written PPM OUT but i don't know if it signal is brought correctly... do you know if there is some test i can perform to verify it?

etracer65, 2018-09-06:
Not without using a scope or logic analyzer.  A quick googling shows that you need to jumper channels 1 & 3 to enable PPM mode which comes out  channel 4.  Did you do that?

jflyper, 2018-09-07:
@PackyFPV 

I wouldn't try a receiver with no guarantee about its correct operation, without proper measurement tools.

Also, there's something wrong with your configuration; you are missing UART1 from the ports tab. I recommend that you do CLI "defaults" and start configuring only the PPM.

jflyper, 2018-09-07:
Since this is clearly a configuration issue, I'm closing the ticket.

PackyFPV, 2018-09-07:
Please wait!!! What do you mean with "you are missing UART1 from the ports tab"? On my fc, when i used a sbus receiver, i connect it to the J7 pins which correspond to the UART6. In the first reply sent me by etracer65, he suggest me to disable all the UART... so i don't understand why are you saying me that UART1 is missing.
In any case, i don't need a really good reliability of my rx, since i'm using it only at home with the simulator; anyway, it was gifted me by a friend and when he used it, it worked normally!!!

@etracer65 
Yes i put a jumper between CH1 and CH3 signal pins and i took the PPM output from the CH4!!!

jflyper, 2018-09-07:
@PackyFPV 

You can goto RCG's Omnibus F4 AIO thread.

closed 2018-09-07 by jflyper

mahmad558, 2019-04-03:
You should solder the jumper of sbus or ppm

mahmad558, 2019-04-03:
https://www.dronetrest.com/t/how-to-identify-what-omnibus-f4-board-you-have/4735