betaflight/betaflight #6705: BF 4.0.0 build 1103 attitude bug

Ralfde, 2018-09-06:
**Describe the bug**
When I fly in the attitude mode and give full roll or nick, when I release the stick to zero position, the copter rolls back in two steps. 
It looks like it stops first at about 20degree and then again at 0 degree.
I don't have this with BF 3.5.0

**To Reproduce**
Flying in attitude mode and release the roll or nick stick from full travel to zero position.

**Flight controller configuration**
Omnibus F4SD

If you need a diff, please let me know.

etracer65, 2018-09-06:
Yes, as requested when you created the issue please provide a `diff` of your settings.  Also a blackbox log and video would be helpful.

What do you mean by "attitude mode"?  Are you referring to angle mode?

Also, please try build 1101 as I suspect changes that were included in 1102 might be related.

Ralfde, 2018-09-06:
Yes, the angle mode.

etracer65, 2018-09-06:
Be sure to do a full chip erase (or do `defaults` in the CLI) before putting your settings back in.  When trying different builds of the development branch you need to always do this as each build may introduce new/changed parameters and not resetting your setting can result in a corrupted configuration and unpredictable results.  For example if you were using an earlier 4.0 build and then upgraded to 1103 without a reset then that could cause a problem.  So if 1101 is okay please confirm that 1102 (or later) has the problem but be sure to reset your settings (1102 introduced settings changes for example).

Ralfde, 2018-09-07:
DIFF:   (flight tests coming later)
# version
# Betaflight / OMNIBUSF4SD (OBSD) 4.0.0 Sep 4 2018 / 22:40:33 (121789367) MSP API: 1.41

board_name OMNIBUSF4SD
manufacturer_id 

# name

# resources
resource MOTOR 1 A03
resource MOTOR 3 B00
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource SONAR_TRIGGER 1 NONE
resource SONAR_ECHO 1 NONE
resource SERIAL_TX 11 A08
resource SERIAL_RX 11 A01

# mixer

# servo

# servo mix

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature SOFTSERIAL
feature GPS
feature TELEMETRY
feature LED_STRIP
feature ESC_SENSOR

# beeper

# beacon

# map
map TAER1234

# serial
serial 0 2 115200 57600 0 115200
serial 2 1024 115200 57600 0 115200
serial 5 64 115200 57600 0 115200
serial 30 8 115200 57600 0 115200

# led
led 0 0,0::C:2
led 1 1,0::C:2
led 2 2,0::C:2
led 3 3,0::C:2
led 4 4,0::C:2
led 5 5,0::CV:2

# color

# mode_color

# aux
aux 0 0 1 1300 2100 0 0
aux 1 1 0 1000 1700 0 0
aux 2 46 2 1000 1200 0 0
aux 3 26 3 1800 2000 0 0
aux 4 28 1 1300 2100 0 0
aux 5 30 3 1000 1200 0 0
aux 6 35 1 1950 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_notch2_hz = 200
set gyro_notch2_cutoff = 100
set mag_declination = 2420
set serialrx_provider = SUMD
set blackbox_p_ratio = 128
set motor_pwm_protocol = DSHOT1200
set failsafe_throttle = 1100
set failsafe_procedure = GPS-RESCUE
set current_meter = ESC
set vbat_scale = 111
set yaw_motors_reversed = ON
set small_angle = 180
set gps_provider = UBLOX
set gps_rescue_angle = 20
set gps_rescue_initial_alt = 20
set gps_rescue_descent_dist = 30
set gps_rescue_ground_speed = 1000
set gps_rescue_throttle_max = 1500
set gps_rescue_sanity_checks = RESCUE_SANITY_OFF
set deadband = 10
set yaw_deadband = 10
set pid_process_denom = 2
set tlm_halfduplex = OFF
set osd_vbat_pos = 2102
set osd_rssi_pos = 2137
set osd_vtx_channel_pos = 2465
set osd_crosshairs_pos = 2253
set osd_ah_sbar_pos = 2254
set osd_ah_pos = 2126
set osd_current_pos = 2094
set osd_mah_drawn_pos = 2089
set osd_gps_speed_pos = 2273
set osd_gps_sats_pos = 2241
set osd_home_dir_pos = 2286
set osd_home_dist_pos = 2306
set osd_altitude_pos = 2262
set osd_pidrate_profile_pos = 2435

# profile
profile 1

set dterm_notch_hz = 260
set vbat_pid_gain = ON
set iterm_rotation = OFF
set iterm_relax = OFF
set iterm_relax_type = GYRO
set yaw_lowpass_hz = 100
set throttle_boost = 0
set acro_trainer_angle_limit = 40
set p_pitch = 45
set d_pitch = 35
set f_pitch = 100
set p_roll = 40
set d_roll = 30
set f_roll = 100
set p_yaw = 45
set d_yaw = 20
set f_yaw = 100
set abs_control_gain = 20

# rateprofile
rateprofile 0

set roll_rc_rate = 106
set pitch_rc_rate = 106
set roll_srate = 76
set pitch_srate = 76
set tpa_rate = 50
set tpa_breakpoint = 1050

Ralfde, 2018-09-07:
Build 1101 does have the same problem, but the strange thing is, not always. Log00004 should show the effect all the time. Log00003 does not have the effect at the beginning and later on in the log the effect starts.

[LOGS.zip](https://github.com/betaflight/betaflight/files/2359710/LOGS.zip)

Ralfde, 2018-09-07:
Build 1072 doesn't have this problem, but I think the max. angle limit is not correct in this version. The angle gets more during flying forward and backward.

etracer65, 2018-09-08:
Your description of the cause of the apparent "two step" return to level is probably correct.

While in flight the attitude of the axes is determined by integrating both the gyro and the accelerometer information by the IMU.  It's not possible to use only the accelerometer because it measures acceleration - not just gravity.  As the quad flies it feels acceleration and this would confuse the "level" calculations.

So a common cause of the level attitude drifting is related to noise (vibration) affecting both the gyros and accelerometer.  So it's not unusual to experience more "drift" when flying rapidly or using higher throttle.  Ways you wan reduce this mainly center around reducing vibrations seen by the flight controller - soft mounting, making sure frame bolts are tight, etc.

Once the gyro activity settles down the IMU that integrates the sensor data begins to converge back to level based on the accelerometer data - correcting the error.

So I'm not sure there's any bug here.

Ralfde, 2018-09-08:
See DVR. In real the two stops are seen more clearly.
I also added a Video with a yaw problem. Really strange.

[Two_Stops_angle_mode.zip](https://github.com/betaflight/betaflight/files/2363441/Two_Stops_angle_mode.zip)
[Yaw_oscillation.zip](https://github.com/betaflight/betaflight/files/2363446/Yaw_oscillation.zip)

Ralfde, 2018-09-08:
Logging to Yaw_problem. At a fpv flight I had the impression also that something like this is also at roll and nick.

[LOG00012.zip](https://github.com/betaflight/betaflight/files/2363447/LOG00012.zip)

Ralfde, 2018-09-08:
Another thing: The HoTT telemetry does normally work, but it can happen that the OSD setting fails after I made settings with the HoTT telemetry. When I try to enter the OSD setting mode then only some **** occur on the screen instead of the OSD setting menu. Maybe this could be done more stable.

Ralfde, 2018-09-08:
Your description of the cause of the apparent "two step" return to level is probably correct.

While in flight the attitude of the axes is determined by integrating both the gyro and the accelerometer information by the IMU. It's not possible to use only the accelerometer because it measures acceleration - not just gravity. As the quad flies it feels acceleration and this would confuse the "level" calculations.

So a common cause of the level attitude drifting is related to noise (vibration) affecting both the gyros and accelerometer. So it's not unusual to experience more "drift" when flying rapidly or using higher throttle. Ways you wan reduce this mainly center around reducing vibrations seen by the flight controller - soft mounting, making sure frame bolts are tight, etc.

Once the gyro activity settles down the IMU that integrates the sensor data begins to converge back to level based on the accelerometer data - correcting the error.

So I'm not sure there's any bug here.

Ralf:
Maybe it's not a bug, but it's not with inav or older BF firmware 3.2 in the same copter. But maybe this is not so important for a Racequad or Freestyle quad.

Ralfde, 2018-09-17:
I found a reason for this problem in the attitude mode:
I use the ICM-20601 gyro. I think the accel scaling is not correct. When I calibrate the accel, Z-Axle shows correct 1g. But when I turn the copter 90 degree, it shows 0.5g. I think the accel scale need to be corrected for this gyro.

The yaw problem you can see in the video is also still there in the latest build 1120 from today. Please have also a look into this.

etracer65, 2018-09-17:
What board is this and where did you buy it?  I'm not aware of any OMNIBUSF4SD targets using the ICM20601 gyro.  Who manufactured this board?

Please provide the output of the `status` command and also the `gyroregisters` command.

If you have a separate issue related to yaw please open a different issue.  It should not be combined under this unrelated issue.

Ralfde, 2018-09-17:
Omnibus F7NXT. I did get a gyro box with this gyro for testing.
Sorry, I looks like I did something wrong before. Accels show this time 1g.
# status
System Uptime: 57 seconds
Current Time: 0000-01-01T00:00:00.000+00:00
Voltage: 0 * 0.1V (0S battery - NOT PRESENT)
CPU Clock=216MHz, Vref=3.31V, Core temp=54degC, GYRO=ICM20601, ACC=ICM20601, BARO=LPS
Stack size: 2048, Stack address: 0x20010000
I2C Errors: 1, config size: 2666, max available config: 16384
CPU:27%, cycle time: 34, GYRO rate: 29411, RX rate: 33, System rate: 9
Arming disable flags: RXLOSS CLI MSP GPS

# Gyro 2
# WHO_AM_I    0xAC
# CONFIG      0x0
# GYRO_CONFIG 0x12

jflyper, 2018-09-17:
Flash target `OMNINXT7` for OmniNXT F7 board.

I will close this issue ticket because it is just a random memorandum of multiple issues spanning multiple boards, which is too confused for others and may be for the OP.

Issues also sounds like configuration problems to me. Please consider using other means of support as instructed in the original pre-formatted text in issue templates.

closed 2018-09-17 by jflyper