betaflight/betaflight #6709: help me please. When i go to the receiver tab on betaflight, nothing shows up on yaw, throttle pitch etc. how can i get stuff to show up

akdsflkasjdf, 2018-09-07:
# If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack (registration at https://slack.betaflight.com/) or other user support forums & groups (e.g. facebook).

## If you believe there is an issue with the firmware itself please follow these steps:
1. Describe your problem;
2. Include ways to reproduce the problem;
3. Provide information about your flightcontroller and other components including how they are connected/wired;
4. Add the used configuration and firmware version;
5. Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/); and
6. Remove this Text :).

akdsflkasjdf, 2018-09-07:
Also it is a xsrf4po

jflyper, 2018-09-08:
Please read the text immediately below the title. Thank you.

closed 2018-09-08 by jflyper