betaflight/betaflight #6716: Add alert by color and blinking

Khaossa, 2018-09-09:
Kt will be really awasome if OSD alert or warning lkke a voltage or current can change the collor depending the value of it an start blinkin when is  ear puted limit. Like some cams do now.

etracer65, 2018-09-09:
The chip that generates the OSD image can only produce levels of gray images.  It cannot produce color overlays.

Khaossa, 2018-09-09:
And can't make the font be bigger when value arrive to limit or it blink?

etracer65, 2018-09-09:
The OSD overlay chip can only store one font at a time so it's not possible to increase the font size dynamically.

There is already a battery capacity alarm.  Just set the Capacity value under alarms to be when you want to be notified.  When the consumption exceeds this, the consumed current, remaining percentage and estimated remaining time will start blinking.

Likewise if the battery is below the WARNING or CRITICAL limits then the battery voltage indicators will start to blink.