betaflight/betaflight #6717: vbat_pid_gain of configurator change also dterm_notch_cutoff

nerilorenzo, 2018-09-09:
**Describe the bug**
by using the interface of the configurator I have selected 
vbat_pid_gain = ON
then using diff I noticed that it where inserted not one but two commands:
set dterm_notch_cutoff = 0
set vbat_pid_gain = ON
while default value for the dterm_notch_cutoff is 160
I notice also that if using the interface I set the vbat_pid_gain to on, then save, then set again to off and save, the dterm_notch_cutoff do not came back to 160 but stay to 0
setting vbat_pid_gain = ON by using the CLI the additional command is not added

**To Reproduce**
I found this behavior also doing it as the first command after flashing and reset 

**Expected behavior**
Only vbat_pid_gain need to be changed by using the button of the interface

**Flight controller configuration**
'''
Entering CLI Mode, type 'exit' to return, or 'help'

# diff

# version
# Betaflight / KAKUTEF7 (KTF7) 4.0.0 Sep  8 2018 / 06:24:00 (39ced6bbf) MSP API: 1.41

board_name KAKUTEF7
manufacturer_id 

# name

# resources

# mixer

# servo

# servo mix

# feature

# beeper

# beacon

# map

# serial

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# vtx

# rxfail

# master

# profile
profile 0

set dterm_notch_cutoff = 0
set vbat_pid_gain = ON

# rateprofile
rateprofile 0

'''

**Setup / Versions**
 - Kakute F7
 - configurator 10.4.0
- betaflight 4.0.0 #1107 and #1075 same behavior 

**Additional context**
Thanks to all developer for the work you are doing

etracer65, 2018-09-09:
I can replicate the issue, but it's not a serious problem.  Basically the `dterm_notch_cutoff` is getting set to 0 if the dterm notch is already disabled - which it is because `dterm_notch_hz` is set to 0.  So the value changing to 0 doesn't actually affect anything.  If the dterm notch really is enabled (both settings are non-zero) then changing `vbat_pid_gain` doesn't change any dterm notch settings.

So we'll look into correcting this, but it's purely cosmetic.

spatzengr, 2018-09-09:
To back up @etracer65 conclusion: https://www.screencast.com/t/bdiFlgWZ

mikeller, 2018-09-10:
This is really an artefact of how the configurator is handling notch filters being disabled, by setting both frequencies of the filter to 0. Not sure if changing the configurator to reflect the original values when changing would be better or not.

etracer65, 2018-09-10:
I think the Configurator behavior is fine, but instead the default for `dterm_notch_cutoff` is the problem.  There's no reason for it to still be set to 160 while the dterm notch is disabled by default with `dterm_notch_hz = 0`.  I was planning on just fixing the default and that will resolve the behavior.

closed 2018-09-10 by borisbstyle

mikeller, 2018-09-10:
@etracer65: This will still give users artefacts of things changing that they did not change if they disable Dterm notch by setting the frequency to 0 in CLI, and then hitting 'Save' on the PID tuning tab...

PR #6731 (2018-09-10): Change dterm_notch_cutoff default to 0
Fixes #6717 

The dterm notch is disabled and `dterm_notch_hz` is already 0 by default. When the Configurator saves changes from the PID Tuning tab and the notch is disabled it sets both `dterm_notch_hz` and `dterm_notch_cutoff` to 0.  Since the default was left at 160, this makes `dterm_notch_cutoff` appear in `diff` outputs.  Ultimately it's just cosmetic since the dterm notch is disabled if either value is zero.  But there's no reason for the `dterm_notch_cutoff` to remain set to the legacy 160 value.

This issue exists in 3.5 as well but I'm not sure  [... cut]