betaflight/betaflight #6719: GPS RESCUE PROBLEM

GARYANTHONYMCDONOGH, 2018-09-09:
QUAD FALLS OUT SKY WHEN ACTIVATING GPS RESCUE MODE. MOTORS DISARM AND QUAD FALLS
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Flight controller configuration**
''' diff

# version
# Betaflight / DALRCF405 (DLF4) 3.5.1 Sep  8 2018 / 05:07:41 (d9fb5ca13) MSP API: 1.40

board_name DALRCF405
manufacturer_id 

# name

# resources
resource SERVO 1 A05
resource CAMERA_CONTROL 1 NONE

# mixer

# servo
servo 0 1000 2000 1500 100 6

# servo mix

# feature
feature SERVO_TILT
feature GPS
feature RSSI_ADC
feature OSD
feature AIRMODE

# beeper

# beacon
beacon RX_SET

# map

# serial
serial 0 64 115200 57600 0 115200
serial 2 2 115200 57600 0 115200
serial 3 4 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 1 1775 2100 0 0
aux 1 1 0 900 1175 0 0
aux 2 46 3 1875 2100 0 0
aux 3 13 0 1850 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail
rxfail 7 s 2000

# master
set acc_trim_pitch = -2
set acc_calibration = -61,21,195
set mag_hardware = NONE
set fpv_mix_degrees = 25
set motor_pwm_protocol = DSHOT300
set failsafe_throttle = 1200
set failsafe_switch_mode = STAGE2
set failsafe_procedure = AUTO-LAND
set yaw_motors_reversed = ON
set gps_provider = UBLOX
set gps_rescue_initial_alt = 35
set osd_warn_arming_disable = OFF
set osd_warn_visual_beeper = OFF
set osd_warn_crash_flip = OFF
set osd_warn_core_temp = OFF
set osd_vbat_pos = 2116
set osd_rssi_pos = 2131
set osd_tim_1_pos = 2476
set osd_flymode_pos = 2455
set osd_anti_gravity_pos = 277
set osd_gps_speed_pos = 2291
set osd_gps_sats_pos = 2123
set osd_home_dir_pos = 2288
set osd_home_dist_pos = 2280
set osd_altitude_pos = 2402
set osd_warnings_pos = 2379
set osd_stat_tim_2 = OFF
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF

# profile
profile 0

set dterm_notch_cutoff = 0
set anti_gravity_gain = 7000
set smart_feedforward = ON
set iterm_relax = RP
set i_pitch = 55
set d_pitch = 45
set p_roll = 50
set i_roll = 55
set d_roll = 45
set p_yaw = 75
set i_yaw = 55
set p_level = 40
set i_level = 40
set level_limit = 85

# rateprofile
rateprofile 0

set roll_srate = 60
set pitch_srate = 60
set yaw_srate = 60
set tpa_rate = 20
set tpa_breakpoint = 1550'''

**Setup / Versions**
 - Flight controller dalrcf405 aio, from dalrc
 - Other components r9 slim rx
 

**Additional context**
other users in my area have experienced the same problem.

GARYANTHONYMCDONOGH, 2018-09-09:
Has anyone else in the world experienced this issues with the motors disarming when GPS rescue mode is activated?

AirBreak69, 2018-09-09:
@GARYANTHONYMCDONOGH I compared your diff to myworking set and found at least two bad settings in your config:
1. failsafe_switch_mode = STAGE2 and failsafe_off_delay = 10 (default: 1s)
This is the reason why the motor are switched off 1s after failsafe condition. failsafe_off_delay should be raised to some seconds (5s ..10s) if the landing should be softer using the failsafe_throttle.
Note: failsafe_off_delay is only used during AUTO-LAND operation.
2. set failsafe_procedure = GPS-RESCUE in CLI is needed to activate GPS-RESCUE. The configurator currently does not support GPS-RESCUE and resets failsafe_procedure = AUTO-LAND on each save. So you have to restore failsafe_procedure = GPS-RESCUE manually after any change in the failsafe tab.

jlandegent, 2018-09-09:
i think it is 's problem of sanity checks. the quad goes into failsave if not the required parameters are met for GPS rescue mode. i had this also twice. you can disable sanity checks

https://github.com/betaflight/betaflight/wiki/GPS-rescue-mode

GARYANTHONYMCDONOGH, 2018-09-10:
Thank you.
Will test and report feedback

crow187, 2018-09-16:
I have the same issue. GPS fix okay, flying away, activate rescue mode by switch, falls out of the sky.

I will deactivate sanity checks, too. I will report back next days...

crow187, 2018-10-08:
Disabling the sanity check does remove the unwanted disarm for me.

dridsfpv, 2018-10-17:
I had the same problem. Fixed it with:
set gps_rescue_sanity_checks = RESCUE_SANITY_OFF
or
set gps_rescue_sanity_checks = RESCUE_SANITY_FS_ONLY

TonyBlit, 2018-11-06:
The unexpected disarm just on activation of GPS Rescue was fixed here: https://github.com/betaflight/betaflight/pull/7034

mikeller, 2018-11-06:
@GARYANTHONYMCDONOGH: Can you check if #7034 fixes your issue, and if so please close this?

ChilledoutUK, 2019-02-04:
Honestly sanity checks should be only on FS as default otherwise it can cause a quad with good radio link to tank into the ground for no reason.

Is there any chance of finding out what these checks are exactly?

rcjarvis, 2019-02-18:
I had the same bug, I have done the following:
set failsafe_procedure = GPS-RESCUE
and:
set gps_rescue_sanity_checks = RESCUE_FAILSAFE_ONLY.. I wil report back hopefully tomorrow!

rcjarvis, 2019-03-21:
Works fine now !

ChilledoutUK, 2019-03-22:
Sanity checks need to be broken into individual items that can be selected by the individual and thus would also allow the selection of the most important ones.
This would also make this less of a mystery as to what checks are being done.

TonyBlit, 2019-03-25:
@mikeller I'd close this issue, it seems resolved
@ChilledoutUK I don't think it's a good idea to selectively activate/deactivate sanity checks, but you can open a feature request with a detailed use case. This is making me think that I'll document what sanity check does exactly in the wiki.

TonyBlit, 2019-04-25:
I've updated the wiki with the detailed explanation of the sanity checks:
https://github.com/betaflight/betaflight/wiki/GPS-Rescue-Mode#dealing-with-failures--sanity-checks-very-important
Hope the mystery is over :-)

closed 2019-06-01 by stale[bot]

aLDime, 2021-06-29:
When activated with a switch Rescue or Failsafe everything works as it should! But when the signal is lost (turning off the RC), instead of the rescue procedure, drop and crash:
[Diff all](https://drive.google.com/file/d/1z9zEpYUtJi797PHwjwWoRw-fnRIKSsJx/view?usp=sharing)
[BLACKBOX_RTH](https://drive.google.com/file/d/15uTxkQBSwf4Tz91JaN5XNSRFJ8dPn9gq/view?usp=sharing)
[Short Video As..](https://youtu.be/GkI_5tqO4mY)

TonyBlit, 2021-06-30:
@aLDime when you are loosing connection at the end of the video, the warning shows "RSSI LOW" instead of "RX LOSS", so it seems the RX is not triggering a proper failsafe, so gps rescue is not even starting (as shown on the BB as well). What RX are you using? Are you sure it's properly configured to trigger a failsafe in case of rx loss?

aLDime, 2021-06-30:
@TonyBlit Grateful for your answer, I have already found solutions start point: https://github.com/betaflight/betaflight/issues/6678
My receiver is the most widespread and long-known FS-IA6B (and no less famous FS-i6 transmitter with 14-channel firmware) ... It's strange that this problem (I would say BUG) has been wandering from version to version so long .. 
But as I see I have several ways to solve it:
1.Try adjusting the pulse range settings in betaflight-configurator ..
2. Try to fix the code (I know a little C ++).
3. Try using the patch https://www.rcgroups.com/forums/showthread.php?2486545-FlySky-FS-i6-8-channels-firmware-patch%21

aLDime, 2021-07-02:
@TonyBlit  Ok, now I got everything set up and got a stable RXLOSS in case of a physical absence of a signal, the RTH worked great several times, but after that I got a DROP for no reason!
In one flight - the first time everything is ok, the second time DROP instead of Rescue..
[Here is a video](https://youtu.be/EFoUUeO6Syc)
[Last Diff all](https://drive.google.com/file/d/12HAHA2TwV7TnzYl9itFeNCzkpMfslIVt/view?usp=sharing)

TonyBlit, 2021-07-02:
The config looks good so I'm not sure why it stays in the "RX LOSS" state without GPS Rescue kicking in. If you have the BB I can try to see if there's any clue there. Also I suggest that you use "set debug_mode = DEBUG_RTH" to increase the chance of getting useful info from the BB regarding this problem.

aLDime, 2021-07-02:
I have a BB for this flight, exactly DEBUG_RTH, but unfortunately this issue didn't enter in the onboard flash (only 16MB with a frequency of 1 kHz), and this wasn't the first flight that day .. But here is it maybe you will see that something: 
https://drive.google.com/file/d/1fVcPjdjO-LnLpFxVyRqGaqf3mPCirLs9/view?usp=sharing

aLDime, 2021-07-05:
Developers, will you do something with it !?

https://youtu.be/J5lDNsEvUog
https://youtu.be/HhJ3g8RhQHo

BTFL_BLACKBOX_LOG: https://drive.google.com/file/d/1Fzo3S4n_VMf58ZnDNl2PmGEBr3P66Xht/view?usp=sharing
BTFL_cli: https://drive.google.com/file/d/1RhcBmJNf80M3PHpMk5EKia6fLmrr1Um5/view?usp=sharing

If you can't do it, remove the RTH GPS Rescue function from BF and don't mislead people !!!

I, in turn, delete your under-Alpha-Flight that didn't pass Alpha-testing, since my nerves, time and money are bordering! ((((((((((((