betaflight/betaflight #6722: Spektrum vtx status and pids on tx

airwolfpv, 2018-09-09:
**Describe the bug**
VTX status on left of TX screen no longer available. And the ability to change the pids through my TX is gone. Flash back to 3. 5 and everything is back

**To Reproduce**
Steps to reproduce the behavior: Flash 3. 1. 5 on jbf4

**Expected behavior** for everything to work as it did on 3. 5

**Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)

**Setup / Versions**
 - Flight controller [what type is it, where was it bought from];
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
 - how are the different components wired up.

**Additional context**
Add any other context about the problem here.

fujin, 2018-09-09:
@airwolfpv Please supply your `diff` from CLI, your flight controller details (type, where you bought it from, revision) etc.

airwolfpv, 2018-09-09:
# diff

# version
# Betaflight / CLRACINGF4 (CLR4) 3.5.1 Sep  8 2018 / 05:03:13 (d9fb5ca13) MSP API: 1.40

board_name CLRACINGF4
manufacturer_id 

# name
name ROOSTER

# resources
resource MOTOR 5 NONE
resource LED_STRIP 1 NONE
resource SERIAL_TX 11 B08

# mixer

# servo

# servo mix

# feature
feature SOFTSERIAL
feature TELEMETRY
feature AIRMODE

# beeper
beeper -GYRO_CALIBRATED
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSED

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# serial
serial 3 64 115200 57600 0 115200
serial 30 8192 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 900 1125 0 0
aux 1 13 0 1825 1950 0 0
aux 2 35 1 1825 2075 0 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_lowpass_hz = 90
set gyro_lowpass2_hz = 290
set dyn_notch_quality = 15
set min_check = 1005
set max_check = 1950
set rssi_channel = 7
set rc_smoothing_type = FILTER
set rc_smoothing_input_type = PT1
set rc_smoothing_derivative_type = PT1
set serialrx_provider = SRXL
set motor_pwm_protocol = DSHOT600
set bat_capacity = 1300
set vbat_max_cell_voltage = 44
set ibata_scale = 373
set beeper_dshot_beacon_tone = 4
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 1
set osd_units = IMPERIAL
set osd_warn_core_temp = OFF
set osd_rssi_alarm = 25
set osd_cap_alarm = 1300
set osd_tim1 = 0
set osd_tim2 = 1
set osd_vbat_pos = 2432
set osd_rssi_pos = 2048
set osd_remaining_time_estimate_pos = 2457
set osd_anti_gravity_pos = 2453
set osd_g_force_pos = 2446
set osd_current_pos = 396
set osd_mah_drawn_pos = 2438
set osd_disarmed_pos = 2314
set osd_stat_tim_2 = OFF
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_bbox = OFF
set vtx_channel = 5
set vtx_power = 3
set vtx_freq = 5820
set vcd_video_system = NTSC

# profile
profile 0

set dterm_lowpass_hz = 95
set dterm_notch_cutoff = 0
set anti_gravity_gain = 9500
set iterm_relax = RPY
set p_pitch = 55
set i_pitch = 64
set d_pitch = 36
set f_pitch = 67
set p_roll = 52
set i_roll = 58
set d_roll = 30
set f_roll = 67
set p_yaw = 68
set i_yaw = 54
set f_yaw = 67

# rateprofile
rateprofile 0

set roll_rc_rate = 123
set pitch_rc_rate = 123
set yaw_rc_rate = 123
set roll_expo = 25
set pitch_expo = 25
set yaw_expo = 25
set roll_srate = 78
set pitch_srate = 78
set yaw_srate = 78
set tpa_rate = 12
set tpa_breakpoint = 1630

airwolfpv, 2018-09-09:
jbf4 purchased from Race Day Quads
v 1.4 i beliave  with the 9v output

AndersHoglund, 2018-09-10:
Verified on SPEEDYBEE F4 with BF3.5.1 . VTX control as such works OK, but no VTX status feedback via telemetry. In fact, no telemetry at all works. 
Reverting commit `2814ef75ba1cc206f3158769bd771cf8f9fa3fdd` (Cherry pick of PR #6670) and rebuilding firmware restores full TM functionality. 
Why was such a refactoring PR merged to this maint branch? Only bugfix patches should go in here.

AndersHoglund, 2018-09-10:
Same problem on current master (BF 4.0 build no 1110). No telemetry. Reverting commit ac5bc7bc5547d01ca898c49b66c4673540aa217c solves the issue.
@airwolfpv can you please change the subject line of this issue. It is not a VTX control nor a PID issue, it is a Spektrum Telemetry issue. Thanks.

mikeller, 2018-09-10:
@AndersHoglund: Because this fixes the problem of not properly delaying the sending of telemetry responses when multiple telemetry requests are received before a response is sent.

closed 2018-10-17 by stale[bot]