betaflight/betaflight #6723: Spektrum vtx status and betaflight screen on TX

airwolfpv, 2018-09-09:
# If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack (registration at https://slack.betaflight.com/) or other user support forums & groups (e.g. facebook).

## If you believe there is an issue with the firmware itself please follow these steps:
1. Describe your problem;
2. Include ways to reproduce the problem;
3. Provide information about your flightcontroller and other components including how they are connected/wired;
4. Add the used configuration and firmware version;
5. Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/); and
6. Remove this Text :).

fujin, 2018-09-09:
@airwolfpv Please follow the steps above, otherwise your issue will be closed.

fujin, 2018-09-09:
Actually closing as you appear to have filed a duplicate item here: #6722.

closed 2018-09-09 by fujin