betaflight/betaflight #6728: 3.5.x F7 targets do not fully boot after flashing

jflyper, 2018-09-10:
There are F7 targets that do not fully boot after flashing. Status LED flashing pattern tells initialization is complete, but the board is non-responsive. The board must be power cycled to boot normally.

There seems to be two symptoms.

1. VCP is visible, but does not connect: NERO

2. VCP is not visible: OMNIBUSF7V2, OMNINXT7, MATEKF722

Interestingly, OMNIBUSF7 does not have the issue.

DieHertz, 2018-09-10:
Know about this issue, couldn't find the cause yet. One time I caught it in a debugger it was a spurious infinite loop in `delay()`, as if `SysTick_Handler`stopped being called.
Sounds like incomplete initialization.

jflyper, 2018-09-10:
Hmm... It looks like the symptom is not deterministic.

The exact same Omnibus F7 V2 became flashed and booted stably now. Will try Matek now.

DieHertz, 2018-09-10:
No, it's definitely not deterministic, but maybe we fail to initialize some clock or SysTick properly and rely on it's state after exit from System Memory, which may presumably be different

TheMrJahn, 2018-09-10:
Have Matek, Kakute and Nero F7 boards - got this issue after flashing 3.5.2, but can say, that Nero and Kakute had this issue since 3.5.1

DieHertz, 2018-09-10:
It occurred in 3.4.

MJ666, 2018-10-10:
Same behavior on AFNG F7 boards. The VCP comes up as unknown device directly after flashing. After repower or reset the board is working normally.

tavdog, 2022-07-28:
Close?

closed 2022-07-28 by haslinghuis