betaflight/betaflight #6733: during Failsafe active !! how to regain control of quad !? - long range!

yo8aiv, 2018-09-10:
**Describe the bug**
a friend is trying to pass from iNav to BF3.5 because he saw gps rescue working but he got stuck at the failsafe not being able to regain control. he says the only way is to remove the lipo.
- i remember i also tested FS since BF 3.2  and had the same issues. retested now and when i hit failsafe switch it acts as if i disarm the quad showing me the summary screen.

**To Reproduce**
assign a failsafe switch and trigger or isolate rc TX to loose signal

**Expected behavior**
not shure, but i suppose i should regain control of the craft if link between craft and tx is recovered.

**Flight controller configuration**
# diff all

# version
# Betaflight / OMNIBUSF4SD (OBSD) 3.5.1 Sep  8 2018 / 05:40:39 (d9fb5ca13) MSP API: 1.40

board_name OMNIBUSF4SD
manufacturer_id 
mcu_id 003400263037471131313731
signature 

# reset configuration to default settings
defaults nosave

# name

# resources
resource SERIAL_TX 11 C09
resource I2C_SCL 2 B10
resource I2C_SDA 2 B11

# mixer

# servo

# servo mix

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature SOFTSERIAL
feature GPS
feature AIRMODE

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -RX_SET
beeper -DISARM_REPEAT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSED

# beacon

# map

# serial
serial 0 64 115200 57600 0 115200
serial 5 2 115200 115200 0 115200
serial 30 2048 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1700 2100 0 0
aux 1 27 1 1300 1700 0 0
aux 2 46 1 1700 2100 0 0
aux 3 13 3 1700 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail
rxfail 4 h
rxfail 5 s 1500

# master
set align_mag = CW270FLIP
set rssi_channel = 7
set serialrx_provider = SBUS
set motor_pwm_protocol = MULTISHOT
set failsafe_procedure = GPS-RESCUE
set align_board_yaw = 180
set ibata_scale = 265
set small_angle = 180
set gps_provider = UBLOX
set gps_rescue_min_sats = 4
set osd_rssi_pos = 2282
set vtx_band = 5
set vtx_channel = 2
set vtx_freq = 5695

# profile
profile 0

# profile
profile 1

# profile
profile 2

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 0

**Setup / Versions**
kakute v1
30A esc - dshot600
rx frsky xsr
tx qx7 
bf 3.5.1

**Additional context**
Add any other context about the problem here.

jflyper, 2018-09-10:
Duplicate of #6719?

mikeller, 2018-09-10:
No, this is desired behaviour, if failsafe stage 1 delay has elapsed without a connection from the TX being re-established then stage 2 will be entered based on the assumption that the connection is lost without a chance of recovery, and the only way to regain control is to power cycle the craft. If this happens too quickly for your use case, increase the stage 1 delay.

yo8aiv, 2018-09-10:
ok. thanks for your fast answer. 
some ppl suggested to disarm >< rearm  or some other stuff but nothing worked.
-  increasing the stage 1 delay might help. 
ok we will test with this.
recently i tested setting gps-rescue as FS procedure  adding it also to stage 1 as well as stage 2 
is this a proper way to do it .. if quad failsafe how to regain control !?
thx in advance.
regards.

mikeller, 2018-09-10:
@yo8aiv: GPS rescue is quite different from the normal failsafe, and to my understanding it should be possible to regain control with it when the connection from the TX is restored during the return flight / landing approach. @s0up or @dbasch  should be able to comment on this.

yo8aiv, 2018-09-10:
great. even better. Thanks . we will be testing .. and probably provide some videos.

ChilledoutUK, 2018-09-12:
In my testing Gps_rescue activated via failsafe is immediately cancelled when signal is recovered which is why i was asking for a failsafe delay to protect against intermittent radio signals.
Has this changed?

lievenvv, 2018-09-15:
I have managed to re-arm my quad in mid air after a stage2 failsafe. I just dropped the throttle to 0% and toggled the arm switch while free falling to the ground. Worked just fine!
It was a micro quad running BF3.3, iirc. Small_angle was set to 180, no other relevant changes that I can think of. Accelerometer was enabled (for crash_recovery).

Nevertheless, if there are any _other_ settings that could increase our chances of succesfully doing this, I would be interested to know!

yo8aiv, 2018-09-15:
https://www.youtube.com/watch?v=FOzbRf3Mpik 
as this guy here .. when stage 2 set to gps rescue .. once link reestablished we should be able to switch back from GPS _Rescue (FS) to normal flight mode 
- i think condition is to have the gps rescue switch activated once in stage2 FS like this when link is solid and craft is close to home can be switched out from gps rescue.
- i will test this feature tomorow 
for now i only tested GPS Rescue - since the place im flying is populated park.
Thanks BF dev's. Nice job. https://youtu.be/O8xWkJU6rh0

yo8aiv, 2018-09-16:
> In my testing Gps_rescue activated via failsafe is immediately cancelled when signal is recovered which is why i was asking for a failsafe delay to protect against intermittent radio signals.
> Has this changed?

hello . i have tested today several times and had the same behaviour.  when in failsafe and rx recovered i could retake control of the craft. it's already good this thing works. i will post some video in following day's.

in my opinion i think best would be to have a different approach :
- to disable failsafe mode / retake control (GPS-Rescue) in our case, only if user input/ interaction on sticks.
i dont know if this is doable but in my opinion is better than some delay. 
thanks dev's.

yo8aiv, 2018-10-17:
anyone with different results ?!