betaflight/betaflight #6735: UART for Caddx Turtle similar to "Runcam Device"

dridsfpv, 2018-09-10:
Runcam Split can communicate with the flight controller through a UART using the Peripherals "RunCam Device" in order to start/stop video recording using a switch with the transmitter.
Caddx Turtle is a similar camera for HD recording with RX/TX pads and I would like to request a similar feature.

mikeller, 2018-09-10:
A 'Drone Area Network (DAN)' protocol for the communication between flight controllers and peripherals like cameras, VTX, ... is being worked on at the moment. Once this is ready, it will be added to Betaflight, and Caddx will be able to implement it on their hardware to allow it to be controlled by Betaflight.

DieHertz, 2018-09-10:
IMO it's a task for hardware manufacturer to request, no Betaflight developer was contacted by Caddx to discuss it nor provide any details.

etracer65, 2018-09-11:
Or better yet the manufacturer steps up and contributes code themselves to support their product.

DieHertz, 2018-09-11:
@etracer65 only after agreeing on the protocol, we can't accept any more proprietary one-device protocols

DieHertz, 2018-09-22:
I guess they really want you to believe in it :-)

dridsfpv, 2018-09-23:
@ghostface I asked caddx too and they told me the same, that they are working with betaflight

mikeller, 2018-09-24:
@dridsfpv: If they were we'd have asked them to join GitHub to be able to provide support for their hardware, so they should see this and be able to respond in here. :wink:

Tiktiki, 2018-09-26:
Excuse me, I have been contacting you @DieHertz back in July.

You told me that you where working on the new camera/OSD/DVR/accessories protocol and refused to work with me.

Stop saying non-sense please.

Also the Caddx Turtle already has a protocol up since July, but the responses I got from the Betaflight team was a straight up "no more protocols" without even seeing it.

Tiktiki, 2018-09-26:
[Uart_Turtletv2.docx](https://github.com/betaflight/betaflight/files/2419338/Uart_Turtletv2.docx)

DieHertz, 2018-09-26:
You seem to misunderstand the issue. This protocol you're talking about was designed with no awareness nor feedback from Betaflight, it doesn't comply with project's desire to withdraw from one-device proprietary protocols in favor of open and universal ones, and in no way can be considered a "communication between Betaflight and Caddx".

Tiktiki, 2018-09-26:
In 3 months there has been no update.

I have asked numerous times to either help with drafting the protocol or at least have a base to work on.

That's hardly "no Betaflight developer was contacted by Caddx to discuss it nor provide any details." (your words).

DieHertz, 2018-09-26:
Unfortunately slack history expires quickly and we won't be able to find your message where you say that you work for caddx to implement the protocol they designed into Betaflight. It still can't be considered a healthy cooperation in my opinion.

Regarding the universal protocol, Caddx never answered with a list of requirements to discuss before designing the camera part of such a protocol.

mikeller, 2018-09-26:
@Tiktiki: Good to know that you are the official representative of Caddx, and welcome in here. In general, it is preferable to have discussions about future developments in here, and not in private, as this way other developers with an interest in Betaflight can weigh in too.

But in general what you were told by @DieHertz does hold true: Taking into consideration the shortage of flash space for a majority of the boards in use today, and the large number of manufacturers with a potential interest in having a proprietary protocol for their peripherals added, there is no interest from Betaflight to do so. I believe Caddx has been told that they can either wait for the non-manufacturer-specific 'Drone Area Network (DAN)' protocol to become ready, or if this does not suit your time frame, implement one of the already existing protocols (IRC Tramp / SmartAudio / RunCam).

Tiktiki, 2018-09-26:
@mikeller We actually talked a few months back on Slack.

Unfortunately I don't think any hardware manufacturer will jump-in with your DAN protocol, especially since there is no signs of development.

You also can't expect all the manufacturers to have custom firmwares made for everyone of their products... Pretty much all cameras used in drones have the firmware set by the sensor manufacturer, the end-assembler (Runcam, Caddx, Foxxer etc.) can flash over their logo, but usually the rest of the firmware can't be changed easily.

Flash space and processing power on the camera-side is also an issue, unless your new protocol is very simple, light and efficient, I don't see how it can work with every accessory and have manufacturers embrace it.

I am not against the idea, but having the firmware re-worked will take a few months and cost a LOT of money. It would be appreciated if your team would talk with the manufacturers about it instead of trying to come up with a universal solution.

In the mean time FlightOne is integrating the CADDX cameras.

madl3x, 2018-09-26:
@Tiktiki your protocol is extremely similar with RunCam. Why can't you do a fork on your Caddx firmware, to be compatible with RunCam proto? Start stop recording will work, at least.

mikeller, 2018-09-26:
@Tiktiki: That's possible - unfortunately our slack has a quite short holding time because we don't have the funding to be able to get a commercial subscription.

Re protocols, we are willing to work with manufacturers, and I believe some of them will see that being able to sell a product that works with a firmware that has 200000 active users (month over month) will be a competitive advantage.

FlightOne is a different story, they only have to support F4 boards, and only 2 or so of them, compared to the 130+ boards that Betaflight supports. Also, their market share is likely to be smaller than the 200000 active users (month over month) that Betaflight has. But if you think that their market share is big enough for Caddx's FPV cameras to be profitable that is up to you.

mikeller, 2018-09-26:
@Tiktiki:

> They don't use a UART port, 

This is wrong. Please revisit their protocol implementation and do a proper evaluation of the possibility to use the RunCam protocol.

> But that's a really bad patch job and won't support long-term developpement.

In my opinion, the analogue wire solution is just as good / bad as using an UART for the same primitive protocol, like what you are proposing. Future development should go towards implementing a fully fledged configuration protocol, not just one that simulates button presses, by serial or analogue pin.

Tiktiki, 2018-09-26:
@mikeller I know about that Slack issue, that's why I much prefer Gitter.im or Discord.

Remember when Betaflight started? Everyone was using Cleanflight and it was fine, supported everything on the market. Betaflight arrived and suddenly every manufacturer built F1 -F3 -F4 boards that are compatible (mostly with the Omnibus build).

If the users have the features they want in FlightOne, a lot of FC manufacturers will start making compatible ones to cater to this market.

CADDX cameras work with or without UART support, and they are willing to work with you to develop a proper digital/serial protocol to integrate to all their products.

If you prefer to work behind closed doors, that's up to you. I don't think that Caddx or any other camera/DVR/OSD manufacturer will wait for you to be finished before developing new products and features.

Users building custom version of the firmware to have the features work should give you hint that it's a required and urgent feature.

Adding yet an other filtering setup that will be visible by the 1% of pilots that are sponsored (and probably can't use anything else) might not be a priority if you want to keep the 200 000 active users (I am pretty sure your numbers are skewed, 40 000 might be a better ballpark figure).

etracer65, 2018-09-26:
@Tiktiki Wow!  Great way to make nobody want to work with you.

ghostface, 2018-09-26:
Quick clarification there are 2 protocols for the runcam cameras.
UART based (start/stop/split controls) and Voltage based (emulating the 5 key pad) - both are supported in betaflight atm.

The caddx uart protocol is pretty similiar to the runcam uart protocol.

DieHertz, 2018-09-26:
@ghostface in this case there are 3, and only 2 of them are for RunCam:
* RunCam Split protocol
* RunCam RCDevice protocol, which uses a UART to talk to a daughterboard, which emulates the 5-key joystick
* Camera Control, which emulated 5-key joystick using PWM

ghostface, 2018-09-26:
@DieHertz ah right I forgot about the uart control board thing. I stand corrected :)

madl3x, 2018-09-26:
@mikeller Despite the attitude @Tiktiki has, I don't it's fair to have a 3 special protocols for RunCam control and none for Caddx. The Betaflight code has special support for the RunCam device running on UART with two different flavors (split and normal camera) and Caddx has none.

I build 3" quads and RunCam Split Mini is merely an option for me, as it occupies a lot of space on the stack (more than 11mm disregarding spacing), while Caddx has only a single PCB that needs to be stacked. Your discussion with Caddx is not only affecting Caddx products, but also drone fans and I kindly ask you to please think about that.

For the moment, I think you can fix the problem with minimum effort, by adding a flag to the RunCam device implementation, so you can differentiate between Caddx protocol and RunCam Split protocol and add a conversion function for the commands and parameters sent to UART, if the camera is Caddx. So, you end up keeping the code as it is right now, you just add a function to adapt now the RunCam is operated if a Caddx device is selected, by converting how start/stop recording and menu commands are sent over TX UART. Wouldn't that work?

etracer65, 2018-09-26:
@Tiktiki Can you please explain your relationship with Caddx? It's not clear whether you are an employee of the company, an independent consultant hired by them, or merely an enthusiast working on your own. If you are an employee, what's your role and authority to agree on design and firmware decisions?

DieHertz, 2018-09-26:
>I don't it's fair to have 3 special protocols
Only 2 are, and that's the right of primacy, rules have changed, no new manufacturer-named protocols.

Tiktiki, 2018-09-27:
@etracer65 I am a consultant working with Caddx to try to fix their PR and code integration.

Despite what most people think, drone hardware companies are really small (Runcam/Caddx/Foxxer included) and most of the staff don't even speak English.

Also just like no FC manufacturer makes gyro chips in-house, no FPV camera manufacturer makes sensors in-house. Most use a Sony sensor actually, and that comes with some limitations.

The Turtle is "special" in the sense that the camera head is interfaced with an external MCU and allows to change the communication protocol to the camera's locked firmware.

Currently that's the only Caddx camera that has that option, but the plan would be to have a unified controller on all the Caddx cameras so that all of them work with the same protocol.

I have spoken with Caddx a lot and with the sensor's manufacturer. Integrating Betaflight's OSD to replace the camera OSD menu would cost in excess of 100 000$. It would literally mean ordering a few 1000s of custom camera sensors. The current FPV market is not able to absord that volume in the next few months, and technology advances would make that camera obsolete within a year.

That leaves us with the only option to develop a protocol to work with.
I would love to work with the BF team to use their "DAN", but there is nothing available yet to work on.

Reprogramming MCUs in the camera can't really be done, as there are no space for ports on the PCB, it has to be done pre-assembly at the SMT factory. Only the Turtlet v2 can be reprogrammed to test firmware after assembly (via the SD card).

User want to have integrated camera controls. Caddx can't really wait for months before starting development on the next product to have the special protocol documentation, especially since there really isn't much that can be done to the camera without a custom sensor board.

fujin, 2018-09-27:
As covered above, an additional manufacturer-specific protocol won't be added - it is not an option. I'd suggest you implement the Runcam RCDevice UART-based protocol until such a time as the Betaflight DAN protocol is available.

The correct way to go about this would have been to discuss at length with the project members prior to developing any Caddx camera that implements a new protocol (especially given the costs involved) instead of just turning up with demands and vague threats.

Caddx chose not to do so. They even bragged about being in contact with us, which they were not. Multiple times. They even ignored us when we challenged them about it.

You chose to continue on this path after the details were you explained to you at length.

Good day, sir.

closed 2018-09-27 by fujin

mikeller, 2018-09-27:
@Tiktiki: If you think you know more about our user numbers than us, and that it is appropriate to show this attitude, then this project is no place for you. Good bye.

@madl3x: Don't try pinning this on Betaflight (and this is not about fair anyway). The RunCam protocols were around at the time that Caddx started to propose their cloned version of the RunCam UART protocol, and they were told multiple times that if they want such a protocol they can use the RunCam protocol. So far they don't seem to have an interest in making this work.