betaflight/betaflight #6741: BF 3.5.0/3.5.1 SmartAudio freeze

Racer715, 2018-09-11:
**Describe the bug**
If I want to change the channel in the OSD, the FC freezes.

**To Reproduce**
Plug in a battery, pitch up and yaw left to enter settings, then go to FEATURES, then SA

**Expected behavior**
In my case, the FC freezes. I can still see the OSD and the camera is still working, but i can't exit the settings or arm the copter. 

**Flight controller configuration**
``` 
# diff

# version
# Betaflight / DALRCF405 (DLF4) 3.5.0 Aug 14 2018 / 03:29:50 (1d710afd7) MSP API: 1.40

board_name DALRCF405
manufacturer_id 

# name
name YS_SRC

# resources
resource MOTOR 1 A08
resource MOTOR 2 A10
resource MOTOR 3 C06
resource MOTOR 4 B00

# mixer

# servo

# servo mix

# feature
feature LED_STRIP
feature OSD

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSED

# beacon

# map
map TAER1234

# serial
serial 1 64 115200 57600 0 115200
serial 3 16384 115200 57600 0 115200
serial 4 2048 115200 57600 0 115200

# led
led 0 4,3::C:12
led 1 4,4::C:12
led 2 4,5::C:12
led 3 4,6::C:12
led 4 7,3::C:12
led 5 7,4::C:12
led 6 7,5::C:12
led 7 7,6::C:12

# color

# mode_color

# aux
aux 0 0 0 1800 1925 0 0
aux 1 15 1 1425 1525 0 0

# adjrange

# rxrange
rxrange 0 1140 1857
rxrange 1 1140 1856
rxrange 2 1142 1855
rxrange 3 1141 1857

# vtx

# rxfail

# master
set rssi_channel = 8
set rc_smoothing_type = FILTER
set serialrx_provider = SPEK2048
set motor_pwm_protocol = DSHOT1200
set align_board_yaw = 180
set small_angle = 180
set deadband = 4
set pid_process_denom = 1
set usb_hid_cdc = ON

# profile
profile 2

# rateprofile
rateprofile 0
``` 

**Setup / Versions**
 - Flight controller: DALRC F405 (not aio)
 - VTX: SabotageRc Bucaneer (same as FXT FX877T 5.8GHz)
 - SmartAudio is connected to TX5

**Additional context**
In Betaflight 3.4.1, everything works as it should.

jflyper, 2018-09-15:
Doesn't reproduce with test setup using test board (Nuk-F405RG) wired as DALRCF405, flashed with v3.5.0 and loaded with your config diff. SmartAudio device is Unify 5G8 Pro (non-HV).

There is not much more devs can do.

To further diagnose the problem, I think you need a debug tools/facility to capture what is going on with your setup.

You may also want to contact your FC and/or VTX manufacturer or distributor for assistance.

[Nuk405RG-DALRC405-BF3_5_0-SA.mov.zip](https://github.com/betaflight/betaflight/files/2385687/Nuk405RG-DALRC405-BF3_5_0-SA.mov.zip)

closed 2018-09-15 by jflyper

Racer715, 2018-09-16:
I reflashed BF 3.5.1like before and now, everything works fine. I really don't know why, but thank you for your help.

vodka-bears, 2018-09-25:
I can confirm, BF 3.5.1, OMNIBUSF4SD, exactly the same problem.

bobbfwed, 2018-11-05:
I had the same problem on BF 3.5.2, with the same flight controller (DALRCF405). It seemed to show itself shortly after enabling usb_hid_cdc.

In my OSD, the VTX band didn't have any option (completely blank on the right side), if I tried to change it, the FC froze. When I tried to change the channel, it would change the frequency to crazy values, like 56420 and 1 and 46680 and 0, and so on. I tried manually setting a frequency through the OSD. I put in a value, then it wouldn't do anything when I selected SET.

When I tried to change the VTX channel using LUA scripts, it threw an error just viewing the VTX page, crashing the LUA script. This was on BF TX LUA v1.0.2 and v1.1.0.

I ended up doing a full chip erase, and reloading all the same settings (except usb_hid_cdc being off), and now all works fine (I wasn't able to disable only that setting because the FC wasn't showing up as a COM port with that setting on).

mikeller, 2018-11-06:
@bobbfwed: What operating system are you using? In some cases, having HID support enabled will only work if extra drivers are installed: https://github.com/betaflight/betaflight/wiki/HID-Joystick-Support

bobbfwed, 2018-11-06:
@mikeller: HID worked just fine, but the COM port for programming wouldn't show up. This was on my computer I normally have where I fly. My home computer had both the HID and COM, so if I had taken it home the way it was, I would have been able to disable the usb_hid_cdc setting, but I needed to change my channel, so I just did a full flash. I might try installing those drivers to see if it fixes the problem with the COM not showing up though. Thanks.