betaflight/betaflight #6745: CMS displayport selection is not cyclic.

AndersHoglund, 2018-09-12:
When the "enter CMS" stick combinations are repeated, CMS used to cycle between all available display ports. For example OSD -> Tx telemetry -> off -> OSD -> Tx telemetry ...etc
Now I find that it is more of a "one shot", hangs and beeps when stick command is repeated. You have to power cycle the FC to resume normal CMS operation.

Bisection found that this behaviour was introduced in build 1055 and onwards, build 1054 and older are OK. Build 1055 unfortunately has a huge amount of commits, 24 in all. https://ci.betaflight.tech/job/Betaflight/1055/changes#detail4

Prime suspect in this build is however the introduction of "CMS on HOTT Text Mode". This turned out to be true by rebuilding with USE_HOTT_TEXMODE disabled. Used PR #6743 as the base and added `#undef USE_HOTT_TEXTMODE` in `REVO/target.h` and now CMS on Spektrum Tx screen via SRXL telemetry works OK. Multiple CMS entry commands cycles normally. 

FC: REVO
Tx: Dx18g1
Rx: SPM4649T

BF Config:
```
defaults nosave
feature TELEMETRY
map TAER1234
serial 0 64 115200 57600 0 115200
set serialrx_provider = SRXL
save

```
Firmware modification to hide the problem:
```
$ git diff
diff --git a/src/main/target/REVO/target.h b/src/main/target/REVO/target.h
index 1a4d07b95..189d35579 100644
--- a/src/main/target/REVO/target.h
+++ b/src/main/target/REVO/target.h
@@ -20,6 +20,8 @@

 #pragma once

+#undef USE_HOTT_TEXTMODE
+
 #if defined(AIRBOTF4)
 #define TARGET_BOARD_IDENTIFIER "AIR4"
 #define USBD_PRODUCT_STRING     "AirbotF4"

```

jflyper, 2018-09-14:
Hmm... Doesn't reproduce with Omnibus F4 V3 with SPM4649T...

I'm also confused why Hott receiver is initialized with `set serialrx_provider = SRXL`.

AndersHoglund, 2018-09-14:
Strange. I made a quick test on another rig with DYSF4PRO on build 1119, and I can see the same problem here.
```
# version
# Betaflight / DYSF4PRO (DYS4) 4.0.0 Sep 13 2018 / 21:40:33 (8207dab3e) MSP API: 1.41

```
Using iX12 and spm4649T and this config:
[betaflight_diff_180914.txt](https://github.com/betaflight/betaflight/files/2384562/betaflight_diff_180914.txt)

I have not looked into the HOTT code and checked how it behaves when not enabled.

AndersHoglund, 2018-09-14:
This was first seen on SPEEDYBEEF4. Still a problem on this FC with build 1119. 
```
# version
# Betaflight / SPEEDYBEEF4 (SBF4) 4.0.0 Sep 13 2018 / 22:14:04 (8207dab3e) MSP API: 1.41
```
[betaflight_SBF4_diff_20180914.txt](https://github.com/betaflight/betaflight/files/2384753/betaflight_SBF4_diff_20180914.txt)

To summarize, problem seen on 3 FCs:
REVO, simplest case with only the FC and Rx on the bench. Nothing else connected. With a minimal config as of first post. 
DYSF4PRO, fully flying rig with Tramp VTX and Blheli_s ESCs etc.
SPEEDYBEEF4. Also a fully flying rig.

jflyper, 2018-09-15:
Okay, loaded NukF405RG with AIRBOTF4 binary, and yes, Hott telemetry is actually registering a display port device for some reason.
```
#0  cmsDisplayPortRegister (pDisplay=0x2000bac4 <hottDisplayPort>)
    at ./src/main/cms/cms.c:91
#1  0x08033ac2 in hottDisplayportRegister ()
    at ./src/main/io/displayport_hott.c:146
#2  0x0803a5fe in initHoTTTelemetry () at ./src/main/telemetry/hott.c:353
#3  0x08038b62 in telemetryInit () at ./src/main/telemetry/telemetry.c:89
#4  0x08018d82 in init () at ./src/main/fc/init.c:607
#5  0x0800e806 in main () at ./src/main/main.c:34
```

Let's find out why.

jflyper, 2018-09-15:
Ah, `initHoTTTelemetry` is calling `hottDisplayportRegister` unconditionally. I think it should (at least) be call only if `portConfig` is non-NULL.
https://github.com/betaflight/betaflight/blob/8207dab3e527e5ee68008b700bb6fd5405ef8f1a/src/main/telemetry/hott.c#L347-L357

PR #6764 (2018-09-23): Avoid inactive telemetry port from being registered as a displayport …
Fixes #6745 

It seemed that CRSF also had the same problem.

Needs testing by someone with Hott text telemetry (@Scavanger ?) and CRSF.

- [ ] Hott case
- [ ] CRSF case
- [x] Non-Hott and non-CRSF case
closed 2018-09-23 by mikeller