betaflight/betaflight #6751: Altitude limitation and NFZ limitation

Hellorheaven, 2018-09-13:
Hi,
Due to a futur european law base on this consultation: 
http://ec.europa.eu/transparency/regexpert/index.cfm?do=groupDetail.groupMeetingDoc&docid=15211
http://ec.europa.eu/transparency/regexpert/index.cfm?do=groupDetail.groupMeetingDoc&docid=15212

I think that could be great to integer 2 functions (activable in configuration tab):
- NFZ: No fly zone if a GPS is connected and have possibility on the FC to download NFZ coordinate on sdcard (the one use by blackbox for exemple)
- Altitude limitation: with the barometer we could know the altitude (more accurate with a gps) so may be we could block the function of throttle when we arrive at the limit which has been configure

An other function to be reflect is how could we send a unique identification of the quad.

All this function has to be a choice from the pilot to apply or not.

the law in Europe will be more strict than actually and I think in other country too so we have to prepare us from this restrictive law.

thanks

DieHertz, 2018-09-13:
I don't think it's Betaflight's job to work on these regulations and limitations

varkokonyi, 2018-09-13:
Just buy a DJI. This doesn't fit into BF

Hellorheaven, 2018-09-13:
excuse me but racer/freestyle will have to observe this regulation too. that not only for dji or yuneec but for all quadcopter so if the european pilots of racer want to be able to continue to fly they will have to comply with the regulation otherwise they will be all the time annoyed by the police forces. result European pilots will no longer come on betaflight but on the competitor who has been able to adapt to regulation.

but it a betaflight choice

DieHertz, 2018-09-13:
They have their own brain, eyes and hands to comply with the regulations, I hope.

varkokonyi, 2018-09-13:
It's not BF's job to comply with the regulations. It's the pilot's job. We don't need the same BS like in the DJI drones. If we want to do this, we have to put every single NFZ into memory, something we are already running low on. Even an average road car doesn't have a no-car-zone limit. Check the airspace before you fly and you're good.

varkokonyi, 2018-09-13:
Maybe you can post this request to the ButF team, they might be interested in this feature. This would fit very well there

wind0r, 2018-09-13:
Would be quite funny to see a Tiny Whoop with a GPS module for fencing and a GSM Module to broadcast unique identification.

Just as a note. DJI does the fencing on the remote/mobile for a reason.
Having people posting proposals is good but atleast they should be 
technical feasibility and fitting to the purpose. 

This purpose may fits INAV but BF is a total strange place for that. BF Quads which would "support this" in the lower one digit % range since they dont have GPS / Baro

also about "it is betaflight choice". betaflight is a open source project so everyone can implement it. people do not need to wait until someone does it for them

ALso about unique identification. It is allready there just use the IR-Transponder and Altitude Limit you easily can do via your Radio and Telementry. Just lower throttle when the telemetry value is over xyz. Similar to GPS Fencing since GPS Fencing via SD Card will be possible which such small MCUs.

mikeller, 2018-09-13:
@Hellorheaven: These are only proposals for future laws. If they are ever implemented then there will be a transition period between the decision and the law coming into effect, giving projects like Betaflight time to review whatever will be made into law, and considering if this will require changes in the firmware to stay within the legal requirements.
Adding support for technical measures proposed in these law proposals at this stage would send the wrong signal (i.e. that we welcome and support these laws).

Hellorheaven, 2018-09-13:
@mikeller i'm totally agree with you. that could be a wrong signal and this is a law proposal. my query feature it's just for having a beginning of reflection because the adoption of this proposal by commission is for 2018/Q4 and the decision is for 2019/Q1 so really near but I think that we have since 2021 for be apply in each country in Europe . The second link explain the classification of all type of quadcopter in 4  classes and all restriction and system requirement. for our racer if this proposal will be apply we will be in C3 class and not in C0 class due to the speed of our quad...

wind0r, 2018-09-13:
Those papers also say that every UAS needs CE.
It looks like those papers just want to kill the hobby.

Hellorheaven, 2018-09-13:
@wind0r effectivly  due to lobbying :(

mikeller, 2018-09-13:
@Hellorheaven: 2021 is a long way off, so if this ever goes ahead there will be ample time to look for solutions. But it's a valid point to bring this onto the radar now.

jflyper, 2018-09-15:
I think features mentioned in this ticket is a mid to long term dev items.

The biggest problem is that we don't know how we can automatically force NFZ and altitude ceiling in non-navigational (i.e. free flying) modes.

closed 2018-10-23 by stale[bot]