betaflight/betaflight #6752: TBS unify nano Smartaudio Lite

vincentholm, 2018-09-13:
TBS unify Nano smartaudio lite not working!? Sorry if this is the wrong place, but the smartaudio on my tbs unify pro nano is not working. I have connected it the same way as all my unify pro hv race on the omnibus f4 v6. (Audio wire to TXsomething). Am i supposed to do it different? I couldn't find information on this

etracer65, 2018-09-13:
The unify nano requires a 10K pulldown resistor to be connected to the audio line for it to work. Connect a 10K resistor from the audio line to ground.

vincentholm, 2018-09-13:
> The unify nano requires a 10K pulldown resistor to be connected to the audio line for it to work. Connect a 10K resistor from the audio line to ground.

What do you mean by audio to ground? Ground on the vtx? Or ground on the FC? I'm sorry, but i have never worked with resistors before..

etracer65, 2018-09-13:
Connect a 10k resistor between the "SMART AUDIO" and "GND" connections on the nano.  TBS should have included this on the board but made a mistake.

[image]

It's also noted in the manual:

[image]

I'm closing this as it's a hardware question and not a Betaflight firmware issue.

closed 2018-09-13 by etracer65