betaflight/betaflight #6754: Is it possible to run rudimentary flight control software on a Calliope Mini (Arduino pendant)?

Horst86, 2018-09-13:
Hi to all you creative Devs!

I am a very experienced Pilot and flying Betaflight from day one and since ;)
I've grown a lot with the hobby. Thanks for giving me that possibility to use such sophisticated code and pretty much take it for granted...

Now having the possibilty to give back something to the community by teaching the younger ones to program and to fly is a great opportunity. For this i'm having the idea on getting a learning-based microprocessor-board to fly. This board is the Calliope Mini (open source Hard- and Firmware, the UK/German Arduino if you will)

Specs are : https://calliope-mini.github.io/v10/

My primary question is: can the Hardware that is on there 

- ARM Cortex M0 
-   16MHz
-   16kB RAM
-   256kB Flash
-   BMX055 ACC/GYRO/MAG
-   UARTs
-   I2C  

handle the amount of code that is needed for stable Angle-mode flight?

Furthermore, would there be a person i can talk to on getting the Betaflight Code to work on this Calliope Mini Target?

Keep up the great work and happy flying ;)?

Philipp Horstmann 

[image]

[image]

[image]

watageek, 2018-09-13:
Have a look on RCGroup for the H8 mini firmware.

https://www.rcgroups.com/forums/showthread.php?2512604-Eachine-H8-mini-acro-firmware

jflyper, 2018-09-13:
I think something like http://mydronelab.com/blog/arduino-quadcopter.html would be a better starting point.

Horst86, 2018-09-13:
Thanks for your fast tips and links on this...

Hardware specs seem alright so far? IMU is onboard in contrary to the arduino copter project from mydronelab...

watageek, 2018-09-13:
https://github.com/silver13
😉

wind0r, 2018-09-13:
@Horst86 Hardware specs for what? Having a MCU and IMU is enough to make a Quadrocopter firmware but for Betaflight that will not that good. The small issue is that there isnt any driver for that IMU and the bigger problem is that the mcu just has 16mhz. Even the slow F1 targets which got dropped some time ago have had 72mhz

using some other project as a base would be the better idea since it should be easier to port to

jflyper, 2018-09-13:
@Horst86 Onboard gyro would be better. Only concern is that you might have to write (or modify) gyro driver to fit the drone lab gyro api.

Another concern is software compatibility, especially in low layers, of m0 based Arduino v.s. original at mega based ones.

I would
1. Try compiling drone labs sketch with m0 Arduino ide first. 
2. Write, if necessary, a fake gyro module that always return flat position and no acc.
3. Test motor outputs with logic analyzer.

Probably a week end job for an experienced programmer 😉

Horst86, 2018-09-13:
Overwhelming answers. Thank you so much guys.

See, that 16MHz was also the first concern that got to my mind (looptime to the moon ;) )
The other issues about writing more sophisticated dummy-code, i will discuss with an experienced programmer buddy since i have very little actual coding experience myself... more on the hardware side

Again, thanks for the very valuable input up to this point. Its much appreciated

Happy flying ;)

jflyper, 2018-09-15:
@Horst86 

Closing, as this is not a BF specific issue.

But I'll be looking forward to hear your progress 😃

closed 2018-09-15 by jflyper