betaflight/betaflight #6755: Where should I change the "taskMainPidLoop"

0Jiahao, 2018-09-13:
Hi, I am trying to add some new functionalities (mainly calculate the desired attitude and assign them to the pid.c) to the betaflight. When I compiled my code the following error message showed up. Any idea where can I change the "taskMainPidLoop"
/////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
make -j ./obj/betaflight_3.3.1_OMNIBUSF4SD.hex
make[1]: Entering directory '/home/jiahao/betaflight_autonomous'
%% version.c 
%% (speed optimised) pid.c 
Linking OMNIBUSF4SD 
/tmp/cc9sEHQ8.ltrans5.ltrans.o: In function `taskMainPidLoop':
<artificial>:(.text+0x41f8): undefined reference to `ol_roll'
<artificial>:(.text+0x50a0): undefined reference to `ol_pitch'
collect2: error: ld returned 1 exit status
Makefile:285: recipe for target 'obj/main/betaflight_OMNIBUSF4SD.elf' failed
make[1]: *** [obj/main/betaflight_OMNIBUSF4SD.elf] Error 1
make[1]: Leaving directory '/home/jiahao/betaflight_autonomous'
Makefile:407: recipe for target 'hex' failed
make: *** [hex] Error 2

jflyper, 2018-09-13:
I think you have added references to variables called `ol_pitch` and `ol_roll` somewhere in your code, but they are not declared as variables.

Btw, `taskMainPidLoop` is in `src/main/fc/core.c`.

closed 2018-09-13 by jflyper

0Jiahao, 2018-09-13:
@jflyper Let me simplify my attempt, I created a '.c' and a '.h' in '/src/flight'.
/////////////////////////////////// my_test.c /////////////////////////////////////////
#include "flight/my_test.h"
float ol_pitch = 0;
float ol_roll = 0;
/////////////////////////////////// my_test.h ////////////////////////////////////////
extern float test_pitch;
extern float test_roll;

When I assigned these two to the pid controller, it failed. Any idea?

DieHertz, 2018-09-13:
Your new files must be listed in source.mk to be compiled

0Jiahao, 2018-09-13:
@DieHertz Hero!