betaflight/betaflight #6756: Dynamic switching of VTX power based on RSSI

jabacz, 2018-09-13:
Most pilots are using VTX power more than 25mW because of flying in challenging areas with trees, buildings etc. 
VTX power set to, perhaps 200mW, causes interference to your friends sitting near to you when you start or approach landing. 

Detect close proximity to pilot using RSSI and switch VTX power according RSSI detected using SA.

Example (subject to user configuration, if agreed). When RSSI shows value over 80, keep VTX at 25mW using SA command. When RSSI drops bellow 80, use SA command to switch VTX to 100mW (200mW). When RSSI comes back over 80, fallback to 25mW. Some spike filtering would be needed to avoid random switching. Perhaps last few seconds average can do the job.

mikeller, 2018-09-13:
In addition to avoiding the waste of CPU cycles, I am also not sure if changing the output power will work for all VTX hardware that is out there **without causing a momentary drop out of the video signal**, happening at a time that the pilot has no control over, which sounds like a recipe for a crash.

jabacz, 2018-09-13:
To the flicker during the change: Did you ever fly in woods? :-) 
The momentary video dropouts are "normal" feature of every FPV. I doubt you will distinguish VTX power change from going behind a tree. At least when I was trying this manually through BF stick config with TBS Unify I did not notice any major video dropout.
To the CPU cycles: Maybe this was true with F1/F3 CPU. I doubt, the RSSI evaluation is a big overhead to displaying the same value in the OSD. Especially with F4/F7 computing power. 
In addition, I had a word about this with a friend going long range. The GPS distance can be another trigger for VTX power change.

SchonRichtig, 2018-09-14:
VTX Power by RSSI would be nice,
if it is too much, maybe we can use a switch, not only for pit mode we already have.
A 3-position switch for 25-200-800mw.
or a potentiometer, that i have on the taranis and have no other use for it :)

jabacz, 2018-09-16:
Confirmed. I have TBS Unify, Unify HV, Unify race, Unify nano and EWRF Mini e7082TM. All do switch without noticeable disruption, or just a flicker. No such thing like dropout worth crashing. 

I can agree with free channel switching instead of RSSI. The automatic would be better, but if it's too complicated, or whatever, the switching like pitmode ON/OFF on RC command could be easier to deploy.

SchonRichtig, 2018-09-17:
Using a Switch/poti solution, you can implement the RSSI switching in your Radio (Open-TX)
and maybe others, if you want to.

mikeller, 2018-09-18:
To me the switch solution looks a whole lot better - even in the presence of a blackout during a switch this will give the pilot the liberty to choose at what time the blackout happens.

SchonRichtig, 2018-09-18:
yeah, I’ve hoped this was implemented in the past but that was the pit mode solution :)
I don’t know if it is possible to use one mode for VTX Power, or one mode for each power level? 
Ideally there is also a short notification in the OSD

jflyper, 2018-09-20:
I'm beginning to recall the past discussion. RSSI based switching is not good because

1. RC RSSI is not directly related with VRX reception (these often run in different frequencies and different reception conditions including antenna topology),

2. RC RSSI characteristic is different by frequency bands and RC technology deployed, and

3. current RSSI implementation for some RC technology is not linear response; it is even a simulated RSSI in some case that present abrupt drop of RSSI value at the edge of reception.

jabacz, 2018-09-20:
Hi jflyper,

1) of course it's not directly related, but generally it correlates. For reason mentioned it has not to be 100% precise.
2+3) Yes, I know, if RC system reports RSSI wrong way, the user should not use this feature :-)

Anyway we could start by adapting current "Pitmode on switch" to "VTX power on the switch". This can help to situation mentioned in the first post of this thread as well. 
Understand: having 25mW -> XXX mW on the switch the same way as we now have Pit_mode -> XXX mW

mikeller, 2018-09-21:
@jabacz: I am a big supporter of 'function chaining', and implementing VTX power as an adjustment range (i.e. use an RC channel to select one of all available power levels) sounds like a good candidate for this. If desired it will be easy for users to build their own 'RSSI based VTX power' function on the TX with this, and even more, they can still have an 'override switch' allowing them to boost VTX power if the RSSI value is not an accurate representation of the required VTX power level.

SchonRichtig, 2018-09-24:
@ghostface 
Great Solution, but you said that the switching can be ignored if I switch to fast through the power levels?
And for your solution you must have a telemetry receiver.
I think a BF integrated solution is better, it will work with every brand and every receiver, and
there also could be a OSD popup with the selected power level

ghostface, 2018-09-24:
@SchonRichtig yeah I agree built in would be nice. Yes it requires a telemetry receiver. It does have the advantage of not taking up an additional channel of the 8 available (8 to get minimum latency).

The ignoring when switching too fast might have something todo with it trying to send another telemetry command before the first one is done but I am not sure, it might also be the vtx who simply rejects/ignores additional instructions while it is still processing one.

A delay between sending the command or something like that would probably fix it in the lua script.or maybe something else is wrong - I am not a lua expert and I just debugged the original script and fixed it up till it worked :)

jabacz, 2018-12-03:
Hi.
I have an update. I have made a personal progress in GIT usage. I can clone Betafilght locally, I can run a build with working output. I was digging into the VTX code. I have tried some changes, but I've got lost in cross reference of modules and variables. 
Is here anyone, possible the author of previous VTX code, to give me a helping hand to understand the consequences?
BR
JaBa

Klaus-Michael, 2019-02-06:
The ability to change the VTX midflight on a Switch is the one feature I'm missing since moved to Betaflight from Kiss.
I would love to see this implemented-

Hawxxer, 2019-08-16:
Progress on this one?

wrenfpv, 2019-10-04:
Any update on this ?

SchonRichtig, 2019-10-14:
Unfortunately, no

I still think that this would be an useful option
But I’m not a programmer, I have no idea how complicated it is to 
Implement a mode for the vtx power levels.
Shout be simple, but I could be soo wrong 😊

yoodog, 2019-11-07:
cool, that would be great add on!

closed 2026-09-26 by blckmn