betaflight/betaflight #6758: Airbot Omnibus F4 V6 LED issue

greennuts, 2018-09-13:
So have issue with Airbot Omnibus F4 V6 LEDs.
Flashed OMNIBUSF4FW firmware, but couldn't use LED or TX1 (S/A) for smart audio. SA didn't work on TX2 as well
As SA is primary, flashed with OMNIBUSF4SD and have at least 3 uarts working

I'm missing something?

jflyper, 2018-09-13:
Yeah, your config `diff`.

jflyper, 2018-09-13:
Btw, LED strip pad might not work unless camera control resource is released.

Here's my personal memo.

---
#### ACC/GYRO selection and orientation
set gyro_to_use = SECOND

#### Orientation
MCU side up, USB connector to the left
set align_gyro = CW0
set align_acc = CW0

MCU side up, USB connector to the right
set align_gyro = CW180
set align_acc = CW180
 
#### LED strip
LED strip signal (labelled LED) does not work if CAM_CTL is used.
Workaround is to use alternative pads for LED strip signal connection.

(a) Option 1
Any one of motor 5, motor 6, TX6 and RX6 pads work.
Once this option is taken, rest of the pads in this group can not be used for timer related functions.
resource led_strip c6  # TX6 
resource led_strip c7  # RX6 
resource led_strip c9  # M5
resource led_strip c8  # M6

(b) Option 2
RX4 works.
resource led_strip a1   # RX4

(c) Option 3
TX2 works, but this option is likely to collide with S/A function on TX2.
resource led_strip a2  # TX2

(d) Option 4
TX1 works, if not used for other purposes such as a receiver.
resource led_strip a9  # TX1