betaflight/betaflight #6761: Note on acc/gyro detection change for configurable acc/gyro

jflyper, 2018-09-14:
Toward an initial release of generic target in 4.0, acc/gyro was made configurable by #5868, and it introduced some peculiarity which needs some attention for users as well as devs and distributor of boards mentioned here.

The lists may not be complete, and there may be additional issues which is not yet found and not described here. Comments are welcome.

#### acc/gyro alignment

Alignment configuration for acc/gyro in target.h will be based occurrence number (1 or 2), as opposed to device type based one. This will force users of some target with multiple acc/gyro possibilities to perform post flashing configuration of sensor alignment.

Following targets are known to have this property.

- FURYF3
- MATEKF405
- MATEKF722
- SIRINFPV

Note that F4 and F7 targets are less problematic as these will go generic target with 4.0.

#### Boards with I2C acc/gyro only (F4 and F7)

Some F4/F7 targets only configures I2C acc/gyro, which can not be handled by the current `USE_MULTI_GYRO` facility which assumes all F4/F7 targets have SPI based acc/gyro.

Following targets are known to have this property:

- CRAZYFLIE2
- NUCLEOF7 (Less problematic as this target is for dev board)

#### `USE_TARGET_CONFIG` and `targetConfiguration( )` is deprecated

This is not configurable acc/gyro specific, but devs should be aware that this facility will be gone.

0crap, 2018-12-20:
Wow, that escalated quicly. 
I can confirm my Matek F405-STD needs an "set align_board_yaw = 270" to fix this.
Target MATEKF405 (MKF4).

FuryFPV, 2019-03-17:
Problem still existing with Matek F405 boards - using 90 on YAW at the moment with boards pointing forward.  Surely not a big fix???

mikeller, 2019-03-17:
@FuryFPV: This will be fixed with the Universal Targets, to be released with a configurator update some time before 4.1.

TonyBlit, 2019-03-25:
I stumbled upon this issue yesterday when upgrading a quad with a Matek 405, from BF 3.5 to BF4. I got a few runaways that left me completely puzzled as I wasn't expecting that simply upgrading BF without any fancy filter config couldn't work straight away.
So I wonder if this issue should not be warned IN CAPS on the release notes, to prevent people upgrading their quads and having unexpected runaways, which is frankly a bit dangerous.

mikeller, 2019-03-25:
@TonyBazz: Good point - seeing that 4.0 will start out with 'legacy targets' there needs to be an entry in release notes (I DON'T DO CAPS, SORRY).

I am aware that this is a problem in practice with MATEKF405 boards, since they are sold in different versions with different gyro models which are mounted with different orientations.

Are you (or anybody else for that matter) aware of any other boards that now need a custom config and need to be mentioned in the release notes?

TonyBlit, 2019-03-25:
So polite @mikeller ^^

I'm not familiar with how the orientation is defined in the legacy targets, but there should be a way to search through the codebase to find all the problematic targets, isn't there?

mikeller, 2019-03-26:
@TonyBlit: Not sure if there is a way to distinguish from looking at the target definition alone if a particular board that has 2 gyros defined does so because it is sold with those 2 gyros installed (often one as an add-on caged version), or if it is because there are different variants of the board sold, each of them with one of the gyros.

So far I am only aware of the MATEKF405 being a case of the latter, hoping for the community to report back if there are others...

0crap, 2019-03-26:
I only know the original Matek F405 as http://www.mateksys.com/?portfolio=f405-std
That board does now need the "set align_board_yaw = 270" fix.

It would be interesting to know if there are users with Matek F405 targets, other then te above, that work correctly with the BF 4.x defaults.
Are there really much around?

0crap, 2019-03-26:
> I stumbled upon this issue yesterday when upgrading a quad with a Matek 405, from BF 3.5 to BF4. I got a few runaways that left me completely puzzled as I wasn't expecting that simply upgrading BF [...]

Did you check if the quad moved correctly in the BF Configurator? ;-)

Twistedsocal, 2019-03-28:
Is this the same thing as the gyro being flipped 180 ie mirrored? If so then cool ill just wait if not that is an issue i got with the 405 in my babayhawk r pro. fixed as soon as i downgraded again

davenose, 2019-04-12:
You might want to update the 4.0.0 release notes; I see there was consideration for that above, but didn't see it in the release notes.  My friend and I wasted a bit of time on our MATEKF405-STD based quads a couple days ago until we stumbled on the solution.  I would speculate we are not the only ones that wasted some time.  FYI, MATEK has updated their website with bold warnings for the affected products.

Is this still planned to be addressed for a pre-4.1 configurator release?

mikeller, 2019-04-14:
@davenose:

From https://github.com/betaflight/betaflight/releases/tag/4.0.0:

> the gyro configuration was unified from being done by gyro hardware model to being generic for all gyro models. This means that for some targets that are used for multiple different boards that come with different gyro models that are mounted with different orientations, it is necessary from Betaflight 4.0 on to manually set gyro_1_sensor_align (and / or gyro_2_sensor_align for boards with a second gyro) to match the gyro orientation on the board that is used. See note #6761 for an explanation, and instructions for individual boards. This is a temporary workaround, per-board configurations with correct gyro alignment will be released as Unified Target configurations (#5868);

Not sure how much more explicit (including the link to this issue) this can be made?

davenose, 2019-04-15:
I did see that on my first review of the release notes, but I suppose as I'm not an expert I skipped over that as a possible concern for our quads by neglecting to consider that the MATEKF405 could be affected.  More explicit would be including the list of affected FCs in the release notes itself, which would have caught our attention.  Your call on how to proceed.  Thanks for all the hard work you guys do!

mikeller, 2019-04-16:
@davenose:

- we do not have a conclusive list of which of the ~ 170 targets we support are affected by this, and publishing an incomplete list would only give some affected users the impression that they were not affected;
- the release notes are very long this time already, making it less likely that they are actually read by users - I tried to minimise them by not dwelling on individual issues longer than needed.

For these reasons I opted to link to the GitHub issue instead for this.

davenose, 2019-04-21:
@mikeller 
Thanks for the explanation ... that makes sense.

Mozzzzi, 2019-04-25:
Hey Folks. For completeness for us not-so-tech-savvy. I have a Matek F405-AIO. The settings I required to fix this in BF4.0.0 are as below. Now all good.  Thanks.
In BF Configurator Config screen:
-	Yaw Degrees = 0 degrees 
-	Sensor 1 Yaw CW90 degrees 
-	This translates to the following CLI Command settings
o	set gyro_offset_yaw = 0
o	set gyro_to_use = FIRST
o	set align_board_roll = 0
o	set align_board_pitch = 0
o	set align_board_yaw = 0
o	set gyro_1_sensor_align = CW90
[image]

abimaputra78, 2019-04-27:
I have Matek F405-STD. in BF4 just set 1st Gyro to “CW180". everything were ok.

tavdog, 2022-06-15:
Close time?

haslinghuis, 2022-06-15:
Outdated

closed 2022-06-15 by haslinghuis