betaflight/betaflight #6766: CLI command presets

jonerFPV, 2018-09-15:
CLI command presets would be a feature where user can pre-define commands that are run when the preset is selected especially from the OSD.

For example one can try pre-defined filter setting combos quickly on flying session without any need of computer or multiple parameter tweaking (through OSD). The user could pre-define these sets at home before going flying. 

Another use case would be when the copter gets beat up, for example props get damaged during race and the pilot doesn't have time to change filter settings. Then the pilot could quickly switch to "harder filtering" + easier PIDs to get the copter flying without burning motors.

BF could include some presets for easier testing, for example the ones listed in here: https://github.com/betaflight/betaflight/wiki/Tuning-Tips-for-Betaflight-3.4

jonerFPV, 2018-09-15:
Closed because similar (not identical) to another issue that has been closed.

jflyper, 2018-09-15:
@jonerFPV In that case, you have to bring up a new point that is strong enough to keep it open.

jonerFPV, 2018-09-16:
The other ticket was referring to AUX switch which is kinda limited/blind system. I was thinking of presets (with title) that works through OSD or configurator perhaps.

The new point is that it's a really preset system with explanations not limiting to only 2-3 different profiles which you need to know when flying. With OSD you could do basic setups without computer and know what you're activating.

I was thinking how to implement this. The "scripts" or presets should be stored in quite efficient way to config. One solution would be to store the "parameter" as config.lookup type following with the parameter data. One parameter record could then vary in length and be efficiently stored. The data would be read sequential so to know where data starts / end. All records would have header with 16 char description. The whole parameter scripting system would have a header with lookup table where different profile data starts at. 

If there's any positive sign I could try to implement this. However I wouldn't like to do it "wrong" and need to rewrite the change because of some global error of thinking how it should be implemented. So if any ideas / go-ahead-sounds-good I could look into this.