betaflight/betaflight #6773: Cant activate/configure LED on Omnibus Target 3.5.1

ps915, 2018-09-16:
I update the Tiny Brushless "ET100" from ldarc.com from 3.1.7 to 3.5.1 but i cant activate the LED Feature (not via GUI and also not via CLI)

I noticed, that the resources have changed. Here is a resource dump of the 3.1.7 where the LED worked:

```
# Betaflight / OMNIBUS 3.1.7 Apr  3 2017 / 22:14:20 (e1c4b5c)
name ps915
resource BEEPER 1 C15
resource MOTOR 1 B08
resource MOTOR 2 B09
resource MOTOR 3 A03
resource MOTOR 4 A02
resource MOTOR 5 B07
resource MOTOR 6 B06
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource MOTOR 9 NONE
resource MOTOR 10 NONE
resource MOTOR 11 NONE
resource MOTOR 12 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 B04
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A14
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A15
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 NONE
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource SERIAL_RX 12 NONE
```

Its resource A08. After installing a fresh (full chip erase) OMNIBUS Firmware 3.5.1, i noticed, that the resource changed to:

```
# resource stock 3.5.1
resource LED 1 B03
resource TRANSPONDER 1 A08
```
But LED nor TRANSPONDER Features are not activated. I then tried to activate the LED Featrue cia GUI. Did not worked. Then i tried to change the resources:

```
resource Transponder 1 none
resource LED 1 A08
```

But in the end i cant activate the LED Feature at all.

jflyper, 2018-09-16:
@ps915 
LED strip support has been dropped from F3 targets due to flash size shortage.

Btw, the resource you are fiddling with, `LED`, is not a LED strip but the onboard status LED.

closed 2018-09-16 by jflyper

ps915, 2018-09-16:
Ah, damn your right! Thanks :( Whats the last release with LED support?

jflyper, 2018-09-16:
I believe it was 3.5.0.

mikeller, 2018-09-16:
3.4.1 was the last version of OMNIBUS with LED_STRIP support:

```
mikeller@pc:~/git/betaflight$ git diff 3.4.1 3.5.0  -- src/main/target/OMNIBUS
diff --git a/src/main/target/OMNIBUS/target.h b/src/main/target/OMNIBUS/target.h
index 3c6a8646f..309ce2fd4 100644
--- a/src/main/target/OMNIBUS/target.h
+++ b/src/main/target/OMNIBUS/target.h
@@ -29,6 +29,12 @@
 #undef USE_ITERM_RELAX
 #undef USE_RC_SMOOTHING_FILTER
 
+#undef USE_LED_STRIP
+
+#undef USE_HUFFMAN
+#undef USE_PINIO
+#undef USE_PINIOBOX
+
 #undef USE_TELEMETRY_HOTT
 #undef USE_TELEMETRY_MAVLINK
 #undef USE_TELEMETRY_LTM
[...]
```

ps915, 2018-09-16:
Danke Michael! :)

andrejpodzimek, 2019-04-22:
Just for the recod, for those who desperately want LEDs back: You can look at the `target.h` file, identify and remove features you don’t need and add LEDs back. I did this to keep my LEDs working with Betaflight 4 on my “vintage” Kingkong 130GT “hot rod” (`SPRACINGF3`).

Here are a few obvious candidates for removal from `SPRACINGF3` (because you always have at most one baro, one compass or one rangefinder &mdash; if any at all):

```c
#define USE_RANGEFINDER
#define USE_RANGEFINDER_HCSR04
```

```c
#define USE_MAG
#define USE_MAG_AK8975
#define USE_MAG_HMC5883
#define USE_MAG_QMC5883
```

```c
#define USE_BARO
#define USE_BARO_BMP280
#define USE_BARO_MS5611
#define USE_BARO_BMP085
```

Remove whatever you don’t need and then add the LEDs back:

```c
#define USE_LED_STRIP
#define USE_LED_STRIP_STATUS_MODE
```

With a bit of luck, it will still fit in the flash space. The obvious drawback is that you’ll be stuck with a custom build, which makes things harder if you want to report a bug. (You’ll need to reproduce it with a regulare release.)