betaflight/betaflight #6777: Betaflight 3.5.1 Yaw bounce

ijustwannafly, 2018-09-17:
I am getting a weird behavior on two quads I flashed to 3.5.1.

Acro mode, no accelerometer enabled so no stability modes.

DYS F4 pro v1 and v2.

All defaults tested and even settings tested from the bf 3.4/3.5 tuning guide.  

For now I have feed forward transition at 0.5 but all feed forward set to 0.  Even pid tuned RPY and still came close to defaults on PID values.

Roll and Pitch are fine but yaw seems to "bounce" back.

In a flat spin with just yaw input it will complete a 360 degree spin but when I go back to center stick (as best as possible) the quad bounces to one side (usually opposite the spin direction) and then goes back to its "desired" position.  It's almost as if it's purposely undershooting after the completion to ensure it stops where the gyro thinks it should be.

I thought it might have been my finger twitching on the stick and not always hitting center stick on yaw.

I put a battery on my bf 3.2.2 quad that has been pid tuned and it does not have this same behavior.

I even dropped yaw p down to 25 and yaw I down to 25 with 0 D and 0 feedforward.  I tried low P high I, high p low I, even high and low feedforward and the yaw insists on bouncing.

It's not a hard bounce.  To be sure again, I flashed the same problem quad back to 3.2.2 and pid tuned it and there is no yaw bounce.

Any ideas?

jflyper, 2018-09-17:
3.5.x is completely different from 3.2.x, so they are not directly comparable.

I suggest you visit one of the support forums/slack.

closed 2018-09-17 by jflyper

ijustwannafly, 2018-09-18:
Although it is not "directly comparable" there was an issue that presented itself that was never present in previous versions.

Since I was directed to a slack/support forum path I did the troubleshooting on my own.

I verified the issue presented itself over multiple quads only changing bf versions.

I now know exactly what setting presents the issue but since it seems to be "ON" as part of the defaults and majority of the users are not complaining but rather proclaiming it "flies locked in" on defaults I won't press the matter.

Heck, I'm using 0 feed forward on rpy and 0 feed forward transition and I still get fast consistent rolls/flips.

I will admit the filters and the separation of the feed forward from d term made it faster/easier to tune d-term so at least thanks for that.

jflyper, 2018-09-19:
We can reopen this if we have something we can all look at, like bb log and diff output.