betaflight/betaflight #6783: Yaw, strange behaviour since BF 3.5 or 3.51 and also latest build BF 4.0

Ralfde, 2018-09-17:
When I make a fast stop, everything seems to be normal.
When I make small or no yaw movements, the yaw does move itself.
Please see the videos and logfiles.

etracer65, 2018-09-17:
Based on your other (now closed) issue, have you tried using the correct target?  It seemed earlier that you were trying to use OMNIBUSF4SD on a OmniNXT F7.

Also you need to provide more information/context.  Where's the settings `diff`?  What flight mode are you using?  Do you have a GPS or compass installed?  Trying to use head-free mode?

Nobody can help you if you don't provide information.  Also remember that this is a place to report problems with Betaflight - not a place to get support on your individual hardware configuration.

Ralfde, 2018-09-17:
Also you need to provide more information/context. Where's the settings diff? 
I used this setting diff on a OmnibusF4SD and OmiNXTF7. Both show the same problem.
What flight mode are you using? attitude mode, I think it's also in the rate mode, but I'm not 100% sure.
Do you have a GPS or compass installed? Yes, a GPS is installed, not compass. Looks like the problem does only occur, when a GPS is installed.
Trying to use head-free mode? no

Ralfde, 2018-09-17:
board_name OMNINXT7
manufacturer_id 
mcu_id 003d00253136511136323831
signature 

# reset configuration to default settings
defaults nosave

# name

# resources
resource MOTOR 1 B01
resource MOTOR 2 B00
resource MOTOR 3 B04
resource MOTOR 4 B05

# mixer

# servo

# servo mix

# feature
feature GPS
feature TELEMETRY

# beeper

# beacon

# map
map TAER1234

# serial
serial 0 2048 115200 57600 0 115200
serial 1 8 115200 57600 0 115200
serial 3 64 115200 57600 0 115200
serial 5 2 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 1 1300 2100 0 0
aux 1 1 0 1000 1700 0 0
aux 2 46 2 1000 1200 0 0
aux 3 26 3 1800 2000 0 0
aux 4 28 1 1300 2100 0 0
aux 5 30 3 1000 1200 0 0
aux 6 35 1 1975 2100 0 0
aux 7 39 8 1800 2000 0 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set align_gyro = CW270
set gyro_lowpass_hz = 90
set gyro_lowpass2_hz = 0
set gyro_use_32khz = ON
set gyro_to_use = SECOND
set align_acc = CW270
set acc_calibration = -2,86,2
set mag_declination = 2420
set rssi_channel = 16
set rssi_scale = 125
set rssi_offset = -12
set rc_smoothing_type = FILTER
set fpv_mix_degrees = 35
set serialrx_provider = SUMD
set blackbox_p_ratio = 128
set motor_pwm_protocol = DSHOT1200
set failsafe_procedure = GPS-RESCUE
set ibata_scale = 200
set ibata_offset = -3500
set yaw_motors_reversed = ON
set small_angle = 180
set gps_provider = UBLOX
set gps_rescue_angle = 20
set gps_rescue_initial_alt = 20
set gps_rescue_descent_dist = 30
set gps_rescue_ground_speed = 1000
set gps_rescue_throttle_max = 1500
set gps_rescue_sanity_checks = RESCUE_SANITY_OFF
set deadband = 10
set yaw_deadband = 10
set pid_process_denom = 2
set osd_vbat_pos = 2102
set osd_rssi_pos = 2137
set osd_vtx_channel_pos = 2497
set osd_crosshairs_pos = 2253
set osd_ah_sbar_pos = 2254
set osd_ah_pos = 2126
set osd_current_pos = 2094
set osd_mah_drawn_pos = 2089
set osd_gps_speed_pos = 2273
set osd_gps_sats_pos = 2241
set osd_home_dir_pos = 2286
set osd_home_dist_pos = 2306
set osd_altitude_pos = 2262
set osd_pidrate_profile_pos = 2467
set osd_stat_max_dist = ON
set osd_stat_max_alt = ON
set vtx_band = 5
set vtx_channel = 3
set vtx_freq = 5732

# profile
profile 0

set dterm_lowpass_hz = 90
set dterm_notch_cutoff = 0
set iterm_rotation = OFF
set iterm_relax_cutoff = 9
set yaw_lowpass_hz = 90
set throttle_boost = 0
set acro_trainer_angle_limit = 40
set p_pitch = 45
set d_pitch = 35
set f_pitch = 100
set p_roll = 40
set d_roll = 30
set f_roll = 100
set p_yaw = 45
set d_yaw = 10
set f_yaw = 100
set abs_control_gain = 20

# profile
profile 1

set dterm_notch_hz = 260
set iterm_rotation = OFF
set iterm_relax = RP
set iterm_relax_type = SETPOINT
set yaw_lowpass_hz = 100
set throttle_boost = 0
set acro_trainer_angle_limit = 40
set p_pitch = 45
set d_pitch = 35
set f_pitch = 100
set p_roll = 40
set d_roll = 30
set f_roll = 100
set p_yaw = 45
set d_yaw = 10
set f_yaw = 100
set abs_control_gain = 20

# profile
profile 2

set dterm_notch_hz = 260
set iterm_rotation = OFF
set iterm_relax = RPY
set iterm_relax_type = SETPOINT
set yaw_lowpass_hz = 100
set throttle_boost = 0
set acro_trainer_angle_limit = 40
set p_pitch = 45
set d_pitch = 35
set f_pitch = 100
set p_roll = 40
set d_roll = 30
set f_roll = 100
set p_yaw = 45
set d_yaw = 10
set f_yaw = 100
set abs_control_gain = 10

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

set roll_rc_rate = 106
set pitch_rc_rate = 106
set roll_srate = 76
set pitch_srate = 76
set tpa_rate = 50
set tpa_breakpoint = 1050

# rateprofile
rateprofile 1

set roll_rc_rate = 106
set pitch_rc_rate = 106
set roll_srate = 76
set pitch_srate = 76
set tpa_rate = 50
set tpa_breakpoint = 1250

# rateprofile
rateprofile 2

set roll_rc_rate = 106
set pitch_rc_rate = 106
set roll_srate = 76
set pitch_srate = 76
set tpa_rate = 50
set tpa_breakpoint = 1250

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save

etracer65, 2018-09-17:
Set yaw D to 0 - traditional quads should not use D on yaw.

Also disable absolute control `set abs_control_gain = 0`

Ralfde, 2018-09-17:
Ok, I will try

Ralfde, 2018-09-18:
Tested yaw D 0 and D10 this morning with set abs_control_gain = 0.
It worked fine. Thanks a lot.
Just I would like to know, why abs_control_gain did make this effect.

ctzsnooze, 2018-09-18:
Most likely it was the D on yaw causing it, not absolute control.

Ralfde, 2018-09-18:
ok. Thanks a lot.

closed 2018-09-18 by Ralfde