betaflight/betaflight #6791: USB HID Joystick - problematic axes range

Cleric-K, 2018-09-18:
Currently, the HID emulation uses single byte values for the axes ranges [ -127; 127 ]
Some games have trouble with the negative axes values - for example FPVFreerider, FPV Air 2.
FPVFreerider completely disregards the negative values (truncates to 0), while FPV Air 2 seems to take the absolute value of the negative. In both cases the controls are unusable.

I looked at https://github.com/betaflight/betaflight/blob/9b51fb321643c55ad8d0b0d5500454faf5d46597/src/main/io/usb_cdc_hid.c#L46-L47

Normally, axes values for modern devices are in the range [0; 32767]. Unfortunately changing the definitions above is of no use, because the HID report itself uses single byte values.

I have no experience with the betaflight USB HID code.
Would it be hard to change the HID report to use two byte values per axis?

I can do it if someone points me at the code where the changes should be made.

mikeller, 2018-09-18:
@chrish did most of the work for the USB HID support in Betaflight, maybe he can help?

In general, USB HID uses a lot of the library code provided by STM, and it would be desirable to keep using it in a form that is modified a little as possible, to reduce the work required to update the libraries at a later stage.

mikeller, 2018-09-21:
I think testing it (or potentially adding a special HID-only mode that does not run other tasks) can't hurt. Another thought: Would changing the range to 8 bit unsigned help?

conkerkh, 2018-09-21:
I think we should test both cases first I can create two branches on my repo and anyone Who cares can test it aparat myself and we will decide what to do. We can have to descriptors, one with higher res (u16) other one with u8

Cleric-K, 2018-09-25:
Sorry for late comment - I was on the road.
@conkerkh I can make the tests. Did you push the branches already?

Personally I think the u16 version would be better, because after all, the single byte loses precision against the 1000 units of input. But it should really be tested to see if the doubling of payload really degrades performance.

conkerkh, 2018-09-27:
@Cleric-K you can check my branch at https://github.com/conkerkh/betaflight/tree/HID_U16 compile for whatever target you need or get it from CI, it doesn't work for F7 targets as it is just quick POC.

conkerkh, 2018-09-27:
btw I didn't check if it works:P

Cleric-K, 2018-09-27:
I'm completely at loss here. I encounter the weirdest behavor.

At the moment I turn on my transmitter the report data becomes almost random. I capture the frames with USBLyzer.

To be certain I replaced the code in `sendRcDataToHid()` with simple `memset(report, 0x1, sizeof(report))`.
When I boot the fc and the transmitter is off I get the proper reports:

Usage | Range Lg/Ph | Value
-- | -- | --
X | [0..2048] | 257
Y | [0..2048] | 257
Z | [0..2048] | 257
Rx | [0..2048] | 257
Rz | [0..2048] | 257
Ry | [0..2048] | 257
Vx | [0..2048] | 257
Wheel | [0..2048] | 257

```
00000000  01 01 01 01 01 01 01 01 01 01 01 01  . . . . . . . . . . . .  
0000000C  01 01 01 01                          . . . .                  
```

At the moment I turn on the transmitter the random data starts to come in. Here are 5 consecutive frames:

Usage | Range Lg/Ph | Value
-- | -- | --
X | [0..2048] | 32814
Y | [0..2048] | 2051
Z | [0..2048] | 0
Rx | [0..2048] | 256
Rz | [0..2048] | 32892
Ry | [0..2048] | 2051
Vx | [0..2048] | 0
Wheel | [0..2048] | 24832

```
00000000  2E 80 03 08 00 00 00 01 7C 80 03 08  .  . . . . . . |  . .  
0000000C  00 00 00 61                          . . . a                  
```

Usage | Range Lg/Ph | Value
-- | -- | --
X | [0..2048] | 32814
Y | [0..2048] | 2051
Z | [0..2048] | 0
Rx | [0..2048] | 256
Rz | [0..2048] | 32814
Ry | [0..2048] | 2051
Vx | [0..2048] | 0
Wheel | [0..2048] | 256

```
00000000  2E 80 03 08 00 00 00 01 2E 80 03 08  .  . . . . . . .  . .  
0000000C  00 00 00 01                          . . . .                  
```

Usage | Range Lg/Ph | Value
-- | -- | --
X | [0..2048] | 0
Y | [0..2048] | 0
Z | [0..2048] | 0
Rx | [0..2048] | 0
Rz | [0..2048] | 32816
Ry | [0..2048] | 2051
Vx | [0..2048] | 0
Wheel | [0..2048] | 256

```
00000000  00 00 00 00 00 00 00 00 30 80 03 08  . . . . . . . . 0  . .  
0000000C  00 00 00 01                          . . . .                  
```

Usage | Range Lg/Ph | Value
-- | -- | --
X | [0..2048] | 32924
Y | [0..2048] | 2051
Z | [0..2048] | 0
Rx | [0..2048] | 24832
Rz | [0..2048] | 32884
Ry | [0..2048] | 2051
Vx | [0..2048] | 0
Wheel | [0..2048] | 24832

```
00000000  9C 80 03 08 00 00 00 61 74 80 03 08    . . . . . a t  . .  
0000000C  00 00 00 61                          . . . a                  
```

Usage | Range Lg/Ph | Value
-- | -- | --
X | [0..2048] | 32750
Y | [0..2048] | 2051
Z | [0..2048] | 0
Rx | [0..2048] | 8448
Rz | [0..2048] | 32810
Ry | [0..2048] | 2051
Vx | [0..2048] | 0
Wheel | [0..2048] | 24832

```
00000000  EE 7F 03 08 00 00 00 21 2A 80 03 08  î  . . . . . ! *  . .  
0000000C  00 00 00 61                          . . . a                  
```

The data is not completely random, there is some pattern, but the big question is how is this at all possible. The `memset`ted report is completely disregarded!

It could be an hardware issue (I don't have another controller to test). It is not a problem on the PC side because the same happens on two different machines.
But this does not happen in the single byte version.

Any ideas?

Cleric-K, 2018-09-27:
Ok, made some progress.

It seems that the `report` buffer, being in the stack, is not consumed synchronously by `USBD_HID_SendReport`. Most probably it is being sent after the function exits and other functions modify the memory that belongs to the same part of the stack.

I moved the `report` variable outside the function, making it global and the problem with the randomness is solved. There are still some problems with the ranges - seems like the 0-2048 range is not liked by Windows unless going through calibration. Also it looks like there's some jerkiness, but I have to go to bed now. Will try to check again tomorrow.

conkerkh, 2018-09-27:
we can just declare it static inside the function should be enough as it will stay on stack and won't be global which isn't really what we wan't I'll test this now anyway

Cleric-K, 2018-09-28:
The weirdness continues...

I made the range from 0 - 32767 (0x7fff) because this seems to be what is most used for 16bit axes. This works on Windows without calibration.

But now there are other problems. Even though the report uses its own memory (as global or static variable) there still seem to occur some strange effects. The bytes in the report are misplaced.

I've put a simple counter in the beginning of the `sendRcDataToHid()` which allows for example only every 100th execution - thus I space apart the executions which gives enough time for the USB system to complete sending before the code of `sendRcDataToHid()` is executed.
This solved the problem.
It seems that `sendRcDataToHid()` writes to the buffer before the USB sending is complete which leads to bizarre behavior.

My conclusion is that there should be some kind of busy checking of the USB port. Only if the port is idle should we continue to fill the `report` buffer and then send it. I don't know if such state checking is possible.

conkerkh, 2018-09-29:
There is busy checking on USB port first of all, second I tested this without your modifications and everything works fine for me on W10, just have static buffer so it doesn't get poped off stack. Also joystick doesn't need calibration with 0-2048 I also looked at raw packets from USB and they look ok.

conkerkh, 2018-09-29:
The problem is with optimization or LTO which messes up everything I'll go again through my code as clearly something goes wrong there, if you compile for debug all is good.

conkerkh, 2018-09-29:
Also if you change range of sent data you need to change descriptor as well.

Cleric-K, 2018-09-29:
I have no other fc so I can't test and be sure if my problem is a corner case.

About the range: yes, I also changed the descriptor.

I think you are right that any range should work. I suspect that Windows caches the calibration data for the device and when I change the descriptor it does not override it automatically. Only if I hit reset in the calibration tab.

About the busy USB: maybe the USBD_HID_SendReport() function is protected from overlapping but the way I read the code I dont see anything that will stop the report buffer to be overwritten on every call to sendRcDataToHid(), which maybe interferes with the usb dma stuff. Correct me if I'm wrong.

By the way single byte unsigned also works ok, I tested it. So if things get too difficult with u16, maybe u8 is an option.

conkerkh, 2018-09-30:
I made some test, and got it working in the end with U16 with -32767, 32767 range, there was one define which was missing... However from what I can see Liftoff doesn't work with only positive values, I checked descriptors for XLite (pretty much same for Taranis) it has range of -127, 127, next XBOX 360 controller has also from negative to positive -32768, 32768. In general this is true for all the the gamepads (controllers) even the HID specification:

"Traditionally, a joystick driver applies its own scaling to values returned from a joystick. That is, the driver simply linearizes and translates the range of values generated by the stick into normalized values between 0 and 64K, where 32K is centered. The application (game) then interprets the normalized values as necessary. Because of this, joysticks normally do not declare Units or Physical Minimum and Physical Maximum values for their axes. Depending on the driver, these items may be ignored if they are declared."

Becuase of this I'm quite against any modification that would make the virtual joy from BF have a range from 0 to X. I would suggest developers to simply fix their sims. BTW FPV Freerider worked totally fine for me with neg values (default descriptor).

Cleric-K, 2018-10-01:
Well, fair enough.
I agree that it is sim's problem.

It is strange that you say you have no problems with Freerider. Just now I tested it both on Windows 7 and 10 and the behavior is the same.
It is also a well known issue, for example with taranis, which as you say have range -127 to 127. Not really an "issue", because the author of Freerider knows about that and that's why he supplied [this guide](https://drive.google.com/file/d/0BwSDHIR7yDwSOGhPMTFyWEpDVWM/view?usp=sharing) for setting it up. Basically it moves the output to the upper half such that the HID axes moves only in the 0 - 127 range.

Btw it is not really Freerider's fault - it turns out that it is [a problem with the Unity engine](https://answers.unity.com/questions/986618/inconsistent-output-from-inputgetaxis-across-contr.html). But strangely only with some joysticks. Other Unity based games which do not have that problem seem to use some other library for Input.

It would be interesting to know what differentiates joysticks which do work ok with unity vs those which do not. If it is a matter of some special descriptor element it would be nice if it could be implemented.
Unfortunately I don't have a joystick that has negative axes and does not have the problem.
1/ taranis has the problem
2/ My Thrustmaster Topgun afterburner (with negative axes) has the problem
3/ xbox controller clone does _not_ have the problem but it uses only positive axes 0 - 127

The true xbox 360 I think uses completely different subsystem (xinput) so it is hard to draw conclusions with it.

Well, as long as no one else has really complained about it we may just close the issue.

mikeller, 2018-10-02:
From following this discussion I think joystick support seems to be a wasteland of missing standards and conventions, and I suspect that whatever we do in the firmware, there will always be at least one use case where the input provided won't work for the app consuming it.

Not sure if any of this could be improved by providing custom joystick drivers for Windows that took care of properly making the supported input range known?

conkerkh, 2018-10-02:
Only thing that comes to my mind is that we can scale up to using U16 and add an option in CLI to configure range with cutting negative values if someone needs it. This will require calibration in Windows when you chose to use them but will take care of all situations I hope.

conkerkh, 2018-10-02:
Descriptor could stay the same it simply won't send negative values

Cleric-K, 2018-10-02:
I thought about that CLI option also, but I refrained to mention it, simply because it really gets a little too technical for a user that just wants to plug the quad and play the sim. It surely is a solution, but it really makes me sad that something that should be so trivial would require such fiddling.

My only hope for seamless experience at this point is if we find some joystick which works with the generic Windows HID driver, uses signed axes and yet does not have problems.
The HID descriptor of such a joystick may give us a hint what makes it different from one which does not work ok.

As I mentioned already, my Thrustmasters stick uses signed axes but has the problem.
My gamepad uses unsigned axes and as such does not have a problem.

I don't have any other game controllers but I'll see if I can find something else to test.

Cleric-K, 2018-10-02:
It is quite strange that Unity has not addressed that issue for so many years.
I've just looked at [https://forum.unity.com/threads/gamepads-and-getaxisraw.67906/](https://forum.unity.com/threads/gamepads-and-getaxisraw.67906/) where developers experience the very same problem. They confirm that the problem is with the negative side of signed axes.

conkerkh, 2018-10-02:
This can be also added to configurator, but I think for whatever reason its Windows specific...

ledvinap, 2018-10-02:
@Cleric-K : It may be easier to copy hid_report on start of transaction in HID driver ...

conkerkh, 2018-10-02:
This is handled by interrupt no DMA is used on USB this is not a problem!

ledvinap, 2018-10-02:
@conkerkh : Even interrupt based USB may try to send report when new version is being written. This will be invisible with 8bit values, but may create glitch  with 16 bits

conkerkh, 2018-10-02:
It's not the case here, try it for yourself, only glitches I get are from RSSI channel (which is not a glitch anyway, RSSI is simply changing all the time). I tested this with MacOS and W10 

`Id	Type	Time	Length	Hex	Ascii	
15	R	0.000000	17	00 83 00 84 00 00 00 41 00 00 00 01 80 f3 3d 00 00	.......A......=..	
15	R	0.007912	17	00 83 00 c5 00 00 00 41 00 00 00 01 80 2e 3d 00 00	.......A......=..	
15	R	0.007990	17	00 83 00 84 00 00 00 41 00 00 00 01 80 28 3c 00 00	.......A.....(<..	
15	R	0.007684	17	00 83 00 84 00 00 00 41 00 00 00 01 80 a5 3b 00 00	.......A......;..	
15	R	0.008437	17	00 83 00 c5 00 00 00 41 00 00 00 01 80 e6 3b 00 00	.......A......;..	
15	R	0.007949	17	00 83 00 c5 00 00 00 41 00 00 00 01 80 2e 3d 00 00	.......A......=..	
15	R	0.008140	17	00 83 00 c5 00 00 00 41 00 00 00 01 80 2e 3d 00 00	.......A......=..	
15	R	0.008065	17	00 83 00 c5 00 00 00 41 00 00 00 01 80 ab 3c 00 00	.......A......<..	
15	R	0.015792	17	00 83 00 84 00 00 00 41 00 00 00 01 80 e6 3b 00 00	.......A......;..	
15	R	0.008089	17	00 83 00 c5 00 00 00 41 00 00 00 01 80 a5 3b 00 00	.......A......;..	
15	R	0.007854	17	00 83 00 c5 00 00 00 41 00 00 00 01 80 e6 3b 00 00	.......A......;..	
15	R	0.008074	17	00 83 00 84 00 00 00 41 00 00 00 01 80 2e 3d 00 00	.......A......=..	
15	R	0.008016	17	00 83 00 84 00 00 00 41 00 00 00 01 80 ab 3c 00 00	.......A......<..	
15	R	0.006826	17	00 83 00 c5 00 00 00 41 00 00 00 01 80 6a 3c 00 00	.......A.....j<..	
15	R	0.009299	17	00 83 00 84 00 00 00 41 00 00 00 01 80 e6 3b 00 00	.......A......;..	
15	R	0.006767	17	00 83 00 84 00 00 00 41 00 00 00 01 80 a5 3b 00 00	.......A......;..	
15	R	1.017059	17	00 83 00 c5 00 00 00 41 00 00 00 01 80 6a 3c 00 00	.......A.....j<..	
15	R	0.008006	17	00 83 00 84 00 00 00 41 00 00 00 01 80 f3 3d 00 00	.......A......=..	
15	R	0.008063	17	00 83 00 84 00 00 00 41 00 00 00 01 80 b1 3d 00 00	.......A......=..	
15	R	0.006943	17	00 83 00 c5 00 00 00 41 00 00 00 01 80 ed 3c 00 00	.......A......<..	
15	R	0.007929	17	00 83 00 c5 00 00 00 41 00 00 00 01 80 ab 3c 00 00	.......A......<..	
15	R	0.007954	17	00 83 00 84 00 00 00 41 00 00 00 01 80 ed 3c 00 00	.......A......<..	
15	R	0.008037	17	00 83 00 c5 00 00 00 41 00 00 00 01 80 b1 3d 00 00	.......A......=..	
15	R	0.009097	17	00 83 00 c5 00 00 00 41 00 00 00 01 80 f9 3e 00 00	.......A......>..	
15	R	0.014819	17	00 83 00 c5 00 00 00 41 00 00 00 01 80 b7 3e 00 00	.......A......>..	
15	R	0.008061	17	00 83 00 84 00 00 00 41 00 00 00 01 80 f3 3d 00 00	.......A......=..	
15	R	0.007995	17	00 83 00 84 00 00 00 41 00 00 00 01 80 2e 3d 00 00	.......A......=..	
15	R	0.008428	17	00 83 00 84 00 00 00 41 00 00 00 01 80 2e 3d 00 00	.......A......=..	
15	R	0.007568	17	00 83 00 c5 00 00 00 41 00 00 00 01 80 b1 3d 00 00	.......A......=..	
15	R	0.007999	17	00 83 00 c5 00 00 00 41 00 00 00 01 80 f9 3e 00 00	.......A......>..	
15	R	0.008959	17	00 83 00 c5 00 00 00 41 00 00 00 01 80 b7 3e 00 00	.......A......>..	
15	R	0.007037	17	00 83 00 84 00 00 00 41 00 00 00 01 80 f3 3d 00 00	.......A......=..	
15	R	0.016005	17	00 83 00 84 00 00 00 41 00 00 00 01 80 2e 3d 00 00	.......A......=..	
15	R	0.007974	17	00 83 00 c5 00 00 00 41 00 00 00 01 80 2e 3d 00 00	.......A......=..	
15	R	0.007985	17	00 83 00 84 00 00 00 41 00 00 00 01 80 b1 3d 00 00	.......A......=..	
15	R	0.008051	17	00 83 00 84 00 00 00 41 00 00 00 01 80 f9 3e 00 00	.......A......>..	
15	R	0.007930	17	00 83 00 84 00 00 00 41 00 00 00 01 80 b7 3e 00 00	.......A......>..	
15	R	0.009041	17	00 83 00 84 00 00 00 41 00 00 00 01 80 76 3e 00 00	.......A.....v>..	
15	R	0.008093	17	00 83 00 84 00 00 00 41 00 00 00 01 80 f3 3d 00 00	.......A......=..	
15	R	0.007812	17	00 83 00 84 00 00 00 41 00 00 00 01 80 34 3e 00 00	.......A.....4>..	
15	R	0.016070	17	00 83 00 84 00 00 00 41 00 00 00 01 80 3a 3f 00 00	.......A.....:?..	
15	R	0.007247	17	00 83 00 c5 00 00 00 41 00 00 00 01 80 41 40 00 00	.......A.....A@..	
15	R	0.008738	17	00 83 00 c5 00 00 00 41 00 00 00 01 80 3a 3f 00 00	.......A.....:?..	
15	R	0.007425	17	00 83 00 c5 00 00 00 41 00 00 00 01 80 34 3e 00 00	.......A.....4>..	
15	R	0.008498	17	00 83 00 c5 00 00 00 41 00 00 00 01 80 2e 3d 00 00	.......A......=..	
15	R	0.007406	17	00 83 00 c5 00 00 00 41 00 00 00 01 80 2e 3d 00 00	.......A......=..	
15	R	0.008212	17	00 83 00 c5 00 00 00 41 00 00 00 01 80 34 3e 00 00	.......A.....4>..	
15	R	0.008521	17	00 83 00 84 00 00 00 41 00 00 00 01 80 b1 3d 00 00	.......A......=..	
15	R	0.015488	17	00 83 00 84 00 00 00 41 00 00 00 01 80 ed 3c 00 00	.......A......<..	
15	R	0.008053	17	00 83 00 84 00 00 00 41 00 00 00 01 80 ab 3c 00 00	.......A......<..	
15	R	0.007658	17	00 83 00 84 00 00 00 41 00 00 00 01 80 6a 3c 00 00	.......A.....j<..	
15	R	0.008805	17	00 83 00 84 00 00 00 41 00 00 00 01 80 ed 3c 00 00	.......A......<..	
15	R	0.007582	17	00 83 00 c5 00 00 00 41 00 00 00 01 80 34 3e 00 00	.......A.....4>..	
15	R	0.008462	17	00 83 00 c5 00 00 00 41 00 00 00 01 80 b1 3d 00 00	.......A......=..	
15	R	0.007496	17	00 83 00 c5 00 00 00 41 00 00 00 01 80 ed 3c 00 00	.......A......<..	
15	R	0.007999	17	00 83 00 c5 00 00 00 41 00 00 00 01 80 28 3c 00 00	.......A.....(<..	
15	R	0.016669	17	00 83 00 84 00 00 00 41 00 00 00 01 80 a5 3b 00 00	.......A......;..	
15	R	0.007775	17	00 83 00 84 00 00 00 41 00 00 00 01 80 6a 3c 00 00	.......A.....j<..	
15	R	0.007597	17	00 83 00 84 00 00 00 41 00 00 00 01 80 f3 3d 00 00	.......A......=..	
15	R	0.008629	17	00 83 00 c5 00 00 00 41 00 00 00 01 80 b1 3d 00 00	.......A......=..	
[... cut ...]
15	R	0.007000	17	00 83 00 84 00 00 00 41 00 00 00 01 80 82 40 00 00	.......A......@..	
15	R	0.008835	17	00 83 00 c5 00 00 00 41 00 00 00 01 80 82 40 00 00	.......A......@..	
15	R	0.008094	17	00 83 00 c5 00 00 00 00 00 00 00 01 80 05 41 00 00	..............A..	
15	R	0.007923	17	00 83 00 84 00 00 00 00 00 00 00 01 80 ca 41 00 00	..............A..	
15	R	0.008172	17	00 83 00 c5 00 00 00 41 00 00 00 01 80 11 43 00 00	.......A......C..	
15	R	0.007791	17	00 83 00 c5 00 00 00 41 00 00 00 01 80 d0 42 00 00	.......A......B..	
`

conkerkh, 2018-10-02:
First of all STM32 has FS Core and works as FS device, second there is a 1.25k buffer inside the FS Core which buffers data for the EP so it doesn't make sense to tripple buffer everything. Tested this on both W10 and MacOS machines, absolutely no glitches. I'll test this on raspberry pi which is absolute piece of shit hardware and if it works there properly I assume there is a problem on your side not the code.

Cleric-K, 2018-10-03:
Ok, thanks, Chris,

So it must be some strange problem with my particular board I guess (OMNIBUSF4 target).

There must be something, otherwise making the buffer `static` would not have any effect if the buffer was copied right away by `USBD_HID_SendReport()`. And I trust you that _it is_ copied right away in the hardware buffer.. but still I'm also quite certain I was not hallucinating when the axis data was jumping around :)

So it is clear that it is a matter of something quite obscure and probably some interaction between different hardware. As I said, the jumping around started when I turned on the transmitter. So could be some weird interference between the UART and the USB ... I have no Idea, just guessing.

I'll try to do some more rigorous testing tonight.

Thank you for your help!

conkerkh, 2018-10-03:
I'm also certain you aren't hallucinating, however, having a scientific approach here is a must IMO. I honestly hate when stuff doesn't work as it should, we don't know why, then we fiddle, do sth it works and we still really don't know why this doesn't really contribute to the future and in most cases is a cause of error somewhere later on.

Did you build the f/w yourself or downloaded from Travis CI?

conkerkh, 2018-10-03:
I'm certainly interested what's going on in your case, you can try force erasing the whole flash. FYI for some reason in my case couple of FCs had some regions locked to the point I had to erase through debugger as it didn't work properly using DFU mode and MCU was behaving weird.

Cleric-K, 2018-10-03:
Alright,

So I cleaned up everything and started over. It works fine!

So the root problem seems to be in the `#define HID_IN_PACKET 16`.

When it is `8` the glitches appear but only after the transmitter has been turned on.
My apologies for insisting on the modifying-buffer-while-sending stuff but here's what has mislead me:
I've just repeated the experiment to be certain that I was not dreaming :)
There are three stages (all with `#define HID_IN_PACKET 8`):
1. When the `report` buffer is in the stack I see the almost random axes values (as I've sent examples from USBlyzer)
2. When I make the buffer `static`, the behavior changes! There's no more random movement but another bizarre thing happens - the first 8 bytes and the remaining 8 are the same! Like this:

```
00000000 2E 80 03 08 00 00 00 01 2E 80 03 08 
0000000C 00 00 00 01                
```
Effectively this makes it look like the axes move in pairs.
I immediately thought (what seems logical even in retrospect) that in the first case the USB subsystem was reading from the buffer which now belongs to the stack of other functions and thus the random data. Just by making the buffer `static` (nothing else) the behavior changes: now there's no more random axes, but still with problem - they move in pairs. Yet I was certain that the problem was because of asynchronous access to the buffer.

3. The third stage was to add some artificial delay to `sendRcDataToHid()`. I simply made a counter and let only every 100th execution to actually execute. Otherwise it simply returns.
Now the joystick works with rather low refresh rate (obviously) but everything works as expected! All axes respond correctly.
This  for me was the final proof that there must be something to do with modifying the buffer while the USB send is not yet completed.

I hope you see the line of my experiments and the reason why I insisted that the buffer should be protected :)

Now, setting `#define HID_IN_PACKET 16` , completely fixes everything. The `USBD_HID_SendReport()` function truly copies its buffer before returning - I tested by making `memset()` to the `hid_report` right after it and it has no effect.
Thus the `static hid_report` is actually not needed anymore. We can save the 16 bytes of RAM.

Well, why in the `HID_PACKET 8` case, everything behaves as it really reads the buffer asynchronously is beyond me. And honestly I don't have the energy nor the time to fiddle with it anymore.

Now to some experimental results:

I tested all the sims that I have with different axes ranges.

1. [-32768; 32767]

Sim| Result
-- | --
Freerider | Negative side of axes is truncated
FPV Air 2 | Negative side of axes is truncated
Velocidrone | Works
Liftoff | Works
DRL | Works

2. [0; 2048], [0; 32767]

Sim| Result
-- | --
Freerider | Works
FPV Air 2 | Works
Velocidrone | Works
Liftoff | Works
DRL | Works

All tests are on a Win7 machine.

Another thing I have found is that Windows uses the descriptor Logical Min/Max as the default calibration data.
After I change the min/max in the descriptor and flash the fc Windows still uses the previous calibration data! Probably it caches it on a USB Port/VID/PID basis and does not expect that the descriptor might have changed.
Only when I Reset to Defaults (and then close the dialog with OK), the calibration is reread from the descriptor values:
[image]

From my experiments it seems that only Velocidrone depends on this Windows calibration data. Other sims seem to read the raw values on a lower level and don't care how it was calibrated.

I'll be grateful if you can make one more test with positive axes (and just in case Reset the calibration data to default, to reread the descriptors) and see if you'll have problems.

PS: I saw that you have corrected the order of Rx, Ry, Rz axes in the descriptor, but maybe `hidChannelMapping[]` should also be corrected.

conkerkh, 2018-10-04:
You had exact same behavior as I did before modifying mentioned define which is why I insisted to test it again:) I didn't dig to deep in the complete mess of ST USB Device drivers I think that's understandable considering absolutely crazy way it's written.

Anyways getting to the root of what we are doing here, in order to keep compatibility with other systems we have to keep descriptor with negative values. Since there are issues with negative axis in some sims I think doing pg is the best case scenario here, it's very easy to implement and the only problem is the fact you need to calibrate when using Velocidrone I think it's quite popular so let's simply provide people with instructions what to do they usually have to read something to enable hid anyway, or simply we have two switches in configurator (one enables HID, the other one only positive axis with a note you need to calibrate joystick in Windows), I think you have some commits in configurator so perhaps you could do it?

mikeller, 2018-10-04:
My hunch would be to extend `usbDevConfig()->type` to contain the different subtypes of the composite device that need to be supported, and reuse the CLI and MSP for that.

bkeinert, 2018-12-15:
Is there still development on this issue? I would really like to have something that works with I16/U16.

Peter-B-, 2021-01-06:
Hello everybody, thanks a lot for your effort on this topic. I'm facing the same problem right now. 

Do you think it makes sense to continue your work and try to rebase it to the current firmware? I'm new to this project, but I might be able to help here.

mikeller, 2021-01-10:
This should have been taken care of by making the Joystick device behave in a way compliant with Windows requirements in #10153, closing this.

closed 2021-01-10 by mikeller