betaflight/betaflight #6802: Omnibus F3 (target OMNIBUS) LED_STRIP not active nor can be ativated (BF 3.4.0 works)

feiticeir0, 2018-09-19:
From BF above 3.4.0 until last version, is not possible to activate the LED_STRIP feature. 
Is activated (either CLI or GUI) and after a reboot is not active. 

**To Reproduce**
Just flash any BF version above 3.4.0

**Expected behavior**
It's expected to activate the LED_STRIP and after a reboot, it stays active

**Flight controller configuration**
```
# diff
# Betaflight / OMNIBUS (OMNI) 3.5.1 Sep  8 2018 / 05:38:57 (d9fb5ca13) MSP API: 1.40

name FT0_ARF95

feature -RX_PPM
feature RX_SERIAL
feature MOTOR_STOP
beeper -GYRO_CALIBRATED
beeper -ARMING_GPS_FIX
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSED
beacon RX_LOST
beacon RX_SET
serial 0 1 115200 57600 0 115200
serial 2 64 115200 57600 0 115200
set gyro_notch1_hz = 400
set gyro_notch1_cutoff = 300
set gyro_notch2_hz = 200
set gyro_notch2_cutoff = 100
set acc_hardware = NONE
set baro_hardware = NONE
set baro_noise_lpf = 0
set baro_cf_vel = 0
set baro_cf_alt = 0
set min_check = 1100
set rssi_scale = 30
set rc_interp_ch = RP
set max_aux_channels = 6
set serialrx_provider = IBUS
set motor_pwm_protocol = DSHOT600
set motor_pwm_rate = 400
set failsafe_delay = 10
set beeper_dshot_beacon_tone = 4
set servo_lowpass_hz = 400
set pid_process_denom = 3
set osd_vbat_pos = 2433
set osd_rssi_pos = 22
set osd_flymode_pos = 364
set osd_throttle_pos = 129
set osd_vtx_channel_pos = 200
set osd_current_pos = 97
set osd_mah_drawn_pos = 111
set osd_craft_name_pos = 2506
set osd_gps_speed_pos = 66
set osd_gps_sats_pos = 1410
set osd_altitude_pos = 161
set osd_pid_roll_pos = 322
set osd_pid_pitch_pos = 354
set osd_pid_yaw_pos = 386
set osd_power_pos = 47
set osd_pidrate_profile_pos = 418
profile 0

set dterm_notch_cutoff = 0
set anti_gravity_threshold = 350
set iterm_windup = 50
set throttle_boost = 0
set p_pitch = 32
set i_pitch = 23
set d_pitch = 24
set p_roll = 30
set i_roll = 20
set d_roll = 24
set p_yaw = 68
set i_yaw = 25
set d_yaw = 20
set f_yaw = 0
set d_level = 100
rateprofile 0

set roll_expo = 30
set pitch_expo = 30
set yaw_expo = 20
set roll_srate = 87
set pitch_srate = 87
set yaw_srate = 80
```

**Setup / Versions**
 - Flight controller : Mini BetaFlight F3 Brushless FC with Integrated OSD - FORTINET F3+OSD (Gearbest)
 - RX is FRSKY Pro Receiver

**Additional context**
Until BF 3.4.0 it works fine.  Craft is the ARFun Pro 95

feiticeir0, 2018-09-19:
version 3.4.1 also works ! Above does not !

jflyper, 2018-09-19:
Duplicate of #6773?

feiticeir0, 2018-09-19:
I really did a search for a similar issue and none I found. 
Closed because it is a duplicate ! 

Thank you

closed 2018-09-19 by feiticeir0