betaflight/betaflight #6828: add LEMONF3 board

Raffa2s, 2018-09-22:
Is possible to add the native support to LEMONF3 board to betaflight? 
Thank you :)

jflyper, 2018-09-22:
I saw Eachine E010S mentioning LEMONF3 as flight controller with Betaflight as firmware. Your earlier posting suggest that it was flashed with a version of Betaflight. If that is a case, Eachine must disclose the Betaflight fork they are using, as demanded by the Betaflight license.

In summary, you should ask Eachine, not here.

Raffa2s, 2018-09-22:
Ok I understood :) thank you! I had written to eachine, I will keep you all update! 
Thank you :)

Raffa2s, 2018-10-23:
i had the hex file for the version 3.3.0 of BF, can it be usefull?

mikeller, 2018-10-23:
@Raffa2s: There is no simple or easy way to extract the required information for the target definition out of a hex file, so I am afraid that won't help. Not sure if @flyinglemonfpv had a hand in the design of this board and could potentially supply the target definition?

closed 2018-11-29 by stale[bot]

tomasris, 2019-03-23:
Any news about this cosmic-scale problem ?

EachineRC, 2020-07-16:
# resources
resource  BEEPER  1  C15
resource  MOTOR  1  A00
resource  MOTOR  2  A01
resource  MOTOR  3  A02
resource  MOTOR  4  A03
resource  MOTOR  5  A06
resource  MOTOR  6  A07
resource  MOTOR  7  B00
resource  MOTOR  8  B01
resource  MOTOR  9  B10
resource  MOTOR  10  B11
resource  MOTOR  11  NONE
resource  MOTOR  12  NONE
resource  SERVO  1  NONE
resource  SERVO  2  NONE
resource  SERVO  3  NONE
resource  SERVO  4  NONE
resource  SERVO  5  NONE
resource  SERVO  6  NONE
resource  SERVO  7  NONE
resource  SERVO  8  NONE
resource  PPM  1  A15
resource  PWM  1  NONE
resource  PWM  2  NONE
resource  PWM  3  NONE
resource  PWM  4  NONE
resource  PWM  5  NONE
resource  PWM  6  NONE
resource  PWM  7  NONE
resource  PWM  8  NONE
resource  LED_STRIP  1  A08
resource  SERIAL_TX  1  A09
resource  SERIAL_TX  2  A14
resource  SERIAL_TX  3  B10
resource  SERIAL_TX  4  NONE
resource  SERIAL_TX  5  NONE
resource  SERIAL_TX  6  NONE
resource  SERIAL_TX  7  NONE
resource  SERIAL_TX  8  NONE
resource  SERIAL_TX  9  NONE
resource  SERIAL_TX  10  NONE
resource  SERIAL_RX  1  A10
resource  SERIAL_RX  2  A15
resource  SERIAL_RX  3  B11
resource  SERIAL_RX  4  NONE
resource  SERIAL_RX  5  NONE
resource  SERIAL_RX  6  NONE
resource  SERIAL_RX  7  NONE
resource  SERIAL_RX  8  NONE
resource  SERIAL_RX  9  NONE
resource  SERIAL_RX  10  NONE
mixer  QUADX

mmix  reset

servo  0  1000  2000  1500  90  90  100  -1
servo  1  1000  2000  1500  90  90  100  -1
servo  2  1000  2000  1500  90  90  100  -1
servo  3  1000  2000  1500  90  90  100  -1
servo  4  1000  2000  1500  90  90  100  -1
servo  5  1000  2000  1500  90  90  100  -1
servo  6  1000  2000  1500  90  90  100  -1
servo  7  1000  2000  1500  90  90  100  -1
smix  reset

feature  -RX_PPM
feature  -VBAT
feature  -INFLIGHT_ACC_CAL
feature  -RX_SERIAL
feature  -MOTOR_STOP
feature  -SERVO_TILT
feature  -SOFTSERIAL
feature  -GPS
feature  -FAILSAFE
feature  -SONAR
feature  -TELEMETRY
feature  -CURRENT_METER
feature  -3D
feature  -RX_PARALLEL_PWM
feature  -RX_MSP
feature  -RSSI_ADC
feature  -LED_STRIP
feature  -DISPLAY
feature  -OSD
feature  -BLACKBOX
feature  -CHANNEL_FORWARDING
feature  -TRANSPONDER
feature  -AIRMODE
feature  -SDCARD
feature  -VTX
feature  -RX_SPI
feature  -SOFTSPI
feature  -ESC_SENSOR
feature  -FEATURE_ANTI_GRAVITY
feature  VBAT
feature  RX_SERIAL
feature  MOTOR_STOP
feature  FAILSAFE
feature  CURRENT_METER
feature  RSSI_ADC
beeper  GYRO_CALIBRATED
beeper  RX_LOST
beeper  RX_LOST_LANDING
beeper  DISARMING
beeper  ARMING
beeper  ARMING_GPS_FIX
beeper  BAT_CRIT_LOW
beeper  BAT_LOW
beeper  GPS_STATUS
beeper  RX_SET
beeper  ACC_CALIBRATION
beeper  ACC_CALIBRATION_FAIL
beeper  READY_BEEP
beeper  MULTI_BEEPS
beeper  DISARM_REPEAT
beeper  ARMED
beeper  SYSTEM_INIT
beeper  ON_USB
map  TAER1234
serial  20  1  115200  57600  0  115200
serial  0  0  115200  57600  0  115200
serial  1  64  115200  57600  0  115200
serial  2  0  115200  57600  0  115200
aux  0  0  0  1300  2100
aux  1  28  1  1550  2100
aux  2  1  1  1100  1550

mikeller, 2020-07-16:
@Raffa2s, @tomasris: Can you please test the target definition provided by @EachineRC?

TimelessNL, 2020-08-01:
@mikeller, I would like to test the target definition. But there is no unified target for the F3 right? I tried flashing the lagacy SPRacingF3EVO image and applying these resource mappings but none of the gyro/accel/baro sensors are detected. Or am I doing something wrong?

TimelessNL, 2020-08-01:
@EachineRC I think @mikeller ment the [Eachine E010S **Pro**](https://www.eachine.com/EACHINE-E010S-PRO-65mm-5_8G-40CH-800TVL-Camera-F3-Built-in-OSD-High-Hold-Mode-RC-Drone-Quadcopter-p-1179.html) which uses the LemonF3 and isn't compliant with Betaflight's license. You guys distributed a [customized Betaflight firmware](https://www.mediafire.com/folder/hm8r8p4nc8d0b/LEMON_F3) with a mediafire link that can be found in the manual

Did you post the correct resource map for the LemonF3?

mikeller, 2020-08-02:
@TimelessNL: Thanks for testing this. Due to flash space limitations, full resource configurability was never added to F3. In order to be able to test the above `dump` you'd have to use it to create a set of `target.mk`, `target.c`, and `target.h` files and then build a custom target.
Also, it is important to make a distinction between the SPRACINGF3, which uses USB over CP210x, and SPRACINGF3EVO which uses USB over VCP (which is much more common) - even everything else being the same, these targets will not be usable interchangeably.

TimelessNL, 2020-08-02:
Ah thanks! With your explanation it seems that the `LemonF3` uses USB over VCP since flashing the `SPRACINGF3EVO` build makes it able to connect with the configurator.

I will try to build a `LemonF3` custom target. But correct me if I'm wrong, back in 2018 you said a firmware image was not enough to restore a target definition but with the `dump` command while running the custom build it seems possible after all? Please don't understand me wrong, I would like to add support for the `LemonF3`. But just curious why the `dump` method was not evaluated back then.

Also `I2C` remapping is supported on the F3's right? I remember I used it once on a `F3-EVO-BRUSH V2.0` to get `I2C` on one of the `UART` (_RX2/TX2_) ports
[image]
And I guess `SPI` remapping is not supported and that's why the gyro and such don't work?

mikeller, 2020-08-02:
@TimelessNL: Damn, you are right - on F3 the dump does not contain timer / DMA assignments.

Re SPI pin assignments for the gyro, these can be done on F3. But since F3 is typically only built with the one driver for the gyro that is on the board, changing the pin assignment will not make it work if the gyro model is different.

@EachineRC: Can you please supply the timer / DMA assignments for this target, and the type of gyro that is used? It will be easiest if you share the `target.c` / `target.h` files used to build it.

TimelessNL, 2020-08-02:
> @TimelessNL: Damn, you are right - on F3 the dump does not contain timer / DMA assignments.

It's always unfortunate being right in these kind of situations 😢

>But since F3 is typically only built with the one driver for the gyro that is on the board

The gyro seems to be the `MPU6050` which is identical to the `SPRACINGF3NEO` and when I closely looked at all the `SPRACINGF3NEO` target pin mappings most of them seemed to match with the `dump` output on the `LEMONF3`. So I tested to see if the board would show some live with the `SPRACINGF3NEO` firmware on it, but unfortunately it didn't. The board booted with the green status `LED`  _slowly flashing -> quick burst of rapid flashes -> then static on_. Which seems to indicate a fault with the target definition? I flashed from the custom-build without full-erase hoping the resource maps would survive, and a 2nd time with a full-erase but neither one of them worked.
```
# status
System Uptime: 110 seconds
Voltage: 2 * 0.1V (0S battery - NOT PRESENT)
CPU Clock=72MHz, GYRO=MPU6050, ACC=MPU6050.n, BARO=BMP280
Stack size: 2048, Stack address: 0x10002000
I2C Errors: 0, config size: 2011, max available config: 4096
CPU:35%, cycle time: 259, GYRO rate: 3861, RX rate: 49, System rate: 9
```
The `LEMONF3` also includes an osd chip `AT7456E` which is also known as the `MAX7456` right? That on is also used in the `SPRACINGF3NEO`.