betaflight/betaflight #6833: SpeedyBeeF4 Softserial Telemetry not working on UART3

RipperDrone, 2018-09-23:
Wired up the R-XSR with latest SBUS firmware to SpeedyBeeF4 FC (latest Nightly 4.0.0 #1142). Connected telemetry wire from R-XSR to TX3 pad of FC. Remapped pin C10 (associated with TX3) to softserial 1 (Set SERIAL_TX 11 C10). Then set feature telemetry and softserial = ON, assigned Softserial 1 (in BFC ports tab) to Smartport.

Result: No telemetry on my Taranis X9D, regardless of settings (all permutations tried out) tlm_inverted and tlm_halfduplex. Sensor search finds no new sensors (except the std RSSI and RXBat which work without Smartport telemetry anyways)

Expected Behavior: Show telemetry values on taranis

Then I hardware modded the R-XSR (extra inverted Smartport wire soldered to it) and configured hardware UART Smartport --> works as intended

Anybody else noticed this or any idea what went wrong with my softserial attempt?

jflyper, 2018-09-23:
As always, we need your CLI `diff` output from your non-working configuration.

jflyper, 2018-09-23:
Nah, this one is easy.

`SPEEDYBEEF4` target does not associate PC10 with a valid timer entry.

Use S5 (PA15) or S6 (PA8) instead.

You can also use S7 (PB8) if you are not using LED strip on LED pad.

closed 2018-09-23 by jflyper

RipperDrone, 2018-09-23:
Wow, thanks that was a straight hit! :-)

RipperDrone, 2018-09-23:
Thank you! (I didn't upload the diff since I suspected something hardware related. It even didn't work straight from defaults, just entering the 3 lines I described in the post (softserial config). Now I know why :-))