betaflight/betaflight #6836: Broken gyro? (no sensor movement)

psraymon, 2018-09-24:
Hello,
I am a newbie and building my first quadcopter
The FC is an F3 with ACC, MAG and BARO sensors
SPRACINGF3 2.4.0
During the configuration in cleanflight, I first noticed that there was no gyro movement in the sensortab f i moved the FC (The line stays flat). The other sensors do move in the sensor tab. In the setup tab, if i move my FC, the model wil move correctly. But if i disable the ACC, MAG and BARO sensors the configuration tab, the model will not move.
Also, It will not arm. (the arm mode will not turn green if the switch is in the right position)
If I check status in CLI, the following comes up:

# status
System Uptime: 280 seconds
Voltage: 110 * 0.1V (3S battery - OK)
CPU Clock=72MHz
Stack size: 2048, Stack address: 0x10002000
I2C Errors: 0, config size: 2186, max available config: 4096
CPU:1%, cycle time: 1754, GYRO rate: 570, RX rate: 129, System rate: 9
**Arming disable flags: CALIB CLI**

From  the last line I make up that a sensor is still calibrating.
I switched off the MAG, BARO and ACC sensor, so it probably is the gyro sensor. 
It will not change, even if I let it rest for minutes.

Things I have tried:
-Change moron_threshold to different values;( 90, 100, 150, 200)
-Change the firmware version to an older version (from 2.4.0 to 2.3.2)
 -Reset all setting.

Nothing works. Is my gyroscope broken?
Or is there an other fix

Thanks for your help

jflyper, 2018-09-24:
@psraymon First thing you have to do is to copy this issue ticket on CleanFlight GitHub issues.

mikeller, 2018-09-24:
@psraymon:
- this is the GitHub project for Betaflight, not cleanflight;
- in general, GitHub is for questions / discussion around **development** of the firmware, not configuration / hardware issues. These are better discussed on RCGroups;
- from your description, it sounds like your gyro is defective or not recognised.

closed 2018-09-24 by mikeller