betaflight/betaflight #6839: Add "rescue unreliable now" OSD indication for risky flights

rrk1, 2018-09-25:
Hi All!

Currently we have a problem with trust to GPS rescue feature. Sanity check can unexpectedly revert to land/drop behavior. Generally, not a problem flying over a simple grass field, but there is not much need for rescue here. Let's say we are doing a nice long-range freestyle and dives in Norway, over mountains and fjords. Then failsafe happens, sanity check silently decides something is wrong and just drops copter into water or inaccessible terrain...

It will be nice to add some kind of OSD notification that rescue is unreliable now. Something like flashing exclamation mark near satellite count or so. That way pilot can decide to stop risky flight and be careful or just turn home, saving the craft.

pkruger, 2018-12-06:
Maybe this should be a warning, i.e. like "Low Battery" so you can't miss it.

pkruger, 2018-12-20:
PR: https://github.com/betaflight/betaflight/pull/7256

PR #7256 (2018-12-23): Add GPS rescue unreliable now warning in OSD
Fixes: #6839
closed 2018-12-23 by mikeller