betaflight/betaflight #6849: Omnibus F4 tumbles in nose dive after max throttle and then throttle down

DrBonesLCD, 2018-09-29:
**Describe the bug**
After a throttle punch and then throttle down, the quad nose dives in a gentle tumble to the ground unless you take control.

**To Reproduce**
Steps to reproduce the behavior:
1. Omnibus F4 flashed with betaflight 3.5.1 - default settings.
2. Full throttle up and then throttle down to minimum.
The quad starts nose diving towards the ground, sometimes with a gentle roll or yaw until throttle is applied and orientation corrected.

**Expected behavior**
Upon throttle max, quad flies straight up and upon throttle down, quad stabilizes in a level position.

**Flight controller configuration**
``` # diff

# version
# Betaflight / OMNIBUSF4 (OBF4) 3.3.0 Mar  2 2018 / 03:56:20 (177472b4f) MSP API: 1.37

# name

# resources

# mixer

# servo

# servo mix

# feature
feature TELEMETRY
feature AIRMODE
feature ANTI_GRAVITY

# beeper

# map
map TAER1234

# serial
serial 0 64 115200 57600 0 115200
serial 5 32 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1800 2100 0
aux 1 13 1 900 1700 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set acc_hardware = NONE
set mag_hardware = NONE
set baro_hardware = NONE
set rssi_channel = 8
set serialrx_provider = SBUS
set motor_pwm_protocol = DSHOT600
set small_angle = 180
set pid_process_denom = 2
set osd_vbat_pos = 2049
set osd_rssi_pos = 27
set osd_current_pos = 480
set osd_mah_drawn_pos = 2059
set osd_warnings_pos = 2538

# profile
profile 0

set p_pitch = 65
set p_roll = 55
set p_yaw = 85
set i_yaw = 50
set d_yaw = 25

# rateprofile
rateprofile 0

set roll_rc_rate = 150
set pitch_rc_rate = 150
set yaw_rc_rate = 120
set tpa_rate = 15
'''

**Setup / Versions**
 - Cicada AIO Omnibus F (30A ESC 4 in 1) on a Chameleon (220) frame. This also happens on HGLRC Omnibus F4 stack on the XJB145 frame;
 - FrSky SBUS XM+

**Additional context**
I upgraded the Cicada from BeF 3.3 to 3.5.1 when this behavior started happening. I switched back to BeF 3.3.1 and this behavior stopped.
I reproduced the same problem with an HGLRC XJB145, which had BeF 3.3 on it. Upon flashing it with 3.5.1 it had the same diving behavior as I had initially observed on the 220 size quad with the Cicada AIO FC.

etracer65, 2018-09-29:
Your diff shows Betaflight 3.3.

DrBonesLCD, 2018-09-29:
> Your diff shows Betaflight 3.3.

Yes - that was the original setup with no problem. I didn't bother with a diff on the BeF 3.5.1 as I didn't change any settings - all on default.

etracer65, 2018-09-29:
You really can’t apply a diff from 3.3.0 to 3.5.1.  For example, d_yaw = 25 is very bad on 3.5.  Redo your config from defaults.

DrBonesLCD, 2018-09-29:
Hmm...I left everything at default...it happens on two separate and different model quads - both flown on maiden at default 3.5.1 settings. Unfortunately, as I realized that it wasn't a motor or ESC problem on either quad...I flashed them to make them flyable again - So, I never bothered with a diff file. The CICADA AIO (Omnibus F4) doesn't have a blackbox and the HGLRC (Omnibus F4 - micro size)..well, that one was full - So, I unfortunately I can't help on that - but I tend to like to fly rather than tinker and tune for ever.

etracer65, 2018-09-29:
Well I’m not sure how you expect anybody to help. We need to see the actual config that exhibits the problem- not just what you say you setup. Maybe you made a mistake, but no way to tell.  It sounds like you can replicate the problem so try capturing a blackbox showing it for diagnosis.

It really sounds like you forgot to enable airmode.  But without the actual diff it’s impossible to tell.

DrBonesLCD, 2018-09-29:
I'll try to be better next time! After happening on two quads - I figured it must be common enough that perhaps there was an easy fix. I'll wait for the 3.5.1 to 'mature' a bit and then try again.

etracer65, 2018-09-29:
There are no reports of problems like this, 3.5.1 is quite mature. You likely just have a setup problem.

etracer65, 2018-09-29:
Closing this as it appears to be a configuration issue - likely that airmode wasn't enabled.

If you can replicate the problem, please reopen and provide the associated settings `diff` and blackbox log demonstrating the problem.

closed 2018-09-29 by etracer65

RipperDrone, 2018-09-29:
@etracer65 this clearly relates to https://github.com/betaflight/betaflight/issues/6018, doesn't it? It's a long known issue but without full understanding about the external forces causing it, not how to fix this...

jflyper, 2018-09-29:
Need the bb log to say anything conclusive.

etracer65, 2018-09-29:
@RipperDrone No, it doesn't.

> After a throttle punch and then throttle down, the quad nose dives in a gentle tumble to the ground unless you take control.

Sounds simply like airmode wasn't enabled.  But as noted, without a log or even a valid `diff` it's impossible to diagnose any further.

RipperDrone, 2018-09-29:
agreed, tumble to ground makes the difference...