betaflight/betaflight #6852: DALRC DLF7 target / wrong vbat scale / BB logging vibes

darksonny, 2018-09-29:
Hi guys.

Was just testing the new DALRC F722 dual gyro FC.

I have two issues:
1: vbat scale is off by a lot, it should be 156.
I'm using the matek vtx pdb hub, vbat pass-through tho.

2: BBL logging above 2khz is causing wiggles, vibes.
I switched BBL on/off while hovering, could not log with 4khz using 32/16 and ICM (second gyro)

My setup:
DALRC F722 (icm gyro active) 32/16 dshot1200
Wraith ESC (48khz, auto timing on)
2306, 2750kv motors on 5.1 props
5 inch setup

# diff

# version
# Betaflight / DALRCF722DUAL (DLF7) 3.5.1 Sep  8 2018 / 05:08:10 (d9fb5ca13) MSP API: 1.40

board_name DALRCF722DUAL
manufacturer_id 

# name
name F7XL5v3

# resources

# mixer

# servo

# servo mix

# feature
feature TELEMETRY
feature AIRMODE

# beeper
beeper -GYRO_CALIBRATED
beeper -ARMING_GPS_FIX
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -MULTI_BEEPS
beeper -ARMED
beeper -ON_USB

# beacon

# map
map TAER1234

# serial
serial 0 0 115200 57600 0 115200
serial 1 64 115200 57600 0 115200
serial 3 8192 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 1 1300 1700 0 0
aux 2 27 2 1700 2100 0 0
aux 3 13 2 1325 1725 0 0
aux 4 26 1 1700 2100 0 0
aux 5 35 3 1700 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_lowpass_hz = 90
set gyro_lowpass2_hz = 180
set gyro_calib_noise_limit = 148
set gyro_use_32khz = ON
set gyro_to_use = SECOND
set acc_calibration = 44,-13,-73
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1025
set max_check = 1850
set rssi_channel = 12
set rc_smoothing_type = FILTER
set serialrx_provider = CRSF
set blackbox_p_ratio = 64
set dshot_idle_value = 500
set motor_pwm_protocol = DSHOT1200
set motor_pwm_rate = 32000
set vbat_scale = 156
set ibata_scale = 175
set beeper_dshot_beacon_tone = 2
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 2
set osd_rssi_alarm = 50
set osd_cap_alarm = 1200
set osd_tim1 = 1280
set osd_tim2 = 3841
set osd_vbat_pos = 2466
set osd_rssi_pos = 2422
set osd_tim_1_pos = 2454
set osd_tim_2_pos = 2486
set osd_flymode_pos = 333
set osd_throttle_pos = 225
set osd_vtx_channel_pos = 377
set osd_ah_pos = 200
set osd_current_pos = 2434
set osd_mah_drawn_pos = 2402
set osd_craft_name_pos = 362
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 2409
set osd_avg_cell_voltage_pos = 76
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 392
set osd_disarmed_pos = 2378
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_stat_max_spd = OFF
set debug_mode = GYRO_SCALED
set vtx_band = 5
set vtx_channel = 3
set vtx_power = 5
set vtx_freq = 5732

# profile
profile 1

set dterm_lowpass_hz = 80
set dterm_lowpass2_hz = 160
set dterm_notch_cutoff = 0
set feedforward_transition = 10
set iterm_relax = RP
set throttle_boost = 8
set p_pitch = 43
set i_pitch = 65
set d_pitch = 26
set f_pitch = 100
set p_roll = 40
set i_roll = 60
set d_roll = 21
set f_roll = 80
set i_yaw = 60

# rateprofile
rateprofile 1

set thr_mid = 40
set roll_rc_rate = 95
set pitch_rc_rate = 95
set yaw_rc_rate = 113
set roll_expo = 10
set pitch_expo = 10
set yaw_expo = 10
set roll_srate = 82
set pitch_srate = 82
set yaw_srate = 82
set tpa_rate = 20
set tpa_breakpoint = 1050

nyway, 2018-09-29:
@darksonny 
About issues 1:Did you measure by  accurate multimeter adn which voltage point you measure?
If 24V in ,156 and 160 calculate have 0.6V difference,There shouldn't be such a big gap

darksonny, 2018-09-29:
I did measure on battery side with an accurate cell meter (multimeter verified already, not that far off) and I checked the OSD voltage. 
Battery was around 16.5v, my OSD was about 11.5v something.
I'm already discussing that topic with the DALRC engineers...

jflyper, 2018-09-29:
Are you saying battery meter scale of 160 gives 11.5V reading while 156 gives 16.5V?

etracer65, 2018-09-29:
> I'm already discussing that topic with the DALRC engineers...

If so then ask them to provide the values of the resistors used in the voltage divider on the board.  Then there will be no guesswork as to what the scale should be.

> 2: BBL logging above 2khz is causing wiggles, vibes.
I switched BBL on/off while hovering, could not log with 4khz using 32/16 and ICM (second gyro)

You're just causing too much CPU load trying to log at the high rates and this is affecting other parts of the system - like the gyro sampling and PID controller.  Log at a lower rate or reduce your gyro/PID loops.

nyway, 2018-09-29:
@darksonny  @etracer65 
I'm an engineer from DALRC.
The resistors was used 15K/1%  and 1K/1% in the voltage divider on the board.The scale was 16.

jflyper, 2018-09-29:
Just a side note.

I just found out that most flash chips supported by Betaflight can run at significantly faster clock than they are configured to. As the current flash driver does busy (software) writes, the faster clock may help logging at higher rates. I'm hoping to come up with a PR for this in a couple of days.

etracer65, 2018-09-29:
@jflyper I don't think logging/vibrations issue is about speed of the flash chip. I think it's related to CPU load.  Running 32/16, DSHOT1200, ACC on - CPU load is likely 35 - 40% at idle.  Throw in trying to log at 4K and it's probably too much load in flight.

nyway, 2018-09-29:
@jflyper 
Does the F7 and F4 target have different in ADC part?Like fillter?
The flash chip was W25Q128 from Winbond.

jflyper, 2018-09-29:
@nyway No, post ADC processing is the same. I mentioned `DEBUG_MODE = BATTERY` graph because it shows raw voltage value (after scaling) without filtering.

jflyper, 2018-09-29:
@etracer65 
> I don't think logging/vibrations issue is about speed of the flash chip. I think it's related to CPU load. Running 32/16, DSHOT1200, ACC on - CPU load is likely 35 - 40% at idle. Throw in trying to log at 4K and it's probably too much load in flight.

No and Yes. Current flash driver is CPU driven. If flash chip can be clocked faster, then less CPU is used. However, as I stashed my previous note, we can't make it faster 😭 

For the record, F405 running non-OC (168MHz) drives flash chip on SPI2 or SPI3 at 21MHz. With this clock, 4KHz sampling takes around 7% of the CPU time (~= 35usec per write every 500usec). 8KHz would double the figure.

Strangedave, 2018-10-19:
I have two of the same FC. DALRC F722, actually 3 now I bought one as a test rig because I was having a lot of trouble with one of the first two. This Is what I have on the power tab, and its working ok. I think I still need to work on the offset

Battery	0 V	
 
160 Scale 
10 Divider Value 
1 Multiplier Value
Amperage Meter	
 
131 Scale [1/10th mV/A] 
-500 Offset [mA]

If your interested in any of the other issues check this link. https://github.com/betaflight/betaflight/issues/6893

Or this one
https://www.facebook.com/groups/betaflightgroup/permalink/241151723247378/

nyway, 2018-10-19:
@Strangedave 
Hey,did you list what is the trouble about DALRCF722?And we will solve as soon as possible.

Strangedave, 2018-10-19:
I don't think the problem is you. I think I changed a setting in the cli unknowingly. Testing today. I'll let you know what happens in a few hours.

EmilWilkens, 2018-10-19:
I have the same problem with viggle/wiered behavior, but only happening when i go full throttle and let right after... feels very artificial and non intended... running DALRC F722/50A stack with F60 Pro V2... DSHOT1200... 

Any new findings on how to fix issue?

Cheers

Strangedave, 2018-10-19:
Still working on this. Flashed betaflight 3.5.2 last night to get back to the beginning. Set everything up through the configurator. I thought for a minute it was fixed. Wanted to make sure I was not copying code that was causing the problem. Ran a test flight and I still have the issue. Switched back to the mpu6000 gyro and ran another test. Problem gone. The sister Kwad still runs great at 32/16. It’s been off and on raining today, so my efforts are being slowed. I’m going to do a short test flight with the GoPro attached tomorrow just to see if the thing always dips towards the same motor. Suggestion from Joshua Bardwell. I will continue to post updates.  Although I’m afraid the root cause is going to be something simple and I’m going to look like a big dummy. There are three of us having issues though so maybe it’s an actual problem. 

Best regards 
Dave Goddard

> On Oct 19, 2018, at 2:13 PM, EmilWilkens <notifications@github.com> wrote:
> 
> Cheers

EmilWilkens, 2018-10-20:
Hi Strangedave, thanks for sharing. 
I look forward to hear back from you and maybe if nyway (DALRC Engineer??) could clarify or confirm if they are working on firmware fix for this issue? 
I am a complete noob in regards to FPV/Miniquads and this is my first build/drone ever and finished it two days ago. Before i started reading this thread i thought maybe it was just a setting i had done wrong in BF... 

You suggest that using the other gyro will remove the problem, can you ellaborate as to why this might not be an optimal solution or as good?

Cheers Emil

Strangedave, 2018-10-20:
When you installed the Flight Controller it was by default set up to use the first gyro. (there are two) on the DALRC F722 that is an MPU6000 which is kind of old technology. When I say that I think I have had that gyro on everything I have done, and I have been playing with this stuff for a little more than a year now. The new thing 6 months ago was to sample the gyro faster, the MPU6000 was only capable of 8000KHz. So people have been adding ICM in this case ICM20602 gyros to flight controllers in an effort to run faster loop times. I'm not going to go into depth here there is a lot of info on the web. Basically faster loop times feels more responsive. so if you turn on the second gyro and crank up your loop time to 32/16 you should get better performance to a point. The Kwad (specifically my Kwad's are prone to crashing) and after the motors are slammed into the ground violently a few times they tend to produce too much vibration for the ICM gyros, but will still work with the MPU6000. The Kwad I'm working on now is brand new and should not be producing those bad vibrations, so I'm trying to get back onto the ICM gyro, but having problems. We will have to see if I have the mechanical skills necessary to find the issue and fix it. Don't worry about being new, the cool thing in this hobby is that almost everyone is new. I have a post on the BETAFLIGHT facebook page https://www.facebook.com/groups/betaflightgroup/permalink/241151723247378/  and I will continue to update both as I try different fixes, but nothing has worked so far. There is a DALRC engineer reading this thread also so I still hope the group will find a fix. 

P.S. Get onto YouTube and watch every Joshua Bardwell video you can. Lots of great info from that man.

EmilWilkens, 2018-10-20:
Thanks for the comment bud. Let us all know if you come closer to a conclusion. I will look further into it during next week. 
My quad is all new parts and has not encountered a single crash yet, everything is soft mounted, so it should be all the best intentions for working as intended also. 

I do already watch most og his vids :D

Strangedave, 2018-10-22:
Still stumped. Tried 3.5.2, here is a video of that flight. https://youtu.be/Pwn2lXsKGjA 
and here is the exact same parts on a second Kwad I built these 1 day apart, and yes I used the same settings initially. Copied the diff from the working rig to the one that is acting crazy.
https://youtu.be/nIfY_LmHVCI

Strangedave, 2018-10-23:
Good morning,

 

I have tested basically everything, and have not found the issue. I
originally was sure it had a bad gyro, because of how the black box log
looked. I ran another flight and the log is very strange, everything looks
pretty smooth and then the thing just jumps. Can you please look at this log
and tell me if you think the gyro is bad on this Kwad. Please keep in mind
that I have replaced the Flight controller once already, so if the Gyro is
bad than I have 2 of these with a bad Gyro. I think that is pretty unlikely,
but I'm out of ideas at this point. This is a link to the flight video this
log is from https://youtu.be/Pwn2lXsKGjA.
<http://s.bl-1.com/h/crTBHMYg?url=https://youtu.be/Pwn2lXsKGjA.> 

 

Thanks In Advance.

 

Dave G

<mention@noreply.github.com>
BB logging vibes (#6852)

 

<https://github.com/betaflight/betaflight/issues/6852#issuecomment-431265883
ZVKy3Koks5umXnEgaJpZM4XAcRt> .
<https://github.com/notifications/beacon/Ak1mkK7QAr6VSTCvYjDNe6DH0b6Cho-jks5
umXnEgaJpZM4XAcRt.gif> 

  <http://s.bl-1.com/7R9mVp5.gif>

Strangedave, 2018-10-23:
Still working on this, however I'm thinking you might just need some help
tuning. Do you already use plasamtree pid analyzer. If not its worth your
time. I always use that to get to 0 then tune to my liking. This is a video
of how to read the data. https://youtu.be/a0E1ua_ASCQ.
<http://s.bl-1.com/h/crVsfYn8?url=https://youtu.be/a0E1ua_ASCQ.> 

BB logging vibes (#6852)

 

<https://github.com/betaflight/betaflight/issues/6852#issuecomment-431602860
GKQ9Q6Pks5um186gaJpZM4XAcRt> .
<https://github.com/notifications/beacon/Ak1mkCR2QfflgDsdDOHArN4qkwefaVzVks5
um186gaJpZM4XAcRt.gif> 

  <http://s.bl-1.com/7RNpXog.gif>

EmilWilkens, 2018-10-23:
Dear Dave,

I really do hope that someone is able to Help you with these problems..!

I can tell you that my problems are NOTHING like what I see in your video. The only wiggle/strange behavior that i encounter, is when I give it full throttle/longer punches, and then let go of the power real quick. Then my quad does wiggle just a little bit and feels like it has trouble finding balance/decent in a fluid and natural way.
Only when the throttle is skamme to zero-pct... 
I dont know if the above is understandeble to anyon at alle, but i will try to capture DVR footage of it soon. 

Cheers Emil

Strangedave, 2018-10-30:
So finally replaced the other two motors. Now it has a new set of motors. Still twitching. https://youtu.be/rCFJWtGcPTQ
 and slomo https://youtu.be/94eQXlxDq8I.

darksonny, 2018-11-06:
I've got two builds with the same DALRC F7 now. It's a damn awesome FC in my opinion, I like the flight characteristics I got after tuning. BUT the BIG BUT... I still couldn't get BBL working.
As soon I flip the switch, my cycle time jumps from pretty consistent 25-35 to a twitchy 35 up to 150 for a split second, which is pretty much the issue that's causing the quad to jiggle.

I wanna say it again, my both FC fly great, no Gyro issue on ICM at all. I'm running 32/16 on Dshot 1200.
The issues and jiggles just happens when I use BBL.

Strangedave, 2018-11-06:
I’m going to turn bb off and fly a test pack. I wonder if I’m having the same issue and never turned bb log off in my testing. I also have 2 rigs. One runs great logging at 1khz. The other is a hot mess. I’ll get back to you on this ASAP.

Strangedave, 2018-11-06:
I can not fully confirm the fix, but it definitely was not visibly twitching. The weather here is really not great for an actual test flight. As soon as the rain stops I will head over to the park with the gopro and see if its for sure good. Here is the video from the test hover. https://youtu.be/LJRGE502ud0  If you compare this to my previous video I think you can assume that turning off black box fixed the problem. Motors were cold to the touch. After my last flight you could smell the motors they were so hot no need to touch them.

Strangedave, 2018-11-06:
Problem seems mostly solved, most of the weird movements were me shaking the sticks, because its 32 degrees real feel. Actual temp is 40 with 20-30 mph winds. There are some unexplained bumps, but I need a better day to fly for testing. Unfortunately; I'm not sure we will get a better day until the spring. here is the video. https://youtu.be/kgw5wQYUYvk . Motors were just slightly warmer than ambient temp. Now we just need to figure out how to fix the BBL problem.

darksonny, 2018-11-08:
did some of you actually try to log with the MPU gyro? it actually looks like the jumps in cycle time starting the log is just showing when using the icm gyro

Strangedave, 2018-11-08:
I have run this rig on the MPU6000 gyro. It was good enough, but I started flying on 32/16, and I think because of that 8/8 feels strange to me. I’m sure I could get used to it over time, but I have had more crashes on 8/8 than anything. To me it feels like my rates are different, and then timing of flips and rolls aren’t what I expect. I also have 2 of these that are set up almost exact copy except for the camera. The first one I built will log at 1khz with no problems. So for now I’m just using the tune from the working rig on the one with no black box. I’m not good enough yet to tune with out help from black box.

Strangedave, 2018-11-08:
The answer is yes. I have logs from the MPU6000 gyro. Worked fine, and I think cou usage was right around 4%

Strangedave, 2018-11-08:
I still have some jumpyness video is here. Its hard to see, but the rig was sputtering the entire time. There was a big bump at 00:31. So better but not fixed completely by turning BBL off. https://youtu.be/ZCA0ZVUgL_M

darksonny, 2018-11-09:
I was just trying around... overclocked cpu, different sampling rates, Accelerameter on/off.
It all comes down to 32khz mode! As soon as you flip the switch, the FC can't log without introducing additioinal cycle time blips and therefore motor noise!

Would be great if the DALRC engineer here could give a statement...

Strangedave, 2018-11-09:
Agreed!!! I would really like someone from DALRC to at least say something also. I don’t need a solution, but maybe “don’t worry, we are on it.” or something. I can wait the weather here is not really flyable anyways.

Strangedave, 2018-11-09:
Nice to know you tried that already. That was going to be my next step.

nyway, 2018-11-09:
@darksonny 
@Strangedave 
We are still trying to figure it out about BBL issue with hardware or software,it doesn't look like a hardware failure.With blackbox logging on the load seems to be getting too high to run 32k,too high occupancy.

Strangedave, 2018-11-09:
Thanks for the update, patiently waiting. please ask if I can be of any help..

darksonny, 2018-11-12:
I just tested another brand F7 running 32k and BBL. No issues at all... DALRC?

nyway, 2018-11-12:
Okay.Sounds great :)

Strangedave, 2018-11-26:
here is the diff from that rig

# Betaflight / DALRCF722DUAL (DLF7) 3.5.3 Nov 14 2018 / 23:23:06 (61d5abf00) MSP API: 1.40

board_name DALRCF722DUAL
manufacturer_id 

# name
name WIN 5R

# resources

# mixer

# servo

# servo mix

# feature
feature TELEMETRY

# beeper
beeper -GYRO_CALIBRATED
beeper -DISARMING
beeper -ARMING_GPS_FIX
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -ON_USB
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSED

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# serial
serial 20 1 57600 57600 0 115200
serial 1 16384 115200 57600 0 115200
serial 3 32 115200 57600 0 115200
serial 4 8192 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 1 900 1125 0 0
aux 2 2 1 1300 1700 0 0
aux 3 13 2 1300 1700 0 0
aux 4 28 1 1300 2100 0 0
aux 5 31 3 1900 2100 0 0
aux 6 35 2 1700 2100 0 0
aux 7 39 3 1300 1700 0 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_lowpass_hz = 125
set gyro_calib_noise_limit = 148
set gyro_use_32khz = ON
set gyro_to_use = SECOND
set acc_calibration = -16,-31,-8
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1040
set max_check = 1960
set rssi_channel = 16
set rc_smoothing_type = FILTER
set fpv_mix_degrees = 30
set min_throttle = 1040
set use_unsynced_pwm = ON
set motor_pwm_protocol = DSHOT1200
set motor_pwm_rate = 32000
set failsafe_switch_mode = KILL
set vbat_scale = 162
set ibata_scale = 131
set yaw_motors_reversed = ON
set small_angle = 35
set pid_process_denom = 2
set osd_units = IMPERIAL
set osd_warn_batt_not_full = OFF
set osd_warn_esc_fail = OFF
set osd_warn_core_temp = OFF
set osd_cap_alarm = 1500
set osd_vbat_pos = 2081
set osd_rssi_pos = 2094
set osd_tim_2_pos = 2103
set osd_flymode_pos = 2382
set osd_vtx_channel_pos = 212
set osd_crosshairs_pos = 2254
set osd_current_pos = 2086
set osd_mah_drawn_pos = 2097
set osd_craft_name_pos = 2445
set osd_warnings_pos = 2410
set osd_stat_max_spd = OFF
set osd_stat_min_batt = OFF
set debug_mode = GYRO_SCALED
set vtx_band = 2
set vtx_channel = 3
set vtx_power = 3
set vtx_freq = 5771
set vcd_video_system = NTSC

# profile
profile 0

set dterm_notch_cutoff = 0
set vbat_pid_gain = ON
set feedforward_transition = 10
set smart_feedforward = ON
set iterm_relax = RP
set p_pitch = 49
set i_pitch = 51
set d_pitch = 32
set f_pitch = 95
set p_roll = 45
set i_roll = 46
set d_roll = 29
set f_roll = 95
set p_yaw = 70
set i_yaw = 53
set f_yaw = 90

# rateprofile
rateprofile 0

set roll_srate = 80
set pitch_srate = 80
set tpa_breakpoint = 2000

dadoribeiro, 2018-12-20:
I have a logging issue with the F722. With 32k Gyro and logging enabled it twitches heavily even in a hover; if I disable logging it works fine, but motors are somewhat hot. On 8k gyro (MPU) logging does not induce the jitters, but motos are still a little hot from a hover. The weird thing is that the MSC command that is supposed to work on a MacOs now does not work and using the Configurator on a Mac takes about an hour for a full log. I have to mount manually through the Terminal using the "diskutil readOnly mount" command for it to work. I think it is a hardware issue but maybe it could be addresses with a software fix.

closed 2019-01-26 by stale[bot]