betaflight/betaflight #6857: Autotune

crackcomm, 2018-09-30:
Hi, I am pretty new to betaflight source code and flying in general but I am interested if there is something in betaflight that would account for different props and motors so you could autotune PIDs for each quad so you could achieve more stable results. Is there something for that already and if not could you point out in the code where that would have place? Thank you.

etracer65, 2018-09-30:
No, there is no auto-tune capability. There have been various attempts over the years but they've all been removed because none ever actually worked.

jflyper, 2018-09-30:
IIRC, Optune PID calculator and associated procedure worked fine for basic PIDs, but that was years ago, and the theory behind the scheme has been deleted together with OP wiki deletion.
We can still take a look at write-ups and YT videos to see what it was like.

DaveFPV, 2018-09-30:
In his last video, Drone Mesh talks about his ongoing project to make a plugin system to learn and dynamically change PIDs by analyzing the quad behavior, relatively to the conditions (windy day, etc.).

DaveFPV, 2018-09-30:
Don't forget about computational power needed to make such demos at a certain speed. But for sure, conventional "fixed" PIDs can't handle all the situations. It can't by design in a complex nonlinear and multi factors system. That's why there a lot of tweaks introduced to compensate for one matter or another, but it becomes probably less reliable with bad interactions possible. There are other ways to enhance PID controller itself with more "intelligence" such as fuzzy logic PID controller for instance, but someone with this knowledge has to dig into it, assuming the MCU has the power needed to do that quickly.

crackcomm, 2018-09-30:
I think that could somehow be a tight fit into current feedback loop.

I'm looking forward to flight controllers that would allow us running LSTMs.

mikeller, 2018-10-01:
@crackcomm: I think the aspect that people are overlooking when trying to extrapolate possibilities from D'Angelo's video for Betaflight or other on-board flight controllers is the availability of good data. Currently, a lot of the complexity of the PID controller comes from the inherent noisiness of the gyro input data. I would think that if this can be augmented with much less noisy data from 3D high speed cameras and image processing then implementing an efficient PID controller and associated tuning will be much easier.
This will apply even more if additional noise is introduced by damaged propellers.

crackcomm, 2018-10-03:
@mikeller I could not agree more.

I am yet to test and debug and see how much of that is really a gyro noise and if I can adjust acceleration to thrust loop because I think it should be fast enough even on slower processors.

closed 2018-11-09 by stale[bot]