betaflight/betaflight #6859: REQ: Pilot nickname.

mixMugz, 2018-09-30:
For now we have only field for configuring craft name. It's good for identify your quad in DVR.
But some people (like me) used this field for nickname (help in race flights to identify pilot). If i have 3 different quads it some time i have trouble to identify each quad when i looking my dvr.

My idea:
- add one more field "nickname" in addition of "craftname"
- add option to show in OSD field "nickname"
- add option to show field "craftname" in end flight screen

This will solve the problem with the identification of the quad and the display of the pilot's nickname.

wx4cb, 2018-10-02:
I usually just have a moniker for m I'll ne eg, wx4cb-IQXL, WX4CB-IQ5 for example, because legally you have to have your ham call to comply with your license which you do have :)

hydra, 2018-10-02:
craftname != pilot name. 

IIRC craftname also appears if Blackbox logs and helps identify log files too.

I don't see it being too much of an issue having separate craft name and pilot name fields as they are used for different purposes.

FieserKiller, 2018-10-02:
as most cams come with osd nowadays you can display craft name in betaflight and the pilot name in your camea osd..

wx4cb, 2018-10-02:
i'm not saying it wouldn't be handy, but IMHO with what @FieserKiller said and the fact that there's enough space to put it in the craft name, i don't see a reason for it. But i'm not a dev, i'm just giving my 2 pence worth :)

smoriarty21, 2018-10-13:
@mikeller @etracer65 I've got this one working, I need to do a bit of cleanup and testing and I need to work out getting it on the OSD and then I will drop a PR in here

smoriarty21, 2018-10-13:
@mikeller @etracer65 I have made a first go at this one, let me know what you think

https://github.com/betaflight/betaflight/pull/6924
https://github.com/betaflight/betaflight-configurator/pull/1215

mixMugz, 2018-10-13:
Also possible to add switch for Craft Name and Pilot name on Flight Statistics in OSD?

closed 2019-02-02 by mikeller