betaflight/betaflight #6860: BF 3.5.1 // SP RACING F3 - Can't enable GPS in configuration tab

johnlenfr, 2018-09-30:
1. 
#Can't enable GPS in configuration table

2. 
Flash SPRACING F3 with fw 3.5.1
Enabled GPS on Port UART2
Enabled GPS in configuration tab but when reboot gps is not activated

3. 
FC: SPRACING F3 Deluxe with integrated magnetometer
GPS on UART 2
Smartport on Softserial 1
Receiver : FrSky R9 Slim (SBUS on UART 3)
LED Strip

4.  
```
# diff
# Betaflight / SPRACINGF3 (SRF3) 3.5.1 Sep  8 2018 / 05:52:39 (d9fb5ca13) MSP API: 1.40

feature -RX_PPM
feature RX_SERIAL
feature SOFTSERIAL
feature AIRMODE
beeper -GYRO_CALIBRATED
beeper -DISARMING
beeper -ARMING
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSED
serial 1 2 115200 57600 0 115200
serial 2 64 115200 57600 0 115200
serial 30 32 115200 57600 0 115200
aux 0 0 0 900 1300 0 0
set acc_calibration = -23,105,87
set mag_calibration = -222,527,826
set serialrx_provider = SBUS
set blackbox_device = NONE
set min_throttle = 1120
set use_unsynced_pwm = OFF
set motor_pwm_protocol = ONESHOT125
set motor_pwm_rate = 480
set align_board_yaw = 90
profile 0

set dterm_notch_cutoff = 0
set anti_gravity_mode = STEP
set anti_gravity_threshold = 350
set anti_gravity_gain = 1000
rateprofile 0

set roll_expo = 30
set pitch_expo = 30
set roll_srate = 80
set pitch_srate = 80
```

Andi252, 2018-09-30:
Hi johnlenfr,

GPS is not part of targets which have 256k flash or less.

All F3 board only have 256k flash, F4 have either 512k or 1024k. F7 between 512k and 2048k.

johnlenfr, 2018-10-02:
Thanks @Andi252 

Issue can be closed then

closed 2018-10-02 by johnlenfr