betaflight/betaflight #6866: STM32F405 bootloader for fresh new chip

supiiik, 2018-10-01:
Hi, I have bought blank STM32F405 chip, but it seems, that it's without DFU bootloader, because soldered on the FC is not communicating over USB. I have successfully flashed HEX over st-link, but I think, that the bootloader is missing there. Is there a way, how to flash the proper bootloader bin/hex there? Thanks

wind0r, 2018-10-01:
DFU is build into the MCU there is no way to flash it, but also there is no way to remove it ;)

This just sounds like some driver issue.

mikeller, 2018-10-01:
@supiiik: The boot loader is in non-erasable ROM, as @wind0r says. But without the firmware properly installed you'll have to use the boot pins to boot into the boot loader, as the soft reboot method used by configurator requires firmware support. Try using the boot pins to get into the boot loader, then using configurator to install the firmware with 'No reboot sequence' enabled.

supiiik, 2018-10-01:
boot pins not working too. firmware is already in the MCU, because I can see, that the LED on the FC is signalizing some activity. As I said, I have already flashed that MCU by st-link over st-link utility. 
When the board is connected with USB, nothing is happening. Windows doesn't even recognize the device. When I try it with boot pins, then windows recognize "unknown device" after a while. I was not able to install DFU driver via zadig or impulseRC_driver_fixer. Seems like a bootloader is really not there...

jflyper, 2018-10-01:
I've never seen one without one. It's a driver problem or USB is broken.

supiiik, 2018-10-01:
It's new MCU from ali https://www.aliexpress.com/item/1PCS-STM32F405-STM32F405RGT6-QFP64-Original/32866720336.html?spm=a2g0s.9042311.0.0.75d04c4dcAqVQH

mikeller, 2018-10-01:
Could be that the USB wiring isn't right then. Or the clock frequency isn't right for  USB (shouldn't affect the boot loader mode though). Hard to judge what is wrong without seeing the schematic.

supiiik, 2018-10-01:
The FC is CLracingF4s, I had broken F4, so I bought new MCU. As this FC doesn't have SWDIO/SWCLK, I have soldered the MCU to old omnibus FC, flashed and resoldered to the CLR. On the both FCs it was same. It's flashed properly because I can see the activity via LED, but the USB is not communicating.

RicoKirstein, 2018-10-01:
If it flashes on the omnibus but not on the CLracingF4s, your boards are probably broken and not the MCU.
Check if you have a short on any power rail, I actually had three OSD chips short the 5V rail on different boards which resulted in the same behavior you're describing.

jflyper, 2018-10-02:
@supiiik 
> I have soldered the MCU to old omnibus FC, flashed and resoldered to the CLR.

Oh, then you have your drivers okay and boot loader is there.

And you have at least one solder bridge or cold joint.

I'm closing this 😩

closed 2018-10-02 by jflyper