betaflight/betaflight #6876: OSD/Headless Mode Warning

buzkill420, 2018-10-03:
Just want to start by saying "I love headless mode for line of sight flying", but it's gotten to the point where i don't want it on my switches anymore due to accidentally flying with goggles with headless mode on. The second you Yaw, you loose complete control of the drone and crash. 

I would like to see betaflight add an OSD warning option for when HEADLESS mode is activated, so that when you put on your goggles, it's obvious you're still in headless mode and should switch it off BEFORE takeoff.

Thank you

timman2er, 2018-10-03:
Agreed. Should the warning display before arming or always? Personally I'd go for the second one as I think no one flies headless with goggles.

buzkill420, 2018-10-03:
I agree with the second option as well.

etracer65, 2018-10-03:
It should be displayed in the flight mode OSD element (Flymode) - not a separate new "warning". I'll work something up.

timman2er, 2018-10-03:
agreed that should be added in flymode, but I think that if a user does this it's near 100% an issue and there should be a warning.

mikeller, 2018-10-03:
Agreed that the flight mode element is a good place to have it - but this will force pilots who don't already have this element active to activate it only to cover a condition they never want to fly FPV with in the first place.

I think it should be added as a warning in the warning element as well - I can't see a reason for not having this on all the time when headless mode is on - in this condition flying FPV is pretty much impossible.

etracer65, 2018-10-03:
Already working on it.  The only question is what priority should the "HEADFREE" warning have? Do we want it "in your face" and override all other warnings or low priority so that all other warnings supersede it?  I'm leaning towards the latter because otherwise it would mask other arming disabled reasons.

etracer65, 2018-10-03:
The current priority for warnings is:

1. DShot beacon arming delayed
2. Failsafe
3. Battery critical (LAND NOW)
4. MCU core temp alarm
5. ESC failure alarms (rpm, temp, current)
6. Crash-flip active
7. Most severe arming disabled reason
8. Low battery
9. RC smoothing not initialized
10. Battery < Full
11. Visual beeper

I think the HEADFREE warning should go between 8 & 9.

etracer65, 2018-10-03:
> I am one of these pilots. FLYMODE is not activated in my OSD ( i always fly in Acro) and therefor having HEADLESS in flymode does not help my issue

Except that you don't always fly in acro, you sometimes fly in HEADFREE mode. So one could argue that you *should* have the flymode activated in the OSD.  But I'm adding it to warnings as well.

However the next line of argument will be "I don't have warnings enabled...".  Sorry, can't help you there. :)

mikeller, 2018-10-03:
@etracer65: I would put the headfree warning between 3 & 4. Reasoning: Users might risk 'a quick test round' even with a temperature / ESC / low battery warning, and these will hide the headfree warning if it's below them. This will result in a crash.

Re arming disabled: If headfree is enabled, there should not be a warning for this, as not arming is desirable _if and only if_ the pilot is flying FPV and therefore sees the OSD at the time.

etracer65, 2018-10-03:
@mikeller I didn't mean to imply add an arming disabled reason and prevent arming.  No way we could do that as we don't know if the pilot is flying FPV or LOS.

I meant that placing the warning above item 7 will prevent the arming disabled reason from displaying and give `HEADFREE` a higher priority. So if there was an arming disabled reason there would be no way to see it if HEADFREE mode was enabled.  Not necessarily saying this is bad, just pointing it out as a potential "why won't my quad arm?" problem.

mikeller, 2018-10-03:
@etracer65:

Agreed on that. But I think the risk of crashing because you take off with headfree on on FPV because that warning was hidden by something that you chose to ignore outweighs the inconvenience of not being able to see why you can't arm in a situation  where not arming is probably the right thing to do.

Also, the not being able to see the arming disabled reason also applies to all of the other warnings above it, maybe we should introduce an 'interrupt priority' for arming disabled that applies only when arming (in a separate pull request).

PR #6877 (2018-10-06): Add HEADFREE indicator to the OSD flight mode and warnings display elements
Fixes #6876 

Provide multiple mechanisms to alert the pilot that he is attempting to fly in HEADFREE mode while in FPV - which will likely result in disorientation and loss of control.

Flight mode will be displayed as `HEAD`
Warning will be displayed as `HEADFREE`
closed 2018-10-06 by mikeller