betaflight/betaflight #6878: 3.5.1 Telemetry not working for Spektrum

paulboes73, 2018-10-04:
**Describe the bug**
Telemetry is not working for Spektrum SPM4649T receiver in 3.5.1 version of Betaflight.  However, in 3.4.1 Telemetry works fine.

**To Reproduce**
Load 3.5.1 firmware the FC.
In the configuration Tab under receiver settings select the following.
Select "Serial-based receiver  (SPEKSAT, SBUS, SUMD) for Receiver Mode
Select "SPEKTRUM2048/SRXL" Serial Reciver Provider

Other Features select Telemetry

Save and Reboot

**Expected behavior**
In the Spektrum radio I should see sensors when selecting Telemetry and select Auto-Config to find the sensors available.

Currently no Telemetry is available in 3.5.1 version, but works fine in 3.4.1

flint723, 2018-10-04:
Known problem, though unfortunate.  Please revert to 3.5.0 or if you are game,  try latest master 4.0.

AndersHoglund, 2018-10-04:
And there is a quickfix waiting: #6743 
This is already on master, but not picked to 3.5.x_maint yet.

closed 2018-11-10 by stale[bot]