betaflight/betaflight #6881: BF4.0.0 - Onboard Flash not working

TheIsotopes, 2018-10-04:
since #6868 the onboard flash on my MatekF405 Mini is no more working :(

@jflyper pls can you see what's wrong

jflyper, 2018-10-04:
#6868 has nothing to do with onboard flash...

[image]

TheIsotopes, 2018-10-04:
maybe, but `build 1164` have no probs with onboard flash ... with `build 1165` the onboard flash are no more working ... `waiting for data` is shown and fc hangs up
[image]

jflyper, 2018-10-04:
Oh, okay. It doesn't reproduce unless the onboard flash is selected for logging (MKF4 has both sdcard and onboard flash configured). Let me look into it.

jflyper, 2018-10-04:
Let me see... It must have something to do with SD card status reporting...

TheIsotopes, 2018-10-04:
I just tested it ... selecting the onboard flash for logging works again 
but if im using `status` in cli the fc hangs up again and the flash is not listed

[image]

jflyper, 2018-10-04:
Yikes!!! How about this one?
[betaflight_4.0.0_MATEKF405-prevent-getmetadata-from-being-called.hex.zip](https://github.com/betaflight/betaflight/files/2447682/betaflight_4.0.0_MATEKF405-prevent-getmetadata-from-being-called.hex.zip)

TheIsotopes, 2018-10-04:
tested again and now all is working perfect ... thx 👍

TheIsotopes, 2018-10-05:
i closed here now ... if #6882 is merged it will fixed this issue

PR #6882 (2018-10-06): SDCARD: Prevent msp from calling sdcard_getMetadata when card is not ready
This surfaced as a byproduct of going VTable oriented SPI/SDIO dispatching.

Fixes: #6881.
closed 2018-10-05 by TheIsotopes