betaflight/betaflight #6883: Betaflight 4.0 ; Mattek 405 ; gyro yaw axis is -90° off.

eyegorefpv, 2018-10-05:
Betaflight 4.0 ; Mattek 405 ; gyro yaw axis is -90° off.

mikeller, 2018-10-06:
This is by design - the way targets will be defined in 4.0 means that different orientations for the different gyro types used on versions of this target cannot be preconfigured in the target, and have to be adjusted as part of the configuration.

The configurator that will be released alongside 4.0 will have the capabilities required to deal with this without the need for user interaction.

eyegorefpv, 2018-10-06:
Hi,  Michael
thanks for the reply
I am new here!😉

closed 2018-11-12 by stale[bot]

eyegorefpv, 2018-11-26:
Hey,

I just signed the petition "Nations Unies: Contre la corruption judiciaire,
pour que les juges appliquent les lois" and wanted to see if you could help
by adding your name.

Our goal is to reach 1,971 signatures and we need more support. You can
read more and sign the petition here:

http://chng.it/MLycFj7VLW

Thanks!
Richard

Mozzzzi, 2019-04-25:
Hey Folks. For completeness for us not-so-tech-savvy. I have a Matek F405-AIO. The settings I required to fix this in BF4.0.0 are as below. Now all good. Thanks.
In BF Configurator Config screen:

Yaw Degrees = 0 degrees
Sensor 1 Yaw CW90 degrees
This translates to the following CLI Command settings
o	set gyro_offset_yaw = 0
o	set gyro_to_use = FIRST
o	set align_board_roll = 0
o	set align_board_pitch = 0
o	set align_board_yaw = 0
o	set gyro_1_sensor_align = CW90
[image]