betaflight/betaflight #6884: No receiver movement in betaflight 3.5.1 with revolt and dsmx receiver

moosestang, 2018-10-05:
I can't get my receiver to work on uart 3 on the revolt.  Works fine with raceflight or betaflight 3.2.5.  I have not tried moving to another uart.  Initially tried the betaflight 4.0 beta and it didn't work so worked my way back.

jflyper, 2018-10-05:
REVOLT target was renovated at the end of march. Before, it was a sub target of REVO target, but it is now an independent target.

You should probably enter CLI command `set serialrx_halfduplex = ON` to make your TX-pad connected serial receiver to work.

https://github.com/betaflight/betaflight/wiki/REVOLT-(V1-&-V2)#revolt-v1-sbus-configuration-after-v340

For further assistance, please consider using other means of communication.

closed 2018-10-05 by jflyper

moosestang, 2018-10-05:
Ah! set serialrx_halfduplex = OFF worked!  I'm using uart 3, not 1