betaflight/betaflight #6885: 32k gyro 100 percent issue.

mchaos, 2018-10-05:
Ever since 3.5 using 32k gryo causes 100 percent CPU.

When is this being addressed?

fujin, 2018-10-05:
* For F405, use 32/16/16 with cpu_overclock 216mHz.
* For F722, use 32/32/32 (DShot1200) with cpu_overclock 240mHz.

Otherwise, see above.

mchaos, 2018-10-05:
> What flight controller and target? 32/32 will likely never be possible on
> F4. Some F7’s can do it with over clocking.
>
> Flight performance improvements are using more processing power and are
> more value then high PID loop rates.

Fpvmodel betaflight f7

Already over locked.

Already over locked.
No issues on 3.4

mchaos, 2018-10-05:
What would be the best performance. Single 32k gyro or dual 8k?