betaflight/betaflight #6888: Compiling new firmware

access3424, 2018-10-06:
I'm compiling some custom firmware but it keeps coming out as betaflight_4.0.0_Betaflightf3.hex instead of the target and version that I'm selecting. Also the file size is a little off from the released firmware. What do I need to change?

access3424, 2018-10-06:
i am compiling with Cygwin with gcc arm 5.4. I am using the command make target=omnibus but the result is always betaflight_4.0.0_betaflightf3.hex.
If I edit the makefile  on line 19 into "TARGET    ?= OMNIBUS" i get a betaflight_4.0.0_OMNIBUS.hex file but still it is not the same as we have in the releases.

McGiverGim, 2018-10-06:
Write the target with all capital letters I think.

timman2er, 2018-10-06:
"make OMNIBUS" works for me.

etracer65, 2018-10-06:
`make TARGET=OMNIBUS` if you only want to generate the hex.  `make OMNIBUS` generates a .bin and .hex

access3424, 2018-10-06:
what would i need the bin file for?

access3424, 2018-10-06:
Yes i know this, the problem is the hex file i get doesn’t work properly and i haven’t even touched the source code yet.

jflyper, 2018-10-06:
@access3424 
The command to execute is `make TARGET=TARGET_NAME`.

For you to generate versions other than 4.0, you either have to install `git` revision control system and `checkout` your desired version.

For further assistance, please use other channels of communication. Thank you in advance.

jflyper, 2018-10-06:
EDIT
Generated binary (and hex) files may not agree exactly in size with ones in distribution, as there is a date string embedded in these files that may differ in size.

closed 2018-10-06 by jflyper