betaflight/betaflight #6892: GPS for F3 FCs

Raumpilot, 2018-10-06:
I am using the TBS Colibri in my midrange Copter.
It would be great to use GPS as well.

jflyper, 2018-10-06:
Due to flash storage shortage, GPS support has been dropped from all F3 targets in favor of flight performance code. It is unlikely to be reinstated in standard distribution in a close future.

closed 2018-10-06 by jflyper

mikeller, 2018-10-07:
@Raumpilot: One option that is left is to build your own custom version of the firmware, with things you don't need disabled, and GPS enabled:

https://github.com/betaflight/betaflight/blob/master/docs/Customized%20Version.md
https://github.com/betaflight/betaflight/wiki/Building-Betaflight-on-Windows,-the-Linux-subsystem-way

Raumpilot, 2018-10-07:
@mikeller that sounds promising, but is not possible for me and my computing skills.

I already fail to set up 'the Linux-subsystem' on my mac...
Is it possible to compile the custom Betaflight firmware with a mac?

mikeller, 2018-10-07:
@Raumpilot: Sorry, I should have asked first, but the 'Linux subsystem' instructions are for Windows and not MacOS. MacOS instructions here:

https://github.com/betaflight/betaflight/blob/master/docs/development/Building%20in%20Mac%20OS%20X.md

timman2er, 2018-10-07:
Another option that might be much simpler is flashing an earlier build with gps in it if it flies good enough for you - then later but F4/F7.

Raumpilot, 2018-10-07:
@timman2er but the GPS rescue Mode (that's what I want to use) first came along with bf3.4 as I remember and that was also the Version in witch the GPS support for F3 Boards has been disabled.

@mikeller 
THX I did this and successfully dowloaded the betaflight repository.
I will now try to undefine some of the features I don't need and define the GPS.
Do I need a special Editor for this? Is there anything I absolutely should not do?

mikeller, 2018-10-07:
@Raumpilot: Any text editor will do.

Raumpilot, 2018-10-08:
@mikeller I have done this. and got stuck again while trying to compile. The Terminal says that the ARM GCC 4.9 series compiler is outdated and that I should update to the latest version. 
Any Tips on this? Is it OK to update? 
I thought the 4.9 Version is needed to handle the betaflight repository.

mikeller, 2018-10-08:
@Raumpilot: `make arm_sdk_install` is the easiest way to get the correct version of the compiler to build the firmware (caution, downloads a couple of 100 MB worth of compiler).